espressif/esp-idf #3418: Deep Sleep consumes high current even after call to shutdown WiFi

mreutman, 2019-05-03:
## Environment

- Development Kit: ESP32-DevKitC
- Kit version (for WroverKit/PicoKit/DevKitC): v1
- Module or chip used:  ESP32-WROOM-32
- IDF version (run ``git describe --tags`` to find it): v3.3-beta3-12-g56afb3bd1
- Build System: Make
- Compiler version (run ``xtensa-esp32-elf-gcc --version`` to find it): xtensa-esp32-elf-gcc (crosstool-NG crosstool-ng-1.22.0-80-g6c4433a) 5.2.0
- Operating System:  Linux
- Power Supply: external 3.3V

## Problem Description
I'm trying to get Deep Sleep to work for a project that also uses WiFi. What I see is that if I go into deep sleep without initializing station mode and connecting to WiFi AP, the current consumption drops down to approximately 58uA. However, if I initialize WiFi and connect to nearby AP, attempt to power down WiFi by calling `esp_wifi_stop` (and `esp_wifi_deinit` to be safe), and then go into deep sleep, the current remains at nearly 60mA (where it jumped to when the WiFi on the ESP32 was turned on).

I'm not sure if there is something wrong in the SDK, if I'm using the API incorrectly, or if there is something in my configuration that is causing this.

### Expected Behavior
When calling functions to disable the WiFi and enter deep sleep, the current consumption to drop to around 58uA.
### Actual Behavior

### Steps to reproduce

1. Initialize WiFi on ESP32 in station mode.
2. Connect to local WiFi AP.
3. Power down WiFi using `esp_wifi_stop` and ensuring it disconnects from WiFi AP.
4. Enter deep sleep using timer function (configured using calls to `esp_sleep_enable_timer_wakeup` and `esp_deep_sleep_start` to cause ESP32 to wake up some time in the future.
5. Measure current draw.

### Code to reproduce this issue

I modified the example application in `esp-idf/examples/wifi/getting_started/station` and added the following to the end of the `app_main` function.

```cpp
void app_main()
{
    //Initialize NVS
    esp_err_t ret = nvs_flash_init();
    if (ret == ESP_ERR_NVS_NO_FREE_PAGES || ret == ESP_ERR_NVS_NEW_VERSION_FOUND) {
      ESP_ERROR_CHECK(nvs_flash_erase());
      ret = nvs_flash_init();
    }
    ESP_ERROR_CHECK(ret);
    
    ESP_LOGI(TAG, "ESP_WIFI_MODE_STA");
    wifi_init_sta();

    ESP_LOGI(TAG, "wait....."); // measure current here for consumption under WiFi
    vTaskDelay(30000 / portTICK_PERIOD_MS);
    ESP_LOGI(TAG, "wifi stop");
    esp_wifi_stop();
    vTaskDelay(5000 / portTICK_PERIOD_MS); // wait a bit to make sure WiFi disconnects
    ESP_LOGI(TAG, "deinit");
    esp_wifi_deinit(); // deinit just to be safe
    esp_sleep_enable_timer_wakeup(15 * 1000000); // wake up 15 seconds in the future
    ESP_LOGI(TAG, "deep sleep");
    esp_deep_sleep_start(); // begin deep sleep
}
```

## Debug Logs

```
I (10124) wifi station: ESP_WIFI_MODE_STA                                  
I (10124) wifi: wifi driver task: 3ffb7bb4, prio:23, stack:3072, core=0    
I (10124) wifi: wifi firmware version: 7240fb7                             
I (10124) wifi: config NVS flash: enabled
I (10124) wifi: config nano formating: enabled
I (10124) system_api: Base MAC address is not set, read default base MAC address from BLK0 of EFUSE
I (10124) system_api: Base MAC address is not set, read default base MAC address from BLK0 of EFUSE
I (10124) wifi: Init dynamic tx buffer num: 32
[... 333 lines cut ...]
CONFIG_SPIFFS_API_DBG=
CONFIG_SPIFFS_GC_DBG=
CONFIG_SPIFFS_CACHE_DBG=
CONFIG_SPIFFS_CHECK_DBG=
CONFIG_SPIFFS_TEST_VISUALISATION=

#
# AWS Configuration
#
CONFIG_AWS_IOT_SDK=y
CONFIG_AWS_IOT_MQTT_HOST=""
CONFIG_AWS_IOT_MQTT_PORT=8883
CONFIG_AWS_IOT_MQTT_TX_BUF_LEN=1024
CONFIG_AWS_IOT_MQTT_RX_BUF_LEN=1024
CONFIG_AWS_IOT_MQTT_NUM_SUBSCRIBE_HANDLERS=5
CONFIG_AWS_IOT_MQTT_MIN_RECONNECT_WAIT_INTERVAL=1000
CONFIG_AWS_IOT_MQTT_MAX_RECONNECT_WAIT_INTERVAL=128000
CONFIG_AWS_IOT_OVERRIDE_THING_SHADOW_RX_BUFFER=
CONFIG_AWS_IOT_SHADOW_MAX_SIZE_OF_UNIQUE_CLIENT_ID_BYTES=80
CONFIG_AWS_IOT_SHADOW_MAX_SIMULTANEOUS_ACKS=10
CONFIG_AWS_IOT_SHADOW_MAX_SIMULTANEOUS_THINGNAMES=10
CONFIG_AWS_IOT_SHADOW_MAX_JSON_TOKEN_EXPECTED=120
CONFIG_AWS_IOT_SHADOW_MAX_SHADOW_TOPIC_LENGTH_WITHOUT_THINGNAME=60
CONFIG_AWS_IOT_SHADOW_MAX_SIZE_OF_THING_NAME=40
```

mreutman, 2019-05-04:
I went ahead and tried `v4.0-dev-459-gba1ff1692` and `v3.0.7-53-g9be6f0c0a` releases of esp-idf to see if changing versions would fix the problem and (un)fortunately, it looks like I get the same behavior for those versions as well.

This leads me to believe that the issue lies either in my use of the API or some configuration setting I have.

igrr, 2019-05-04:
@mreutman  You have mentioned that you are using a DevKit-C. Can you please describe the method of current measurement? DevKit-C includes some supporting circuits which are not powered down in sleep, so it is very hard to obtain accurate results, at least not without modifying the PCB.

mreutman, 2019-05-04:
Hi @igrr, thanks for taking some time to help me out! I have a multimeter wired in series to the 3.3V input pin and my PSU when running these tests.

I'm not too concerned with getting accurate results, but simply some degree of power savings that is in the ballpark of what is advertised in the datasheet. Ultimately, I want to use this with a product I am working on that utilizes the ESP32 and has additional circuitry in place. The Dev C kit exhibits the same behavior as my custom board, so I'm trying to use that to debug for now given it is understood by many more individuals.

One thing that seems very concerning is that I can get around 58uA when entering deep sleep mode without enabling WiFi on the Dev C board (i.e. I just comment out the call to `wifi_init_sta` in the application I've been using for testing), but stay at around 60mA when entering deep sleep with WiFi enabled/disabled beforehand. The three orders of magnitude difference between the two tests makes me think that I'm somehow not successfully shutting down the WiFi on the ESP32.

igrr, 2019-05-05:
Is the USB-UART connected to the PC in both cases (when you are doing a no-wifi deep sleep measurement, and when doing a with-wifi deep sleep measurement)? 60mA looks about the amount of current that the USB-Serial chip consumes when it is not in the power-down mode.

mreutman, 2019-05-05:
All tests have been performed with USB disconnected; unit powered by PSU through 3.3V input pin.

I just reran my test case where I put it into deep sleep before ever initializing WiFi and I measured a current draw of 2.3mA on the Dev C board. The 58uA value I initially reported for deep sleep was actually taken from the board I'm working on for a company product. I did reconfirm though that they both idle at around 60mA when trying to go to sleep after WiFi enable/disable. Apologies for the inaccurate data.

Does 2.3mA sound reasonable for a Dev C board with the ESP32 in deep sleep? And similarly, does 60mA sound reasonable for a Dev C board with WiFi enabled? Hoping I'm not overlooking something in my tests.

mreutman, 2019-05-08:
I believe my issue has been solved. It looks like the multi-meter I was using was affecting overall power delivery to the ESP32. As soon as I bumped my PSU's voltage from 3.3V to 4V, the unit entered into deep sleep consuming around 43uA.

I actually think that the meter was possibly even preventing WiFi from initializing, so that 60mA current I observed was some state before WiFi was fully enabled. The switch to 4V causes the ESP32 to jump to ~100mA current consumption where I believe it to be enabling WiFi. This is then followed by the disconnect and deep sleep putting the current at 43uA.

@igrr, thanks for taking the time to help trouble shoot. Sorry for the rabbit chase this caused.

closed 2019-05-08 by mreutman

Stasikiii, 2019-05-20:
Same issue :(
In my case i have 7uA without wifi and 1.3mA after wifi connection...

XxKavosxX, 2020-06-09:
2020, this help topic helped me