espressif/esp-idf #3419: make: *** No rule to make target '/home/mosma/esp/blink/build/ldgen_libraries', needed by '/home/mosma/esp/blink/build/esp32/esp32.project.ld'.  Stop.

Muhammad1997, 2019-05-03:
## Environment

- Development Kit:      [ESP32-DevKitC]
- Kit version (DevKitC): [v1]
- Module or chip used:  [ESP32-WROOM-32]
- IDF version : v4.0-dev-459-gba1ff1692
- Build System:         [Make]
- Compiler version : 5.2.0
- Operating System:     [Windows]
- Power Supply:         [USB]

## Problem Description

i am trying to ``make flash`` but this message appears every time 
i have tried make clean , delete all files and re cloned , searched for 
```/home/mosma/esp/blink/build/esp32/esp32.project.ld ``` or ```/home/mosma/esp/blink/build/ldgen_libraries``` but i didn't find them , tried to make another project but the same problem appeared 
## Debug Logs
```
make: *** No rule to make target '/home/mosma/esp/blink/build/ldgen_libraries', needed by '/home/mosma/esp/blink/build/esp32/esp32.project.ld'.  Stop.
```

[image]

negativekelvin, 2019-05-04:
https://github.com/espressif/esp-idf/issues/3394

vetaldudnik, 2019-05-04:
There is error in file ldgen.mk line 17

Actual:

$(BUILD_DIR_BASE)/ldgen.**section_infos**: $(LDGEN_LIBRARIES) $(IDF_PATH)/make/ldgen.mk

Expected:

$(BUILD_DIR_BASE)/**ldgen_libraries**: $(LDGEN_LIBRARIES) $(IDF_PATH)/make/ldgen.mk

P.S. I'm new user and now I'm studying how to do pull request :)

igrr, 2019-05-04:
Closing as duplicate of #3394. Please follow the link to that issue and click "Subscribe" to be notified when the fix is merged. In the meantime you can apply the patch from https://github.com/espressif/esp-idf/issues/3394#issuecomment-488528047.

closed 2019-05-04 by igrr

Muhammad1997, 2019-05-04:
@igrr  thanks a lot ...  gonna use stable version of it tell its merged

piercalderan, 2019-05-22:
On eclipse it's impossible to add a Windows or Unix like full path.

Espressif says: Click “Add…” again, and enter name IDF_PATH. The value should be the full path where ESP-IDF is installed.
Any idea?

piercalderan, 2019-05-22:
C:/msys32/home/pier/esp/esp-idf/make/project.mk:104: *** IDF_PATH cannot contain colons. If overriding IDF_PATH on Windows, use MSYS Unix-style /c/dir instead of C:/dir.  Stop.
Atter doing a unix like path
makefile:8: /c/msys32/home/pier/esp/esp-idf/make/project.mk: No such file or directory

Vadano a cagare sulle ortiche sia eclipse che espressif!