espressif/esp-idf #3430: What is the cpu type for each core

Herwey, 2019-05-06:
Hi:
Esp32 has two cores, so what is the type cpu for each core ?
Thanks !

negativekelvin, 2019-05-06:
> 1.4.1 CPU and Memory
> • Xtensa® single-/dual-core 32-bit LX6 microprocessor(s), up to 600 MIPS (200 MIPS for ESP32-S0WD, 400
> MIPS for ESP32-D2WD)
> • 448 kB ROM
> • 520 kB SRAM
> • 16 kB SRAM in RTC

Read the datasheet

Alvin1Zhang, 2019-05-06:
Thanks for reporting the issue. Would you please help provide more details as suggested in the [issue template](https://github.com/espressif/esp-idf/blob/master/docs/issue_template.md)? Thanks.

Herwey, 2019-05-06:
@negativekelvin emm... The datasheet shows the cpu type ? which line is ? include two cores?

igrr, 2019-05-06:
@Herwey This issue tracker is for bug reports and feature requests related to the ESP-IDF. For general questions about the ESP32, please use esp32.com forum or use the support page on espressif.com website. For hardware documentation related requests (such as Datasheet or TRM), you can also use this forum topic: https://esp32.com/viewtopic.php?f=5&t=905.

(At the same time, @negativekelvin gave the answer: both cores are Xtensa LX6 cores, [identically configured](https://github.com/espressif/esp-idf/blob/master/components/xtensa/esp32/include/xtensa/config/core-isa.h)).

closed 2019-05-06 by igrr

negativekelvin, 2019-05-06:
Wikipedia shows it too https://en.wikipedia.org/wiki/ESP32