espressif/esp-idf #3431: ESP32 modbus slave example data error

jackyc58, 2019-05-06:
I use the latest (git at 2019/04/30) ESP-IDF and the accompanying example modbus_slave, I tested it to get the following result:

Modbus Master send: 01 01 00 00 00 04 3D C9
Modbus_slave replies to two results:
01 01 01 05 91 8B (The first reply after booting)
40 40 40 41 64 F1 (Reply after the second time)

Q1. “40 40 40 41 64 F1” Is this correct? If so, how do you interpret the content?
Q2. If it is the format of the first reply (01 01 01 05 91 8B), how to set it in the program?

Thanks!

alisitsyn, 2019-05-06:
Hi jackyc58,

Thank you for this issue report. I tried to reproduce your issue with my master but I can not see the same result after second request.  In my case the response is correct: (01 01 01 05 91 8B):

I (293) cpu_start: Starting scheduler on PRO CPU.
I (0) cpu_start: Starting scheduler on APP CPU.
I (295) uart: queue free spaces: 20
COILS READ: time_stamp(us):24237474, mb_addr:0, type:32, st_address:0x3ffb1f90, size:4
COILS READ: time_stamp(us):30086072, mb_addr:0, type:32, st_address:0x3ffb1f90, size:4
COILS READ: time_stamp(us):30334676, mb_addr:0, type:32, st_address:0x3ffb1f90, size:4

Could you answer my questions: 
1. Which Modbus master sw do you use to communicate with Modbus slave example?
2. What is the delay between 1st and second requests.
3. Your selected baud rate.

Please provide other information you think can be helpful to reproduce this issue.
Thank you.

--
Alex

jackyc58, 2019-05-07:
Alex, thank you very much for your help.

1. Modbus master SW: use Modbus RTU Master test software [link](https://ebook.yourplc.net/modules/wfdownloads/singlefile.php?cid=11&lid=103)
2. Interval: I use manual mode, separated by more than 1 second.
3. baud rate: 115200, N, 8, 1

ESP32 module: ESP32-S (manufactured by Ai-Thinker)
Modbus_slave UART config:
   (5) UART RXD pin number
   (17) UART TXD pin number
   (16) UART RTS pin number

Is there a recommended Modbus master SW?
Thanks.

--
Jacky.

jackyc58, 2019-05-07:
I use RealTerm and I have the same result.
[image]

Send FIle: 
[send.txt](https://github.com/espressif/esp-idf/files/3150699/send.txt)

jackyc58, 2019-05-07:
Hi Alex, I found the problem, because of the hardware design, the baud rate only supports 57600, thank you for your help.

closed 2019-05-07 by jackyc58

alisitsyn, 2019-05-07:
Hi Jacky,
It is good that your issue is solved. There is a list of possible Modbus software the can be used. I would recommend you to use ModbusPoll (ModbusSlave) but it is trial SW. Also it is possible to use PyModbus python support which works well with the examples (https://pymodbus.readthedocs.io/en/latest). Also you can use the open source project: https://sourceforge.net/projects/qmodmaster and some other tools.
There are plans to support the Modbus TCP master/slave and they will work through the same interface API but it is on the work in progress stage and postponed.
Thank you for your interest to Espressif Systems products!

--
Alex

jackyc58, 2019-05-07:
Hi Alex, thank you for your valuable comments.

jackyc58, 2019-05-10:
Hi Alex, 
Is there an example code for modbus tcp?

Regard and Thanks,
Jacky.