espressif/esp-idf #3435: How do I increase BT Classic SPP baudrate?

amoghskulkarni, 2019-05-07:
Hello all,

I am working on a relatively simple BT classic App which looks very similar to the [BT classic SPP initiator](https://github.com/espressif/esp-idf/blob/master/examples/bluetooth/bt_spp_acceptor/main/example_spp_acceptor_demo.c) demo app. The GAP callbacks, SPP callbacks and initialization in main are virtually the same. I got everything working, except I could not find any documentation/example using which I could configure the SPP baudrate. The default baudrate seems to be 9600.

Where can I configure the baudrate for the serial port exposed on SPP? 

(P.S. Since I could not find any good examples/results on my searches, I don't even know if this is a valid question. Sorry if it is not, would like to know more about this in that case.)

blueMoodBHD, 2019-05-08:
Yes, the default baudrate seems to be 9600, and there is no API to configure it. But it will not affect the actual data throughput in SPP.
     In the event that a baud rate is set for a particular port through the RFCOMM service interface, that will not affect the actual data throughput in RFCOMM; i.e. RFCOMM does not incur artificial rate limitation or pacing. However, if either device is a type 2 device (relays data onto other media), or if data pacing is done above the RFCOMM service interface in either or both ends, actual throughput will, on average, reflect the baud rate setting.

Alvin1Zhang, 2019-05-16:
@amoghskulkarni Thanks for reporting the issue. Has your issue been fixed? Thanks.

amoghskulkarni, 2019-05-17:
@blueMoodBHD Thanks for the reply. That explains everything. My application doesn't have any type 2 device, so it seems I'm good to go here.

Alvin1Zhang, 2019-05-20:
@amoghskulkarni Thanks for reporting the issue. Feel free to reopen the issue if it still exists. Thanks.

closed 2019-05-20 by Alvin1Zhang