espressif/esp-idf #3436: Usage of UART1 pins

hvishu, 2019-05-08:
In one of our design there are two processors, ESP wroom module and Renesas micro controller.  UART is communication mechanism between them.

From ESP module side we are currently using SD2 and SD3 pins.  and the SPI mode used is DIO.  So we are thinking GPIO9 and GPIO10 are free and can used for UART1.

With this design we are seeing that on application bootup randomly there is crash in nfs_flash_init() or nvs_open().  This is the first function to be called in app_main()

When we remove the UART1 pins connected between Renesas and ESP module, we don't see this issue.  Or when we use the same binary with ESP wroom module reference board, don't see this issue either.

So my question is can something be done, to continue to use GPIO9 and GPIO10 for UART1 by doing some configuration or changing some software?

I do read that from ESP module datasheet it is mentioned that GPIO9 and GPIO10 shouldn't be used for other purposes.
Regards,
Vishwa

negativekelvin, 2019-05-08:
Not really because they are still connected internally to the flash chip so when uart1 is active you are toggling those flash chip pins and gpio9 can still put the chip in hold mode.

hvishu, 2019-05-14:
Ok.  Then it would be better to remove the below from the ESP wroom data sheet at Table 3 of section 2.2.  Here it say's these pins can be used, but at the bottom of the table it say's not recommended to use these pins,  It's pretty confusing.  I am closing this issue, as we used other pins and issue got resolved.  You can choose to correct the documentation.

SHD/SD2* 17 I/O GPIO9, SD_DATA2, SPIHD, HS1_DATA2, U1RXD
SWP/SD3* 18 I/O GPIO10, SD_DATA3, SPIWP, HS1_DATA3, U1TXD
SCS/CMD* 19 I/O GPIO11, SD_CMD, SPICS0, HS1_CMD, U1RTS
SCK/CLK* 20 I/O GPIO6, SD_CLK, SPICLK, HS1_CLK, U1CTS
SDO/SD0* 21 I/O GPIO7, SD_DATA0, SPIQ, HS1_DATA0, U2RTS
SDI/SD1* 22 I/O GPIO8, SD_DATA1, SPID, HS1_DATA1, U2CTS

closed 2019-05-14 by hvishu