espressif/esp-idf #3437: Compiling MenuConfig application fails (make MENUCONFIG)

MartinTJDK, 2019-05-08:
My build system has troubles compiling the MenuConfig application:

$ make
/usr/bin/gcc -c  -I/usr/include/ncursesw -DCURSES_LOC="<curses.h>"  -DNCURSES_WIDECHAR=1 -DLOCALE -MMD  /esp/esp-idf-v3.2/tools/kconfig/mconf.c -o mconf.o
flex -L -Pzconf -ozconf.lex.c /esp/esp-idf-v3.2/tools/kconfig/zconf.l
bison -t -l -p zconf -o zconf.tab.c /esp/esp-idf-v3.2/tools/kconfig/zconf.y
/usr/bin/m4:/usr/share/bison/yacc.c:616: ERROR: end of file in string
/esp/esp-idf-v3.2/tools/kconfig/zconf.y: fatal error: invalid @ in skeleton: @
make: *** [Makefile:344: zconf.tab.c] Error 1

The result is an empty 'zconf.tab.c' file.

What could cause this issue?
Wrong EOF in 'zconf.y'?

Best regards
Martin

negativekelvin, 2019-05-08:
https://github.com/espressif/esp-idf/issues/3412

Alvin1Zhang, 2019-05-13:
@negativekelvin Thanks for the notes. @MartinTJDK Hi, has your issue been fixed? Thanks.

MartinTJDK, 2019-05-13:
It is "fixed" now. The issue occures due to git changing line endings from "LF" to "CRLF" on Windows.
After forcing to use "LF" during clone it works.

Alvin1Zhang, 2019-05-23:
@MartinTJDK Thanks for reporting the issue. Feel free to reopen if the issue still exists. Thanks.

closed 2019-05-23 by Alvin1Zhang