espressif/esp-idf #3438: Mutual Authentication OTA - what's going on?

kifantidis, 2019-05-08:
Hello,
I would like to ask if there is a way to use a TLS connection with mutual authentication in order to achieve OTA updates. I've read [#2688 ](https://github.com/espressif/esp-idf/pull/2688) from Anders Kaloer, in the past,  had written but I can not find anything that works or anything that is implemented in the newer ESP-IDFs. 
The structure "esp_http_client_config_t" of esp_http_client.h file does not contain client certification and key variables. 

Also after implemented the code from Kaloer's post I did get error related to "esp_transport_ssl_set_client_cert_data" and "esp_transport_ssl_set_client_key_data" functions. 

I'm using ESP-IDF v3.1.2-99-gcf5dbadf4 atm and I'm thinking to upgrade to v3.2. 

Any help or guidance would be appreciated.
Best regards, Kostas.

kifantidis, 2019-05-09:
Good morning.

Thank you for the reply mahavirj. I'll let you know about it when I'll have the results. 

Regards, Kostas.

mahavirj, 2019-05-17:
@kifantidis Do you have any update here?

kifantidis, 2019-05-24:
Hello @mahavirj , 
sadly I don't have time this month due to other projects have to be finished before I get to work with ESP32 again. I'll let you know in this post when I get something new. 

Regards, Kostas.

mahavirj, 2019-05-28:
@kifantidis Fixed with https://github.com/espressif/esp-idf/commit/d81c79d79ff3af59c26fc1bb091b6ef9cce61b48. If you still observe any issues, please feel free to raise issue.

closed 2019-05-28 by mahavirj