espressif/esp-idf #3439: OTA fails with xQueueGenericReceive- assert failed!

korstiaanS, 2019-05-08:
Hi,

I want to use the esp_https_ota function in my program.
The same code works fine in a small program but as soon as I want to use it in my program it starts but then crashes:

```
I (45383) esp_https_ota: Starting OTA...
I (45383) esp_https_ota: Writing to partition subtype 17 at offset 0x210000
/Users/skor/esp/esp-idf/components/freertos/queue.c:1448 (xQueueGenericReceive)- assert failed!

abort() was called at PC 0x40093761 on core 1

ELF file SHA256: 55bf5c84c64d26562f0334a52684743eca83700f662a35fdb86e6e786a264512

Backtrace: 0x4008b430:0x3ffd6e50 0x4008b679:0x3ffd6e70 0x40093761:0x3ffd6e90 0x400da10d:0x3ffd6ed0 0x40092a15:0x3ffd7240

```

What can cause this?

Korstiaan

korstiaanS, 2019-05-08:
Hi,

It gave:

0x4008b679: abort at /Users/skor/esp/esp-idf/components/esp32/panic.c:715
0x40093761: xQueueGenericReceive at /Users/skor/esp/esp-idf/components/freertos/queue.c:2044
0x400da10d: gsm_task at /Users/skor/Documents/vscode/SkorTrack/main/gsm.c:1665
0x40092a15: vPortTaskWrapper at /Users/skor/esp/esp-idf/components/freertos/port.c:435

gsm.c is a task and at the moment of OTA is has a loop where it continuously reads the uart:

```
// read data from the UART
        len = uart_read_bytes(GSM_UART, dataRead, readLength, 100 / portTICK_RATE_MS);
```

But, now for the moment, it worked by changing the http_event_handler by adding a bOTABusy!#?
The gsm.c is not changed.
I don't know why?

```
case HTTP_EVENT_ON_DATA:
            if (bOTABusy == false) {
                if (!esp_http_client_is_chunked_response(evt->client)) {
                    ESP_LOGI(TAG, "writing to receive buffer");
                    strncpy(rcv_buffer, (char *)evt->data, evt->data_len);
                }
            }
            break;

```

korstiaanS, 2019-05-08:
Killing the gsm task before the OTA also works but is not so fine in my wanted result.

mahavirj, 2019-05-09:
@korstiaanS Difficult to say from above information, maybe there is some memory corruption, especially in `rcv_buffer` resulting in corrupt queue handle? You may use `esp_set_watchpoint` to set custom watchpoints to debug further.

korstiaanS, 2019-05-09:
No, it cannot be rcv_buffer. I don't come there.
I rather think that the OTA don't like the call xSemaphoreTake.
If I suspend this call (it is in a continuous loop in a task) during OTA it also stays fine.
Also the backstrace points to this task and I don't do anything more.
This and reading from an uart.

```
if (bOTABusy == false) {
            if (xSemaphoreTake(semGSM, (3000 / portTICK_RATE_MS)) == true) {
                uLocalGsmCommand = uGsmCommand;
                uGsmCommand      = nothing_gsm;  // for next cycle
                xSemaphoreGive(semGSM);
            }
        }
```

korstiaanS, 2019-05-09:
Anyway, to make sure, I will delete all tasks just before the OTA to make sure.
Since I can't find the source of the problem I can't take the risk.

mahavirj, 2019-05-10:
@korstiaanS Okay, I will go ahead and close the issue then (since it does not belong to core `esp_https_ota` component). However, please feel free to raise new issue if you still observe something that is not correct.

closed 2019-05-10 by mahavirj

JFPayeur, 2022-08-26:
Got a similar issue with lora chip using hardware interrupt on a pin , and the OTA script
The OTA was Failing 50% of the time.
I've Disabled  the hardware interrupt at the begining off the ".onStart([]() {"
Never happened again 
Might be similar to you , or not IDK