espressif/esp-idf #3442: Touch Sensor Internal Capacitor: is there any?

giordano-cm, 2019-05-08:
Hello, guys. How are you?

I'm working with the ESP-WROOM-32 and i have a question about its internal hardware: do the touch sensors have an internal (real and physical) capacitor? I know some microcontrollers that have one and uses it for doing **hardware calibration** (i.e., they don't do **software calibration**). Does the ESP-WROOM-32 do it? Does any microcontroller from Espressif do it? I have read the main docs from Espressif but i found nothing about it. Surely, i found some chapters talking about the touch sensor and how to implement it, but i didn't found any specific doc that talks about the touch sensor internal hardware, mainly about the capacitor value i have been looking for. Anyone?

Thanks in advance.

giordano-cm, 2019-05-09:
I found out, at ESP32-WROOM-32 Datasheet, in section 5.3, that Cin (pin Capacitance, but not necessarely the touch sensor pin alone) is typically 2pF. Can i consider that the touch sensor pins have this value of internal capacitance? Or maybe consider that the internal touch sensor hardware has this capacitor?

fuAlex, 2019-05-10:
> I found out, at ESP32-WROOM-32 Datasheet, in section 5.3, that Cin (pin Capacitance, but not necessarely the touch sensor pin alone) is typically 2pF. Can i consider that the touch sensor pins have [...]

Hi, ESP32 internal circuit is different from others, it don't have internal capacitance. If you want learn more about the design of touch sensor, please refer this doc <[Touch Sensor Application Note](https://github.com/espressif/esp-iot-solution/blob/master/documents/touch_pad_solution/touch_sensor_design_en.md)>.  
In addition, what's your requirement about  internal capacitor or hardware calibration?

giordano-cm, 2019-05-10:
> Hi, ESP32 internal circuit is different from others, it don't have internal capacitance. If you want learn more about the design of touch sensor, please refer this doc <[Touch Sensor Application [...]

Thanks for the answer.

About the doc you linked, i've read it, but there is no information about internal capacitance or any deeper detail about the touch sensor hardware.

About the requirement, i've seen some microcontrollers that have internal capacitance for better touch sensing and for auto-calibration. I was also worried about the touch sensor controller, if it was hardware or firmware controlling. I found out that it is done by hardware, though. It has a better response than other microcontrollers that do it by software.

The real "problem" is that there is no stack for auto-calibration and the hardware isn't THAT good for doing it (actually, if it is, there is no doc about it). In the same link you sent, the 4.3 section talks about doing it by hand. I do it at the beggining of my code. But if you put something over the touch pads (or maybe there is an overlay and you put water on in, for example), the touch sensing gets crazy, differently from others microcontrollers.

Anyway, you helped me answering the capacitance question. This issue can be closed, but you may write anything more if you want.

Thanks a lot.

fuAlex, 2019-05-10:
Sure. Maybe, I understand your requirement. so, you want to know if the ESP32 support water-proof function in hardware. right ?  In the case, i have something to say:  
- ESP32 don't support water-proof in hardware. but ESP's next chip will support this.
- The internal capacitance has nothing to do with water-proof function. Meanwhile, GPIO outputs a specific masking signal to achieve this function.

giordano-cm, 2019-05-10:
> ESP32 don't support water-proof in hardware. but ESP's next chip will support this.

Wow. Nice! Can you tell us when this is going to happen? When the ESP's next chip is gonna be released?

> The internal capacitance has nothing to do with water-proof function.

So, what is it for? The internal capacitance..

> Meanwhile, GPIO outputs a specific masking signal to achieve this function.

About it, i didn't find nothing documented. Is there an API for it? Can i use it? Or does the ESP do it by itself?

fuAlex, 2019-05-13:
- The official website has not given the release time. So we all need to pay attention to the official website.  
- The manual of the new chip should talk about the hardware design structure of the waterproof design, we  need to wait for it to come out. : )

giordano-cm, 2019-05-13:
Very nice! Thank you so much for the answers. I'll pay attention to the official website.

That's it for now! Thanks!

closed 2019-05-13 by giordano-cm