espressif/esp-idf #3443: Kconfig can't open Kconfig.projbuild ( issue #2812 back ? )

dnguyen76, 2019-05-08:
Hello_World example compiles and run  succesfully with esp-idf commands in msys32

But Compiling Hello_World project with Eclipse Version: 2019-03 (4.11.0) generate this error :-1: 
IOError: C:/msys32/home/Dan/esp/esp-idf/Kconfig:59: Could not open '/home/Dan/esp/esp-idf/components/bootloader/Kconfig.projbuild' (ENOENT: No such file or directory). Perhaps the $srctree environment variable (which was unset) is set incorrectly. Note that the current value of $srctree is saved when the Kconfig instance is created (for consistency and to cleanly separate instances). Also note that e.g. $FOO in a 'source' statement does not refer to the environment variable FOO, but rather to the Kconfig Symbol FOO (which would commonly have 'option env="FOO"' in its definition).

Is issue  # 2812 coming back with  ?   

--------------------------------------------------------------------------------------------
17:49:43 **** Build of configuration Default for project Hello_World ****
make -j4 all 
including /home/Dan/esp/esp-idf/components/app_update/Makefile.projbuild...
including /home/Dan/esp/esp-idf/components/bootloader/Makefile.projbuild...
including /home/Dan/esp/esp-idf/components/bootloader_support/Makefile.projbuild...
including /home/Dan/esp/esp-idf/components/efuse/Makefile.projbuild...
including /home/Dan/esp/esp-idf/components/esp32/Makefile.projbuild...
including /home/Dan/esp/esp-idf/components/esptool_py/Makefile.projbuild...
including /home/Dan/esp/esp-idf/components/mbedtls/Makefile.projbuild...
including /home/Dan/esp/esp-idf/components/nghttp/Makefile.projbuild...
including /home/Dan/esp/esp-idf/components/partition_table/Makefile.projbuild...
including /home/Dan/esp/esp-idf/components/spiffs/Makefile.projbuild...
including /home/Dan/esp/esp-idf/components/ulp/Makefile.projbuild...
including /home/Dan/esp/esp-idf/components/unity/Makefile.projbuild...
Toolchain path: /opt/xtensa-esp32-elf/bin/xtensa-esp32-elf-gcc
Toolchain version: crosstool-ng-1.22.0-80-g6c4433a5
Compiler version: 5.2.0
python /home/Dan/esp/esp-idf/tools/check_python_dependencies.py
Python requirements from C:/msys32/home/Dan/esp/esp-idf\requirements.txt are satisfied.
mkdir -p /home/Dan/esp/hello_world/build/include/config
python /home/Dan/esp/esp-idf/tools/kconfig_new/confgen.py --kconfig /home/Dan/esp/esp-idf/Kconfig --config /home/Dan/esp/hello_world/sdkconfig --env "COMPONENT_KCONFIGS=/home/Dan/esp/esp-idf/components/app_trace/Kconfig /home/Dan/esp/esp-idf/components/bt/Kconfig /home/Dan/esp/esp-idf/components/driver/Kconfig /home/Dan/esp/esp-idf/components/efuse/Kconfig /home/Dan/esp/esp-idf/components/esp32/Kconfig /home/Dan/esp/esp-idf/components/esp_adc_cal/Kconfig /home/Dan/esp/esp-idf/components/esp_event/Kconfig /home/Dan/esp/esp-idf/components/esp_http_client/Kconfig /home/Dan/esp/esp-idf/components/esp_http_server/Kconfig /home/Dan/esp/esp-idf/components/esp_https_ota/Kconfig /home/Dan/esp/esp-idf/components/esp_wifi/Kconfig /home/Dan/esp/esp-idf/components/espcoredump/Kconfig /home/Dan/esp/esp-idf/components/ethernet/Kconfig /home/Dan/esp/esp-idf/components/fatfs/Kconfig /home/Dan/esp/esp-idf/components/freemodbus/Kconfig /home/Dan/esp/esp-idf/components/freertos/Kconfig /home/Dan/esp/esp-idf/components/heap/Kconfig /home/Dan/esp/esp-idf/components/libsodium/Kconfig /home/Dan/esp/esp-idf/components/log/Kconfig /home/Dan/esp/esp-idf/components/lwip/Kconfig /home/Dan/esp/esp-idf/components/mbedtls/Kconfig /home/Dan/esp/esp-idf/components/mdns/Kconfig /home/Dan/esp/esp-idf/components/mqtt/Kconfig /home/Dan/esp/esp-idf/components/nvs_flash/Kconfig /home/Dan/esp/esp-idf/components/openssl/Kconfig /home/Dan/esp/esp-idf/components/pthread/Kconfig /home/Dan/esp/esp-idf/components/spi_flash/Kconfig /home/Dan/esp/esp-idf/components/spiffs/Kconfig /home/Dan/esp/esp-idf/components/tcpip_adapter/Kconfig /home/Dan/esp/esp-idf/components/unity/Kconfig /home/Dan/esp/esp-idf/components/vfs/Kconfig /home/Dan/esp/esp-idf/components/wear_levelling/Kconfig" --env "COMPONENT_KCONFIGS_PROJBUILD=/home/Dan/esp/esp-idf/components/app_update/Kconfig.projbuild /home/Dan/esp/esp-idf/components/bootloader/Kconfig.projbuild /home/Dan/esp/esp-idf/components/esptool_py/Kconfig.projbuild /home/Dan/esp/esp-idf/components/partition_table/Kconfig.projbuild" --env "IDF_CMAKE=n" --output config /home/Dan/esp/hello_world/sdkconfig --output makefile /home/Dan/esp/hello_world/build/include/config/auto.conf --output header /home/Dan/esp/hello_world/build/include/sdkconfig.h
Traceback (most recent call last):
  File "C:/msys32/home/Dan/esp/esp-idf/tools/kconfig_new/confgen.py", line 505, in <module>
    main()
  File "C:/msys32/home/Dan/esp/esp-idf/tools/kconfig_new/confgen.py", line 187, in main
    config = kconfiglib.Kconfig(args.kconfig)
  File "C:/msys32/home/Dan/esp/esp-idf/tools/kconfig_new/kconfiglib.py", line 656, in __init__
    self.top_node)  # prev_node
  File "C:/msys32/home/Dan/esp/esp-idf/tools/kconfig_new/kconfiglib.py", line 1583, in _parse_block
    self._enter_file(sourced_file)
  File "C:/msys32/home/Dan/esp/esp-idf/tools/kconfig_new/kconfiglib.py", line 1167, in _enter_file
    .format(self._filename, self._linenr, e.message))
IOError: C:/msys32/home/Dan/esp/esp-idf/Kconfig:59: Could not open '/home/Dan/esp/esp-idf/components/bootloader/Kconfig.projbuild' (ENOENT: No such file or directory). Perhaps the $srctree environment variable (which was unset) is set incorrectly. Note that the current value of $srctree is saved when the Kconfig instance is created (for consistency and to cleanly separate instances). Also note that e.g. $FOO in a 'source' statement does not refer to the environment variable FOO, but rather to the Kconfig Symbol FOO (which would commonly have 'option env="FOO"' in its definition).
make: *** No rule to make target '/home/Dan/esp/hello_world/build/include/config/auto.conf', needed by '/home/Dan/esp/hello_world/build/bootloader/bootloader.bin'.  Stop.
"make -j4 all" terminated with exit code 2. Build might be incomplete.

17:49:51 Build Failed. 2 errors, 0 warnings. (took 8s.74ms)

dnguyen76, 2019-05-08:
ESP-IDF version used
# git describe --tags --dirty
v4.0-dev-478-gdf61612f8

renzbagaporo, 2019-05-09:
Hi @dnguyen76. This looks like a separate issue, perhaps even related to https://github.com/espressif/esp-idf/pull/3426 since the circumstances seem to be the same. Can you take a look if making the changes in the said PR resolves this?

Dazza0, 2019-05-09:
@dnguyen76  Alternatively, you can also try running `make menuconfig` or `make defconfig` before attempting to build the project in Eclipse

dnguyen76, 2019-05-09:
I had done "make menuconfig" with ESP-IDF v4.0-dev-478-gdf61612f8 , but error still there

I have installed stable ESP-IDF v3.2  ;  It's working , no error with v3.2

ralphs99, 2019-05-14:
Hello, I'm a new user with limited software experience, but I'm also having the same issue; I can build with the ESP-IDF but not with Eclipse. 

I've followed the Espressif setup instructions for Eclipse having set the path environment variables etc so I don't understand why the compiler can't see the file in question? Any help will be greatly appreciated, preferably with simple to understand instructions! For example, I don't understand the content of #3426. 

Other information that may be relevant?
I'm using windows 10-64bit and was instructed during installation that I needed a Java JRE8 which I installed. Using the link to the installer completed C/C++ installation but I received the error: "the eclipse executable launcher was unable to locate its companion shared library" and after some searching learned that the bundle pool may be the problem. Reinstalling with the bundle pool turned off fixed that problem. 

Thanks again for any advice or assistance, Ralph

renzbagaporo, 2019-05-16:
Hi @dnguyen76, have you tried applying the patch as previsouly suggested?

@ralphs99 Thanks for checking out ESP-IDF! So anyways, the modifications in the patch is so that we can reliably detect if the build is being performed on Windows or not. This is because the path style between Windows and Unix and Unix-like operating systems are different, and some of the tools invoked in the build do not recognize Windows-style paths, which is why we need to do conversion. The previous check, `ifdef MSYSTEM` was apparently reported to be an unreliable, so the conversion does not get executed.

ralphs99, 2019-05-17:
Hi Renzbagaporo, 
Thanks for your reply.
I assume you're referring to #3426 "Fixes failed build on Windows (Eclipse). Condition ifdef MSYSTEM not performs. "make" uses /bin/sh as default shell interpreter where variable MSYSTEM is not defined. I suggest to use OS variable instead"?
Sorry for my ignorance but I don't know what the "OS variable" is or how to apply it. Can you please explicitly tell me what I need to do to get Eclipse working with the ESP-IDF? 
Thanks, Ralph

renzbagaporo, 2019-08-26:
Hi @dnguyen76, any updates to this issue?

@ralphs99 You don't have to do anything with regards to #3426. My response a simple explanation of what's happening in #3426. Are you still experiencing an issue? You man open a separate Github Issue and state the details of your problem there. Thanks.

renzbagaporo, 2019-09-23:
Closing due to lack of updates.

closed 2019-09-23 by renzbagaporo

kurta999, 2020-01-08:
I also have the same issue with newly cloned ESP-AT, v3.3.

Any ideas please?

> App "esp-at" version: ESP-AT
Generating esp32.project.ld
Traceback (most recent call last):
  File "C:/msys32/home/Ati/esp/esp-idf/tools/ldgen/ldgen.py", line 113, in <module>
    main()
  File "C:/msys32/home/Ati/esp/esp-idf/tools/ldgen/ldgen.py", line 96, in main
    sdkconfig = SDKConfig(kconfig_file, config_file, args.env)
  File "C:/msys32/home/Ati/esp/esp-idf/tools/ldgen/sdkconfig.py", line 56, in __init__
    self.config = kconfiglib.Kconfig(kconfig_file.name)
  File "C:/msys32/home/Ati/esp/esp-idf/tools/kconfig_new/kconfiglib.py", line 656, in __init__
    self.top_node)  # prev_node
  File "C:/msys32/home/Ati/esp/esp-idf/tools/kconfig_new/kconfiglib.py", line 1583, in _parse_block
    self._enter_file(sourced_file)
  File "C:/msys32/home/Ati/esp/esp-idf/tools/kconfig_new/kconfiglib.py", line 1167, in _enter_file
    .format(self._filename, self._linenr, e.message))
IOError: C:/msys32/home/Ati/esp/esp-idf/Kconfig:59: Could not open 'C:msys32homeAtiespesp-idfcomponentsapp_updateKconfig.projbuild' (ENOENT: No such file or directory). Perhaps the $srctree environment variable (which was unset) is set incorrectly. Note that the current value of $srctree is saved when the Kconfig instance is created (for consistency and to cleanly separate instances). Also note that e.g. $FOO in a 'source' statement does not refer to the environment variable FOO, but rather to the Kconfig Symbol FOO (which would commonly have 'option env="FOO"' in its definition).
make: *** [/home/Ati/esp/esp-idf/components/esp32/Makefile.projbuild:51: /home/Ati/esp/esp-at/build/esp32/esp32.project.ld] Error 1

shahulakthar, 2020-01-08:
@kurta999 I too got same issue with ESP-AT

`shahul@shahul-PC MINGW32 /home/shahul/esp/esp32-at
$ make flash SILENCE=1
Makefile:34: There is no module_config/module_wroom-32,Using module_config/modul      e_esp32_default
/home/shahul/esp/esp32-at/tools/mkfatfs/Makefile.projbuild:13: warning: overridi      ng recipe for target 'clean'
/home/shahul/esp/esp32-at/esp-idf/components/app_update/Makefile.projbuild:48: w      arning: ignoring old recipe for target 'clean'
Toolchain path: /opt/xtensa-esp32-elf/bin/xtensa-esp32-elf-gcc
Toolchain version: crosstool-ng-1.22.0-80-g6c4433a5
Compiler version: 5.2.0
Python requirements from C:/msys32/home/shahul/esp/esp32-at/esp-idf/requirements      .txt are satisfied.

App "esp-at" version: ESP-AT
Generating esp32.project.ld
Traceback (most recent call last):
  File "C:/msys32/home/shahul/esp/esp32-at/esp-idf/tools/ldgen/ldgen.py", line 1      13, in <module>
    main()
  File "C:/msys32/home/shahul/esp/esp32-at/esp-idf/tools/ldgen/ldgen.py", line 9      6, in main
    sdkconfig = SDKConfig(kconfig_file, config_file, args.env)
  File "C:/msys32/home/shahul/esp/esp32-at/esp-idf/tools/ldgen/sdkconfig.py", li      ne 56, in __init__
    self.config = kconfiglib.Kconfig(kconfig_file.name)
  File "C:/msys32/home/shahul/esp/esp32-at/esp-idf/tools/kconfig_new/kconfiglib.      py", line 656, in __init__
    self.top_node)  # prev_node
  File "C:/msys32/home/shahul/esp/esp32-at/esp-idf/tools/kconfig_new/kconfiglib.      py", line 1583, in _parse_block
    self._enter_file(sourced_file)
  File "C:/msys32/home/shahul/esp/esp32-at/esp-idf/tools/kconfig_new/kconfiglib.      py", line 1167, in _enter_file
    .format(self._filename, self._linenr, e.message))
IOError: C:/msys32/home/shahul/esp/esp32-at/esp-idf/Kconfig:59: Could not open '      C:msys32homeshahulespesp32-atesp-idfcomponentsapp_updateKconfig.projbuild' (ENOE      NT: No such file or directory). Perhaps the $srctree environment variable (which       was unset) is set incorrectly. Note that the current value of $srctree is saved       when the Kconfig instance is created (for consistency and to cleanly separate i      nstances). Also note that e.g. $FOO in a 'source' statement does not refer to th      e environment variable FOO, but rather to the Kconfig Symbol FOO (which would co      mmonly have 'option env="FOO"' in its definition).
make: *** [/home/shahul/esp/esp32-at/esp-idf/components/esp32/Makefile.projbuild      :51: /home/shahul/esp/esp32-at/build/esp32/esp32.project.ld] Error 1
`

I don't no what to fix. Please someone help to solve the issue.