espressif/esp-idf #3444: i2c crashes without giving any info

trustmiao, 2019-05-09:
## Environment

- Development Kit:     Goouuu-ESP32
- Module or chip used:  ESP32-WROOM-32
- IDF version (run ``git describe --tags`` to find it):     v3.3-beta1-210-gb173cf281
- Build System:         [Make]
- Compiler version (run ``xtensa-esp32-elf-gcc --version`` to find it):
    5.0.2
- Operating System:     [Windows]
- Power Supply:         [USB]

## Problem Description
cpu reset without giving out any info after operating on i2c after couple thousand reads

### Expected Behavior
no crash/or crash giving some info for debug
### Actual Behavior
cpu reset
### Steps to repropduce
connect a i2c device, run the following code 

// It helps if you attach a picture of your setup/wiring here.

### Code to reproduce this issue

```cpp

`#include <stdio.h>
#include "driver/i2c.h"
#include "driver/uart.h"
#include "sdkconfig.h"

uint8_t _i2cAddr = 0x4B;
#define ACK_EN 0x01
#define DEFAULT_UART_TIMEOUT_MS 30

void app_main()
{
    
    uart_config_t uart_config = {
        .baud_rate = 115200,
        .data_bits = UART_DATA_8_BITS,
        .parity = UART_PARITY_DISABLE,
        .stop_bits = UART_STOP_BITS_1,
        .flow_ctrl = UART_HW_FLOWCTRL_DISABLE};
    uart_param_config(UART_NUM_1, &uart_config);
    uart_set_pin(UART_NUM_1, GPIO_NUM_1, GPIO_NUM_3, UART_PIN_NO_CHANGE, UART_PIN_NO_CHANGE);
    uart_driver_install(UART_NUM_1, 2048 * 2, 0, 0, NULL, 0);

    i2c_config_t conf;
    conf.mode = I2C_MODE_MASTER;
    conf.sda_io_num = 22;
    conf.scl_io_num = 21;
    conf.sda_pullup_en = GPIO_PULLUP_ENABLE;
    conf.scl_pullup_en = GPIO_PULLUP_ENABLE;
    conf.master.clk_speed = 100000; 
    i2c_param_config(I2C_NUM_0, &conf);
    i2c_driver_install(I2C_NUM_0, I2C_MODE_MASTER, 0, 0, 0);

    uint8_t buffer[4];
    esp_err_t errx;
    while (1)
    {
        uart_write_bytes(UART_NUM_1, (const char *)"1,", 2);
        i2c_cmd_handle_t cmd = i2c_cmd_link_create();
        uart_write_bytes(UART_NUM_1, (const char *)"2,", 2);
        cmd = i2c_cmd_link_create();
        uart_write_bytes(UART_NUM_1, (const char *)"3,", 2);
        i2c_master_start(cmd);
        uart_write_bytes(UART_NUM_1, (const char *)"4,", 2);
        i2c_master_write_byte(cmd, (_i2cAddr << 1) | I2C_MASTER_READ, ACK_EN);
        uart_write_bytes(UART_NUM_1, (const char *)"5,", 2);
        i2c_master_read(cmd, buffer, 4, (i2c_ack_type_t)0x02);
        uart_write_bytes(UART_NUM_1, (const char *)"6,", 2);
        i2c_master_stop(cmd);
        uart_write_bytes(UART_NUM_1, (const char *)"7,", 2);
        for (int round = 1; round <= 64; round++)
        {
            errx = i2c_master_cmd_begin(I2C_NUM_0, cmd, DEFAULT_UART_TIMEOUT_MS / portTICK_PERIOD_MS);
            if (errx == ESP_OK)
            {
                break;
            }
            else if ((errx != ESP_OK) && (round < 64))
            {
                continue;
            }
            else
            {
                i2c_cmd_link_delete(cmd);
                uart_write_bytes(UART_NUM_1, (const char *)"receive ERROR\n", 20);
            }
        }
        i2c_cmd_link_delete(cmd);
        uart_write_bytes(UART_NUM_1, (const char *)"8\n", 2);
    }
    vTaskDelete(NULL);
}
`
```
// If your code is longer than 30 lines, [GIST](https://gist.github.com) is preferred.

## Debug Logs

```

we have a program monitor the uart output when it crashes, here is what we got:

`1,1|2|3|4|5|2,3
1,1|2|3|4|5|2,3
1,1|2|3|4|5|ets Jun  8 2016 00:22:57

rst:0xc (SW_CPU_RESET),boot:0x13 (SPI_FAST_FLASH_BOOT)
count of rcv: 2253 , millis:10712240
1,1|2|3|3
1,1|2|3|4|5|2,3
1,1|2ets Jun  8 2016 00:22:57

rst:0xc (SW_CPU_RESET),boot:0x13 (SPI_FAST_FLASH_BOOT)
count of rcv: 8919 , millis:34073118
1,1|2|3|4|5|2,3
1,1|2|3|4|5|2,3
1ets Jun  8 2016 00:22:57

rst:0xc (SW_CPU_RESET),boot:0x13 (SPI_FAST_FLASH_BOOT)
count of rcv: 6913 , millis:26023427
1,1|2|3|4|5|2,3
1,1|2|3|4|5|2,3
1,1|2|3|4|5|2,3ets Jun  8 2016 00:22:57

rst:0xc (SW_CPU_RESET),boot:0x13 (SPI_FAST_FLASH_BOOT)
count of rcv: 6908 , millis:25861603
1,1|2|3|4|5|2,3
1,1|2|3|4|5|2,3
1,1|2|3|4|5ets Jun  8 2016 00:22:57

rst:0xc (SW_CPU_RESET),boot:0x13 (SPI_FAST_FLASH_BOOT)
count of rcv: 6913 , millis:25928563
1,1|2|3|4|5|2,3
1,1|2|3|4|5|2,3
ets Jun  8 2016 00:22:57

rst:0xc (SW_CPU_RESET),boot:0x13 (SPI_FAST_FLASH_BOOT)
count of rcv: 6907 , millis:25790741
1,1|2|3|4|5|2,3
1,1|2|3|4|5|2,3
1,1|2|3|4|5|2,3ets Jun  8 2016 00:22:57

rst:0xc (SW_CPU_RESET),boot:0x13 (SPI_FAST_FLASH_BOOT)
count of rcv: 6913 , millis:25981870`
```

## Other items if possible

- [ ] sdkconfig file (attach the sdkconfig file from your project folder)
- [ ] elf file in the ``build`` folder (**note this may contain all the code details and symbols of your project.**)
- [ ] coredump (This provides stacks of tasks.)

trustmiao, 2019-05-09:
taking out uart part, I got back trace:
`abort() was called at PC 0x4008271f on core 0

Backtrace: 0x40084dec:0x3ffb47d0 0x40085015:0x3ffb47f0 0x4008271f:0x3ffb4810 0x40082740:0x3ffb4830 0x4008288d:0x3ffb4860 0x400d8b12:0x3ffb4880 0x400dbd6d:0x3ffb4b90 0x40082695:0x3ffb4bc0 0x400e18f7:0x3ffb4c10 0x400e2637:0x3ffb4c30 0x400d2339:0x3ffb4c70 0x400d0b22:0x3ffb4cb0 0x4008776d:0x3ffb4cd0

Rebooting...
ets Jun  8 2016 00:22:57

rst:0xc (SW_CPU_RESET),boot:0x13 (SPI_FAST_FLASH_BOOT)
`

koobest, 2019-05-09:
Hi, @trustmiao 
If you tried 64 times and all failed, "i2c_cmd_link_delete" will be executed twice. I think this caused a crash.

trustmiao, 2019-05-09:
We will rewrite it in better way, however, that wasn't where the esp32 reset, we could take out the whole part of code in  for {} , and it still resets

koobest, 2019-05-09:
@trustmiao 
Do you have WiFi task or  some of the tasks that will erase the flash?

trustmiao, 2019-05-09:
i don't think so, is there a way to check just for sure?

trustmiao, 2019-05-09:
address decoded with idf monitor 

``

abort() was called at PC 0x4008271f on core 0
0x4008271f: lock_init_generic at G:/msys32/home/deyeen/esp/esp-idf/components/newlib/locks.c:81

Backtrace: 0x40084dec:0x3ffb47d0 0x40085015:0x3ffb47f0 0x4008271f:0x3ffb4810 0x40082740:0x3ffb4830 0x4008288d:0x3ffb4860 0x400d8b12:0x3ffb4880 0x400dbd6d:0x3ffb4b90 0x40082695:0x3ffb4bc0 0x400e18f7:0x3ffb4c10 0x400e2637:0x
3ffb4c30 0x400d2339:0x3ffb4c70 0x400d0b22:0x3ffb4cb0 0x4008776d:0x3ffb4cd0
0x40084dec: invoke_abort at G:/msys32/home/deyeen/esp/esp-idf/components/esp32/panic.c:707

0x40085015: abort at G:/msys32/home/deyeen/esp/esp-idf/components/esp32/panic.c:707

0x4008271f: lock_init_generic at G:/msys32/home/deyeen/esp/esp-idf/components/newlib/locks.c:81

0x40082740: lock_acquire_generic at G:/msys32/home/deyeen/esp/esp-idf/components/newlib/locks.c:134

0x4008288d: _lock_acquire_recursive at G:/msys32/home/deyeen/esp/esp-idf/components/newlib/locks.c:171

0x400d8b12: _vfprintf_r at /Users/ivan/e/newlib_xtensa-2.2.0-bin/newlib_xtensa-2.2.0/xtensa-esp32-elf/newlib/libc/stdio/../../../.././newlib/libc/stdio/vfprintf.c:860 (discriminator 2)

0x400dbd6d: vprintf at /Users/ivan/e/newlib_xtensa-2.2.0-bin/newlib_xtensa-2.2.0/xtensa-esp32-elf/newlib/libc/stdio/../../../.././newlib/libc/stdio/vprintf.c:39

0x40082695: esp_log_write at G:/msys32/home/deyeen/esp/esp-idf/components/log/log.c:215

0x400e18f7: i2c_cmd_link_append at G:/msys32/home/deyeen/esp/esp-idf/components/driver/i2c.c:1036

0x400e2637: i2c_master_stop at G:/msys32/home/deyeen/esp/esp-idf/components/driver/i2c.c:1036

0x400d2339: app_main at G:/esp32_apps/i2c_bno_debug/main/i2c_example_main.c:50

0x400d0b22: main_task at G:/msys32/home/deyeen/esp/esp-idf/components/esp32/cpu_start.c:518

0x4008776d: vPortTaskWrapper at G:/msys32/home/deyeen/esp/esp-idf/components/freertos/port.c:403
``

koobest, 2019-05-09:
Please check the return value of `i2c_cmd_link_create()`

trustmiao, 2019-05-10:
thanks alot, problem solved, we should not i2c_cmd_link_create twice.

closed 2019-05-10 by trustmiao