espressif/esp-idf #3445: Can't recive Broadcast message by lwip_recvfrom

nopnop2002, 2019-05-09:
My Environment:

	freeRTOS version:V8.2.0
	NEWLIB version:2.2.0
	lwIP version:2-0-3-255
	ESP-IDF version:v4.0-dev-311-g70eda3d22-dirty

	$ git describe --tags
	v4.0-dev-311-g70eda3d22

Problem Description:
Can't recive Broadcast message by lwip_recvfrom

Transmitter Side Source:
Transmitting is sending broadcast UDP every 500Tick.

	#include <string.h>
	#include "freertos/FreeRTOS.h"
	#include "freertos/task.h"
	#include "freertos/event_groups.h"
	#include "esp_system.h"
	#include "esp_wifi.h"
	#include "esp_event_loop.h"
	#include "esp_log.h"
	#include "nvs_flash.h"

	#include "lwip/err.h"
	#include "lwip/sockets.h"
	#include "lwip/sys.h"
	#include "lwip/netdb.h"
	#include "lwip/dns.h"

	/* The examples use WiFi configuration that you can set via 'make menuconfig'.

	   If you'd rather not, just change the below entries to strings with
	   the config you want - ie #define EXAMPLE_WIFI_SSID "mywifissid"
	*/
	#define EXAMPLE_ESP_WIFI_SSID      CONFIG_ESP_WIFI_SSID
	#define EXAMPLE_ESP_WIFI_PASS      CONFIG_ESP_WIFI_PASSWORD
	#define EXAMPLE_MAX_STA_CONN       CONFIG_MAX_STA_CONN

	#define UDP_PORT 9876

	QueueHandle_t xQueue;
	TimerHandle_t timerHandle;
	//static EventGroupHandle_t s_wifi_event_group;

	UBaseType_t uxQueueLength = 10;
	UBaseType_t uxItemSize = 64;

	static const char *TAG = "wifi softAP";

	// Timer call back
	static void timer_cb(TimerHandle_t xTimer)
	{
	    char buffer[uxItemSize];
	    TickType_t nowTick;
	    nowTick = xTaskGetTickCount();
	    ESP_LOGI(pcTaskGetTaskName(0),"Start");
	    sprintf(buffer,"Hello World!! %d",nowTick);
	    xQueueSend(xQueue, &buffer, 0);
	}

	// Bradcast Task
	void BroadSend(void *pvParameters)
	{
	  ESP_LOGI(pcTaskGetTaskName(0),"Start");

	  /* set up address to connect to */
	  struct sockaddr_in addr;
[... 368 lines cut ...]
	I (220) system_api: Base MAC address is not set, read default base MAC address from BLK0 of EFUSE
	I (230) system_api: Base MAC address is not set, read default base MAC address from BLK0 of EFUSE
	I (260) wifi: Init dynamic tx buffer num: 32
	I (260) wifi: Init data frame dynamic rx buffer num: 32
	I (260) wifi: Init management frame dynamic rx buffer num: 32
	I (270) wifi: Init management short buffer num: 32
	I (270) wifi: Init static rx buffer size: 1600
	I (270) wifi: Init static rx buffer num: 10
	I (280) wifi: Init dynamic rx buffer num: 32
	I (380) phy: phy_version: 4100, 2a5dd04, Jan 23 2019, 21:00:07, 0, 0
	I (380) wifi: mode : sta (30:ae:a4:ca:e1:40)
	I (380) wifi station: wifi_init_sta finished.
	I (380) wifi station: connect to ap SSID:PixRacer password:pixracer
	I (390) BroadRead: Start
	I (500) wifi: new:<1,1>, old:<1,0>, ap:<255,255>, sta:<1,1>, prof:1
	I (510) wifi: state: init -> auth (b0)
	I (510) wifi: state: auth -> assoc (0)
	I (520) wifi: state: assoc -> run (10)
	I (540) wifi: connected with PixRacer, channel 1, bssid = 30:ae:a4:ff:24:b1
	I (550) wifi: pm start, type: 1

	I (1160) event: sta ip: 192.168.4.2, mask: 255.255.255.0, gw: 192.168.4.1
	I (1160) wifi station: got ip:192.168.4.2
	I (1160) BroadRead: xEventGroupWaitBits WIFI_CONNECTED_BIT
	I (1170) BroadRead: socket fd=54

igrr, 2019-05-09:
Can you try calling `esp_wifi_set_ps(WIFI_PS_NONE);` on the receiver side?

nopnop2002, 2019-05-09:
Thank you for quick replay.
I added esp_wifi_set_ps(WIFI_PS_NONE) in wifi_init_sta().

	void wifi_init_sta()
	{
	    xEventGroup = xEventGroupCreate();

	    tcpip_adapter_init();
	    ESP_ERROR_CHECK(esp_event_loop_init(event_handler, NULL) );

	    wifi_init_config_t cfg = WIFI_INIT_CONFIG_DEFAULT();
	    ESP_ERROR_CHECK(esp_wifi_init(&cfg));
	    wifi_config_t wifi_config = {
	        .sta = {
	            .ssid = EXAMPLE_ESP_WIFI_SSID,
	            .password = EXAMPLE_ESP_WIFI_PASS
	        },
	    };

	    ESP_ERROR_CHECK(esp_wifi_set_mode(WIFI_MODE_STA) );
	    ESP_ERROR_CHECK(esp_wifi_set_config(ESP_IF_WIFI_STA, &wifi_config) );
      //-----------------------------------------------
	    ESP_ERROR_CHECK(esp_wifi_set_ps(WIFI_PS_NONE));
      //-----------------------------------------------
	    ESP_ERROR_CHECK(esp_wifi_start() );

	    ESP_LOGI(TAG, "wifi_init_sta finished.");
	    ESP_LOGI(TAG, "connect to ap SSID:%s password:%s",
	             EXAMPLE_ESP_WIFI_SSID, EXAMPLE_ESP_WIFI_PASS);
	}

It's work fine.
Recerver side Log
Receiver side can receive all broadcast UDP.

	I (0) cpu_start: App cpu up.
	I (383) heap_init: Initializing. RAM available for dynamic allocation:
	I (390) heap_init: At 3FFAE6E0 len 00001920 (6 KiB): DRAM
	I (396) heap_init: At 3FFB8E28 len 000271D8 (156 KiB): DRAM
	I (402) heap_init: At 3FFE0440 len 00003AE0 (14 KiB): D/IRAM
	I (408) heap_init: At 3FFE4350 len 0001BCB0 (111 KiB): D/IRAM
	I (415) heap_init: At 40090F50 len 0000F0B0 (60 KiB): IRAM
	I (421) cpu_start: Pro cpu start user code
	I (103) cpu_start: Starting scheduler on PRO CPU.
	I (0) cpu_start: Starting scheduler on APP CPU.
	I (169) wifi station: ESP_WIFI_MODE_STA
	I (169) wifi: wifi driver task: 3ffc0a0c, prio:23, stack:3584, core=0
	I (169) wifi: wifi firmware version: 955b7af
	I (169) wifi: config NVS flash: enabled
	I (179) wifi: config nano formating: disabled
	I (179) system_api: Base MAC address is not set, read default base MAC address from BLK0 of EFUSE
	I (189) system_api: Base MAC address is not set, read default base MAC address from BLK0 of EFUSE
	I (219) wifi: Init dynamic tx buffer num: 32
	I (219) wifi: Init data frame dynamic rx buffer num: 32
	I (219) wifi: Init management frame dynamic rx buffer num: 32
	I (229) wifi: Init management short buffer num: 32
	I (229) wifi: Init static rx buffer size: 1600
	I (239) wifi: Init static rx buffer num: 10
	I (239) wifi: Init dynamic rx buffer num: 32
	I (259) wifi: Set ps type: 0

	I (349) phy: phy_version: 4100, 6fa5e27, Jan 25 2019, 17:02:06, 0, 0
	I (349) wifi: mode : sta (30:ae:a4:24:bc:5c)
	I (349) wifi station: wifi_init_sta finished.
	I (359) wifi station: connect to ap SSID:PixRacer password:pixracer
	I (359) BroadRead: Start
	I (479) wifi: new:<1,1>, old:<1,0>, ap:<255,255>, sta:<1,1>, prof:1
	I (1459) wifi: state: init -> auth (b0)
	I (1459) wifi: state: auth -> assoc (0)
	I (1469) wifi: state: assoc -> run (10)
	I (1479) wifi: connected with PixRacer, channel 1, bssid = 3c:71:bf:4f:c1:a1
	I (1489) wifi: pm start, type: 0

	I (2169) event: sta ip: 192.168.4.2, mask: 255.255.255.0, gw: 192.168.4.1
	I (2169) wifi station: got ip:192.168.4.2
	I (2169) BroadRead: xEventGroupWaitBits WIFI_CONNECTED_BIT
	I (2179) BroadRead: socket fd=54
	I (6479) BroadRead: lwip_recv ret=19
	I (6479) BroadRead: lwip_recv buffer=Hello World!! 12590
	I (11479) BroadRead: lwip_recv ret=19
	I (11489) BroadRead: lwip_recv buffer=Hello World!! 13090
	I (16479) BroadRead: lwip_recv ret=19
	I (16479) BroadRead: lwip_recv buffer=Hello World!! 13590
	I (21479) BroadRead: lwip_recv ret=19
	I (21479) BroadRead: lwip_recv buffer=Hello World!! 14090
	I (26479) BroadRead: lwip_recv ret=19
	I (26479) BroadRead: lwip_recv buffer=Hello World!! 14590
	I (31479) BroadRead: lwip_recv ret=19
	I (31479) BroadRead: lwip_recv buffer=Hello World!! 15090

Please close this issues.
Thank you.

closed 2019-05-10 by Alvin1Zhang