espressif/esp-idf #3446: gatt_security_server Example: Added services are displayed twice at nrfConnect App

secureschwob, 2019-05-09:
## Environment

- Development Kit:      [ESP32-Wrover-Kit]
- Kit version (for WroverKit/PicoKit/DevKitC): [v4]
- Module or chip used:  [ESP32-WROVER|]
- IDF version: v4.0-dev-478-gdf61612f8
- Build System:         [Make]
- Compiler version : xtensa-esp32-elf-gcc (crosstool-NG crosstool-ng-1.22.0-80-g6c4433a) 5.2.0
- Operating System:     [Ubuntu]
- Power Supply:         [USB]

## Problem Description
I extended the `gatt_security_server` example provided by `esp-idf` with additional services, namely the Battery Level Service and the Environmental Sensing Service. In order to do so, I followed the tutorials provided in the example folder and copied the lines of code used to create the Heart Rate Service in the original example.
I played around with any combination of HRS, Battery Level and ESS and then connect to the ESP32 using the nrfConnect Android and Linux apps.
As soon as I have created more than one GATT service at the ESP32, I see every service created *twice* (see screenshot. Let's number the services 1 to 4 beginning from the topmost Battery Service). I can read the characteristics of services 1 to 4 to get the dummy values written into the source code. However, *only reading services 3 and 4* (i.e. the Battery Service and the Environmental Sensing Service displayed at the bottom of the screenshot) trigger an event in the debug logs on the monitor (e.g. `V (1491156) HWS_SENSOR: event = 1`).

Stepping through my Sourcecode with GDB shows, that the debug log messages
```text
E (1076) BT_GATT: Active Service Found 
E (1076) BT_GATT: Duplicate Service start - Service already started
```
appear directly after having stepped over line 56 of the following GDB log:
```sh
739         ret = esp_ble_gatts_app_register(ESP_BATTERY_APP_ID);
(gdb) s
Target halted. PRO_CPU: PC=0x400EFF42 (active)    APP_CPU: PC=0x4014C44A 
esp_ble_gatts_app_register (app_id=87)
	at /opt/esp32_Development/esp-idf/components/bt/bluedroid/api/esp_gatts_api.c:44
44          ESP_BLUEDROID_STATUS_CHECK(ESP_BLUEDROID_STATUS_ENABLED);
(gdb) l
39      esp_err_t esp_ble_gatts_app_register(uint16_t app_id)
40      {
41          btc_msg_t msg;
42          btc_ble_gatts_args_t arg;
43
44          ESP_BLUEDROID_STATUS_CHECK(ESP_BLUEDROID_STATUS_ENABLED);
45          
46          //if (app_id < ESP_APP_ID_MIN || app_id > ESP_APP_ID_MAX) {
47          if (app_id > ESP_APP_ID_MAX) {
48              return ESP_ERR_INVALID_ARG;
(gdb) s
Target halted. PRO_CPU: PC=0x400EFD6F (active)    APP_CPU: PC=0x4014C44A 
esp_bluedroid_get_status ()
	at /opt/esp32_Development/esp-idf/components/bt/bluedroid/api/esp_bt_main.c:28
28          if (bd_already_init) {
(gdb) n
29              if (bd_already_enable) {
(gdb) n
30                  return ESP_BLUEDROID_STATUS_ENABLED;
(gdb) n
esp_ble_gatts_app_register(app_id=85)
	at /opt/esp32_Development/esp-idf/components/bt/bluedroid/api/esp_gatts_api.c:47
47          if (app_id > ESP_APP_ID_MAX) {
(gdb) n
51          msg.sig = BTC_SIG_API_CALL;
(gdb) n
52          msg.pid = BTC_PID_GATTS;
(gdb) n
53          msg.act = BTC_GATTS_ACT_APP_REGISTER;
(gdb) n
54          arg.app_reg.app_id = app_id;
(gdb) n
#stepping over this line prints "Active Service Found" message shown above
56          return (btc_transfer_context(&msg, &arg, sizeof(btc_ble_gatts_args_t), NULL) == BT_STATUS_SUCCESS ? ESP_OK : ESP_FAIL);	
```
Trying to step into `btc_transfer_context` causes my debugger to stop, thus I could not dig deeper.

Am I mistaken about how to add services? 

### Expected Behavior
Every service created should be visible only once.

### Actual Behavior
Every service created is shown twice.

### Steps to reproduce
1. Add the code snippets posted below to the `gatt_security_server` example provided by `esp-idf`
2. use the nrfConnect or similar app to connect to the ESP32 and discover services

### Code to reproduce this issue
Please see relevant snippets at https://gist.github.com/secureschwob/8e829e909a695b1338f269e07950c618.

## Debug Logs

```
I (606) phy: phy_version: 4100, 2a5dd04, Jan 23 2019, 21:00:07, 0, 0
I (846) HWS_SENSOR: app_main init bluetooth
D (846) nvs: nvs_open_from_partition bt_config.conf 1
D (846) nvs: nvs_get_str_or_blob bt_cfg_key0
D (846) nvs: nvs_get_str_or_blob bt_cfg_key0
D (856) nvs: nvs_close 2
D (856) nvs: nvs_open_from_partition bt_config.conf 1
D (866) nvs: nvs_set_blob bt_cfg_key0 475
D (876) nvs: nvs_close 3
V (996) HWS_SENSOR: event = 0

V (1006) HWS_SENSOR: event = 0

V (1006) HWS_SENSOR: event = 16

I (1006) HWS_SENSOR: The number handle = 3
V (1016) HWS_SENSOR: event = 16

I (1016) HWS_SENSOR: The number handle = 3
V (1016) HWS_SENSOR: event = 16

I (1026) HWS_SENSOR: The number handle = 3
V (1026) HWS_SENSOR: event = 16

I (1026) HWS_SENSOR: The number handle = 3
V (1046) HWS_SENSOR: event = 16

I (1046) HWS_SENSOR: The number handle = 3
V (1046) HWS_SENSOR: event = 16

I (1046) HWS_SENSOR: The number handle = 3
V (1056) HWS_SENSOR: event = 16

I (1056) HWS_SENSOR: The number handle = 3
V (1056) HWS_SENSOR: event = 16

I (1066) HWS_SENSOR: The number handle = 3
V (1066) HWS_SENSOR: GAP_EVT, event 22

E (1076) BT_GATT: Active Service Found 
E (1076) BT_GATT: Duplicate Service start - Service already started
E (1086) BT_GATT: Active Service Found 
E (1086) BT_GATT: Duplicate Service start - Service already started
E (1096) BT_GATT: Active Service Found 
E (1096) BT_GATT: Duplicate Service start - Service already started
V (1106) HWS_SENSOR: GAP_EVT, event 22

E (1106) BT_GATT: Active Service Found 
E (1116) BT_GATT: Duplicate Service start - Service already started
E (1116) BT_GATT: Active Service Found 
E (1126) BT_GATT: Duplicate Service start - Service already started
E (1136) BT_GATT: Active Service Found 
E (1136) BT_GATT: Duplicate Service start - Service already started
V (1146) HWS_SENSOR: event = c

V (1146) HWS_SENSOR: event = c

V (1146) HWS_SENSOR: event = c

V (1156) HWS_SENSOR: event = c

V (1156) HWS_SENSOR: event = c

V (1156) HWS_SENSOR: event = c

V (1166) HWS_SENSOR: event = c

V (1166) HWS_SENSOR: event = c

V (1166) HWS_SENSOR: event = c

V (1176) HWS_SENSOR: event = c

V (1176) HWS_SENSOR: event = c

V (1186) HWS_SENSOR: event = c

V (1186) HWS_SENSOR: event = c

V (1186) HWS_SENSOR: event = c

V (1186) HWS_SENSOR: event = c

V (1196) HWS_SENSOR: event = c

V (1196) HWS_SENSOR: GAP_EVT, event 0

V (1196) HWS_SENSOR: GAP_EVT, event 1

V (1206) HWS_SENSOR: GAP_EVT, event 0

V (1206) HWS_SENSOR: GAP_EVT, event 1

V (1226) HWS_SENSOR: GAP_EVT, event 6

I (1226) HWS_SENSOR: advertising start success
V (1236) HWS_SENSOR: GAP_EVT, event 6

I (1236) HWS_SENSOR: advertising start success
V (1246) HWS_SENSOR: GAP_EVT, event 6

I (1246) HWS_SENSOR: advertising start success
V (1246) HWS_SENSOR: event = 0

V (1246) HWS_SENSOR: event = 0

E (1256) BT_BTM: Advertising or scaning now, can't set privacy 
V (1266) HWS_SENSOR: event = 16

I (1266) HWS_SENSOR: The number handle = 3
V (1266) HWS_SENSOR: event = 16

I (1266) HWS_SENSOR: The number handle = 3
V (1276) HWS_SENSOR: event = 16

I (1276) HWS_SENSOR: The number handle = 3
V (1286) HWS_SENSOR: event = 16

I (1286) HWS_SENSOR: The number handle = 3
E (1296) BT_BTM: Advertising or scaning now, can't set privacy 
V (1306) HWS_SENSOR: event = 16

I (1306) HWS_SENSOR: The number handle = 3
V (1306) HWS_SENSOR: event = 16

I (1306) HWS_SENSOR: The number handle = 3
V (1316) HWS_SENSOR: event = 16

I (1316) HWS_SENSOR: The number handle = 3
V (1326) HWS_SENSOR: event = 16

I (1326) HWS_SENSOR: The number handle = 3
V (1326) HWS_SENSOR: GAP_EVT, event 22

E (1336) HWS_SENSOR: config local privacy failed, error status = 1
E (1346) BT_GATT: Active Service Found 
E (1346) BT_GATT: Duplicate Service start - Service already started
E (1356) BT_GATT: Active Service Found 
E (1356) BT_GATT: Duplicate Service start - Service already started
V (1366) HWS_SENSOR: GAP_EVT, event 22

E (1366) HWS_SENSOR: config local privacy failed, error status = 1
E (1376) BT_GATT: Active Service Found 
E (1376) BT_GATT: Duplicate Service start - Service already started
E (1386) BT_GATT: Active Service Found 
E (1396) BT_GATT: Duplicate Service start - Service already started
E (1396) BT_GATT: Active Service Found 
E (1406) BT_GATT: Duplicate Service start - Service already started
E (1406) BT_GATT: Active Service Found 
E (1416) BT_GATT: Duplicate Service start - Service already started
V (1426) HWS_SENSOR: event = c

V (1426) HWS_SENSOR: event = c

V (1426) HWS_SENSOR: event = c

V (1436) HWS_SENSOR: event = c

V (1436) HWS_SENSOR: event = c

V (1436) HWS_SENSOR: event = c

V (1446) HWS_SENSOR: event = c

V (1446) HWS_SENSOR: event = c

V (1446) HWS_SENSOR: event = c

V (1456) HWS_SENSOR: event = c

V (1456) HWS_SENSOR: event = c

V (1456) HWS_SENSOR: event = c

V (1456) HWS_SENSOR: event = c

V (1466) HWS_SENSOR: event = c

V (1466) HWS_SENSOR: event = c

V (1466) HWS_SENSOR: event = c
```

Weijian-Espressif, 2019-05-09:
@secureschwob, Can you provide your test code?

secureschwob, 2019-05-09:
Dear @Weijian-Espressif , thanks for your fast response.
All lines of code, which modified the original `esp-idf/examples/bluetooth/gatt_security_server/` example can be found at https://gist.github.com/secureschwob/8e829e909a695b1338f269e07950c618 . 
Are you missing something particular?

Weijian-Espressif, 2019-05-09:
@secureschwob, Sorry I can not open the URL, probably because of China's firewall. Can you send an attachment?

secureschwob, 2019-05-09:
@Weijian-Espressif Please find the source code from the Gist below:

```cpp
/* File: example_ble_sec_gatts_demo.h */

// Battery Service
enum {
	BS_IDX_SVC,

	BS_IDX_MEAS_CHAR,
	BS_IDX_MEAS_VAL,

	BS_IDX_NB,
};

// Environmental Sensing Service (ESS) (partly implemented)
enum {
	ESS_IDX_SVC,            // ESS service index

	ESS_IDX_ES_TEMP_MEAS_CHAR,   // Characteristic index
	ESS_IDX_ES_TEMP_MEAS_VAL,    // Characteristic value index
	 ESS_IDX_ES_MEAS_NTF_CFG,     ESS notifications configuration index

	ESS_IDX_NB,             // number of table elements
};

/* File: example_ble_sec_gatts_demo.c */
#include "example_ble_sec_gatts_demo.h"
#include "environmental_sensing_service.h"	#includes UUIDs of the service not found in 

#define PROFILE_NUM                 2

#define BATTERY_PROFILE_APP_IDX         0
#define ESP_BATTERY_APP_ID              0x57	//arbitrarily chosen
#define BATTERY_SVC_INST_ID             0
static uint16_t battery_service_handle_table[BS_IDX_NB];

#define ESS_PROFILE_APP_IDX         1
#define ESP_ESS_APP_ID              0x75    //arbitrarily chosen
#define ESS_SVC_INST_ID             0
static uint16_t ess_handle_table[ESS_IDX_NB];

static struct gatts_profile_inst profile_tab[PROFILE_NUM] = {
		[BATTERY_PROFILE_APP_IDX] = {
			.gatts_cb = gatts_profile_event_handler,    //TODO set another event handler?
			.gatts_if = ESP_GATT_IF_NONE,
		},
	[ESS_PROFILE_APP_IDX] = {
		.gatts_cb = gatts_profile_event_handler, //TODO set another event handler?
		.gatts_if = ESP_GATT_IF_NONE,
	},
};

/*Environmental Sensing PROFILE ATTRIBUTES */
// Environmental Sensing Service
static const uint16_t ess_svc = ESS_UUID_SERVICE;   //0x181A

/// ESS Characteristic
static const uint16_t temperature_uuid = ESS_UUID_TEMP_CHAR;    //0x2A6E
static const uint8_t temperature_value[1] = {0x0A};

static const esp_gatts_attr_db_t ess_gatt_db[ESS_IDX_NB] = {
	// Environmental Sensing Service Declaration
	[ESS_IDX_SVC]                    =
	{{ESP_GATT_AUTO_RSP}, {ESP_UUID_LEN_16, (uint8_t *)&primary_service_uuid, ESP_GATT_PERM_READ,
	  sizeof(uint16_t), sizeof(ess_svc), (uint8_t *)&ess_svc}},

	// Battery Service Characteristic Declaration - wenn ich das auskommentiere sehe ich nur einen Leeren Battery Service mit dem ich nichts machen kann
	[ESS_IDX_ES_TEMP_MEAS_CHAR]  =
	{{ESP_GATT_AUTO_RSP}, {ESP_UUID_LEN_16, (uint8_t *)&character_declaration_uuid, ESP_GATT_PERM_READ,
[... 91 lines cut ...]
void app_main()
{
	// ---

	ret = esp_ble_gatts_app_register(ESP_BATTERY_APP_ID);
	if (ret){
		ESP_LOGE(GATTS_TABLE_TAG, "gatts app register error, error code = %x", ret);
		return;
	}

	ret = esp_ble_gatts_app_register(ESP_ESS_APP_ID);
	if (ret){
		ESP_LOGE(GATTS_TABLE_TAG, "gatts app register error, error code = %x", ret);
		return;
	}
}

An additional observation: Not only are the services shown twice, each characteristic within a service is also shown twice (please see the screenshot).

![nrf2](https://user-images.githubusercontent.com/48680239/57451954-d6f12480-7262-11e9-8085-402f5ef5e306.png)

Removing everything related to the Environmental Sensing Service, thus the Battery Sensing Service being the only service present, also removes the duplicate characteristics:
![nrf3](https://user-images.githubusercontent.com/48680239/57452003-00aa4b80-7263-11e9-9013-43aa75d0d999.png)

Thank you for your efforts!

Weijian-Espressif, 2019-05-09:
@secureschwob  I can not visit the Gist because of China's firewall, please send .c and .h files as an attachment. or send your source code to the mail   zhiweijian@espressif.com

Weijian-Espressif, 2019-05-09:
@secureschwob 
[image]
 I have modified your test code, please replace your `` hws_secure_sensor_node.c``  file and try and restart your phone Bluetooth to clean the cache.

Weijian-Espressif, 2019-05-09:
Please check your mailbox

secureschwob, 2019-05-09:
You solved my problem, thank you very much! 

My key error obviously was this:
```cpp
static void gatts_event_handler(esp_gatts_cb_event_t event, esp_gatt_if_t gatts_if,
                                esp_ble_gatts_cb_param_t *param)
{
    //printf("%s app_id %d gatts_if %d", __func__, param->reg.app_id, gatts_if);
    /* If event is register event, store the gatts_if for each profile */
    if (event == ESP_GATTS_REG_EVT) {
        if (param->reg.status == ESP_GATT_OK) {        
            //this!
            profile_tab[param->reg.app_id].gatts_if = gatts_if; 
            // rather than this:
            // profile_tab[ESS_PROFILE_APP_IDX].gatts_if = gatts_if;
            // profile_tab[BATTERY_PROFILE_APP_IDX].gatts_if = gatts_if;

        } else {
            ESP_LOGI(GATTS_TABLE_TAG, "Reg app failed, app_id %04x, status %d\n",
                    param->reg.app_id,
                    param->reg.status);
            return;
        }
    }
```
together with using different `gatts_profile_event_handler` for each of my services.

closed 2019-05-10 by Alvin1Zhang