espressif/esp-idf #3447: esp_http_client_perform never returns if the server does not send anything after connection

sysizlayan, 2019-05-09:
- Development Kit:      none
- Module or chip used:  ESP32-WROVER
- IDF version: v3.2
- Build System: Make
- Compiler version: 1.22.0-80-g6c4433a
- Operating System: Linux
- Power Supply: External 3.3V

## Problem Description
All the fields in http client is set and the esp_http_client_perform is called.
The connection is opened but the server does not respond to the device and esp_http_client_perform did not return.
### Expected Behavior
After some time, it must return even if the function failed.
### Actual Behavior
Infinite loop in the esp_http_client_perform function.
### Steps to repropduce

1. Prepare a server which does not respond to any request but accepts tcp requests.
2. Send a request using esp_http_client_perform.

Thanks in advance.

sysizlayan, 2019-05-16:
According to the BugFix section of the IDF3.3Beta3, this problem may have been solved. Infinite loop on esp_http_client_fetch_headers may be the reason behind it. The test will be done after the official release of the 3.3.

shubhamkulkarni97, 2019-11-01:
@sysizlayan I tried to send a request from ESP32 HTTP Client to a HTTP server hosted on another ESP32 which does not respond to any request. 

esp_http_client_perform returns with the following ERROR message:
HTTP_CLIENT: HTTP PUT request failed: ERROR

Can you please check once?
If it works fine, we can close this issue.

mahavirj, 2019-11-04:
@sysizlayan Closing per above comment, please feel free to reopen if you observe any issues on your side.

closed 2019-11-04 by mahavirj