espressif/esp-idf #3448: esp_tls_conn_http_new aborts on core 0

sshwarts, 2019-05-09:
with esp-idf V3.2, when I call esp_tls_conn_http_new on a server that is currently offline (browser shows a 500 error), I get an abort. This seems to take place deep inside tcpip.c in a tcpip_thread I believe after a call to getaddrinfo in esp_tls.c

I'm calling esp_tls_conn_http_new with a esp_tls_cfg_t that has been set to zero and then the cacert_pem_buf & cacert_pem_bytes fields set (this call works when the server is online).

If I turn on exceptions, is this the kind of thing that can be caught vs throwing an abort? Is there a work-around for now to check server availability prior to the call?

sshwarts, 2019-05-10:
It turns out, if the wifi DHCP does not set the DNS properly, then you get an abort. I had to set the DNS myself to get this call not to abort. This needs a better fix. It's very likely that a wifi router might not set the DNS properly.

jitin17, 2019-05-30:
@sshwarts Hi, would you mind sharing the console log with us to understand this better?

Alvin1Zhang, 2019-06-05:
@sshwarts Thanks for reporting the issue, would you please help share more log? Thanks.

sshwarts, 2019-06-05:
I'm not sure at this point if this was actually the problem. I did some more investigation. What was happening is that I was calling:

esp_wifi_connect();
And then waiting for the eventhandler to indicate connected. I was then right away calling:
esp_tls_conn_http_new(WEB_URL, &cf);

(All this on the same RTOS task)

As it turns out, I have to wait a bit before establishing the connection. I don't know if this is a router issue or wifi setup issue, but introducing a 100ms delay before calling esp_tls_conn_http_new solved the problem.

mahavirj, 2019-07-04:
@sshwarts I will go ahead and close the issue (seems to be resolved at your end).

closed 2019-07-04 by mahavirj