espressif/esp-idf #3450: airkiss入网成功后，如何让微信发现设备

kixiaomu, 2019-05-10:
通过smart_config例程

`esp_smartconfig_set_type(SC_TYPE_ESPTOUCH_AIRKISS)`

设备能成功入网。但是微信上只会一直显示:**"正在搜索可添加的物联网关"**

xuhongv, 2019-05-10:
- 你所说的是这个搜索发现设备 是近场发现功能。目前esp32还没适配的。
- 目前有 esp8266 适配了 ：https://github.com/xuhongv/xLibEsp8266Rtos3.1AirKiss

kixiaomu, 2019-05-13:
好的 谢谢

MinjunHuang, 2019-06-08:
@xuhongv 
你好， ESP32近期有计划支持近场发现功能吗？谢谢。

xuhongv, 2019-06-16:
这个要等官网的资料。我是个人开发者，不是官网的呢

MinjunHuang, 2019-06-18:
哦，好，多谢

xuhongv, 2019-10-05:
- 已经出来了啦！
 xuhongv/xLibEsp32IdfAirKiss: 【持续维护更新】微信公众号airkiss配网以及近场发现功能在esp32 esp-idf框架 sdk的实现。 https://github.com/xuhongv/xLibEsp32IdfAirKiss

@MinjunHuang @kixiaomu

YouDONG-ESP, 2020-11-03:
Hi @kixiaomu , Airkiss is available now, It can be found in the smart config example https://docs.espressif.com/projects/esp-idf/en/latest/esp32/api-reference/network/esp_smartconfig.html?highlight=smart%20config

closed 2020-11-03 by Alvin1Zhang