espressif/esp-idf #3453: link failed 链接失败

judsonzhu, 2019-05-10:
A fatal error occurred: Invalid segment count 23 (max 16). Usually this indicates a linker script problem.
/home/work/esp32/esp-idf/components/esptool_py/Makefile.projbuild:59: recipe for target '/home/work/esp32/tbmqtt/ssl/build/tbmqtt_ssl.bin' failed

Alvin1Zhang, 2019-05-16:
@judsonzhu Thanks for reporting the issue. Would you please help provide more details as suggested in the [issue template](https://github.com/espressif/esp-idf/blob/master/docs/issue_template.md)? Thanks.

keanuyaoo, 2019-05-21:
@judsonzhu  I got the same error.Have you solved it?

projectgus, 2019-06-04:
Hi @judsonzhu @keanuyaoo ,

Without more details it's not possible to provide any support. This is why the template for a new ESP-IDF issue asks a number of questions.

To start with, what ESP-IDF version are you each using? And what OS platform are you each using?

Can you give any details about the project you're building, any custom components, etc?

giroudon, 2019-06-27:
Hi.
This looks like bug #4631 on micropython github repo.
I got exactly the same issue on another project, and solve it this way :
- In esp-idf : "git submodule update --init --recursive "
- In workspace : "rm -rf build", then "make"

WayneKeenan, 2019-06-28:
I'm getting this using esp-idf v3.2.2 on macOS.

The extra step instructions from @giroudon worked for me, I had checked out using:

`git clone --depth 1 -b v3.2.2 --recursive https://github.com/espressif/esp-idf.git esp-idf-v3.2.2`

bradd-pktlzyer, 2019-07-21:
The suggestions in bug #4631 referenced by @giroudon worked for me as well. The key for me was the removal of build dir, that fixed all my issues.

projectgus, 2020-02-07:
Closing as the original reporter never followed up with more info, and it seems like the related issues were resolved with a clean build.

closed 2020-02-07 by projectgus