espressif/esp-idf #3455: mdns returns unrecognizable address

davidzuhn, 2019-05-10:
## Environment

- Development Kit:      Adafruit Huzzah-32 
- Module or chip used:  ESP32D0WDQ6
- IDF version (run ``git describe --tags`` to find it):  v4.0-dev-489-gc2aef48a5
- Build System:       Make
- Compiler version (run ``xtensa-esp32-elf-gcc --version`` to find it): gcc version 8.2.0 (crosstool-NG esp32-2019r1-rc2) 
- Operating System:    Linux 
- Power Supply:     USB

## Problem Description 

mDNS query results sometimes contain incorrect address entries.   

I've been performing queries on my network (searching for _ssh._tcp, say) and getting back result sets that when printed (using the example mdns_print_results function) showed unusual addresses.  I would see 192.168.44.1 [good] and 254.128.0.0 [bad].  I also am not seeing my advertised IPv6 addresses.   I'm using IPv6 link local addressing, and 254.128.0.0 looks suspiciously like the start of fe80::a16e:2f4f:b67d:5eb3 (which is the advertised IPv6 address).

### Expected Behavior

See  IPv4 address of 192.168.44.1 and IPv6 address of fe80::a16e:2f4f:b67d:5eb3.

### Actual Behavior

See IPv4 address of 192.168.44.1 and IPv4 address of 254.128.0.0

### Steps to repropduce

Query _ssh._tcp on my local network

### Code to reproduce this issue

Example code mdns_print_results on https://docs.espressif.com/projects/esp-idf/en/latest/api-reference/protocols/mdns.html

## Debug Logs

## Other items if possible

The problem is that mDNS and IP both have enums for the address type (mdns_ip_protocol_t and lwip_ip_addr_type).   The values for IPv4 are the same (0), but the values for IPv6 are not the same (mdns == 1, lwip == 6).   There are several places in the mDNS code that assigns an ip address type to the mdns address type.   

When code to check the type is of the form:

```if (type == IPv4) { do one thing } else { do another }```

this works just fine, but when the code is

```if (type == IPv6) { do one thing } else { do another }```

this will break down, since the IPv6 address check doesn't work (2 != 6) and so the IPv6 address data is treated as IPv4.   

I've got a short-term workaround (which I'll submit as a pull request shortly) that synchronizes the values of the two enums, so the comparison & assignments work.   I believe that the better solution would be to remove the existence of mdns_ip_protocol_t and use only the lwip_ip_addr_type enum).

davidzuhn, 2019-05-10:
See #3456 for the workaround.

david-cermak, 2019-05-11:
Hi @davidzuhn 

This was fixed in https://github.com/espressif/esp-idf/commit/fbef5297fcc587310142c06df8b561171847ede5 directly in example code.
But, as you've pointed out esp-idf documentation has not been updated accordingly.

Thank you for reporting this, will update the API reference.

davidzuhn, 2019-05-12:
Looking at the code again, I don't know what I was thinking .   This is indeed the right change to make, and when I made the corresponding change in my code things work as expected.

closed 2019-05-12 by davidzuhn