espressif/esp-idf #3457: Skip compile of specific file

Sagigamil, 2019-05-10:
I working on the last IDF release.
I have a component with source file that I want to skip on build and I didn't find any document for this option in IDF.
I cannot remove this file because it's belong to a subrepository.

Thank you for your help,

Sagi

negativekelvin, 2019-05-10:
https://docs.espressif.com/projects/esp-idf/en/latest/api-guides/build-system.html#specifying-source-files

Alvin1Zhang, 2019-05-13:
@negativekelvin Thanks for the notes. @Sagigamil Has your issue been fixed? Thanks.

suda-morris, 2019-05-13:
@Sagigamil If you're using a third-party CMake subrepository, maybe you can [refer to the mbedtls component in IDF to control the source file to build](https://github.com/espressif/esp-idf/blob/master/components/mbedtls/CMakeLists.txt#L75).

suda-morris, 2019-09-18:
We will close this issue since it's not a bug in esp-idf. If you have other question about how to use ESP-IDF, please read the [programming guide](https://docs.espressif.com/projects/esp-idf/en/latest/index.html) or raise your question in our [forum](https://www.esp32.com).

closed 2019-09-18 by suda-morris