espressif/esp-idf #3459: Build fails on esp-idf v4.0-dev-512 on Mac OSX

0cca, 2019-05-11:
## Environment

- Development Kit:    (n/a for this issue, but typically ESP32-DevKitC-v4)
- IDF version: v4.0-dev-512-g6fd535c
- Build System:         Make
- Compiler version: xtensa-esp32-elf-gcc (crosstool-NG crosstool-ng-1.22.0-80-g6c4433a) 5.2.0
- Operating System:     macOSX v10.11.6 El Capitan
- Power Supply:        (n/a for this issue, but using USB)

## Problem Description

"make flash" fails using IDF versions v4.0-dev-512-g6fd535c and esp-idf-v4.0-dev-459-gba1ff16.  Everything works fine with v4.0-dev-141-g106dc05-dirty, but after updating to esp-idf-v4.0-dev-459, and then to v4.0-dev-512, the build process fails during compilation of even simple programs like the hello_world example.

### Steps to Reproduce

1.  Moved old version of esp-idf to backup directory and downloaded latest version of esp-idf using `$ git clone --recursive https://github.com/espressif/esp-idf.git`

2.  Attempted to make a a simple example.
```
$ cd esp-idf
$ git describe --tags --dirty
v4.0-dev-512-g6fd535c-dirty

$ cd examples/get-started/hello_world/
$ make menuconfig

Python requirements from /Users/username/esp/esp-idf/requirements.txt are satisfied.
DEFCONFIG
MENUCONFIG
configuration written to /Users/username/esp/esp-idf/examples/get-started/hello_world/sdkconfig

*** End of the configuration.
*** Execute 'make' to start the build or try 'make help'.

GENCONFIG
App "hello-world" version: v4.0-dev-512-g6fd535c-dirty

$ make flash monitor
Toolchain path: /Users/username/esp/xtensa-esp32-elf/bin/xtensa-esp32-elf-gcc
Toolchain version: crosstool-ng-1.22.0-80-g6c4433a
Compiler version: 5.2.0
Python requirements from /Users/username/esp/esp-idf/requirements.txt are satisfied.
CC build/bootloader/bootloader_support/src/bootloader_clock.o
CC build/bootloader/bootloader_support/src/bootloader_common.o
CC build/bootloader/bootloader_support/src/bootloader_flash.o
CC build/bootloader/bootloader_support/src/bootloader_init.o
CC build/bootloader/bootloader_support/src/bootloader_random.o
CC build/bootloader/bootloader_support/src/bootloader_utility.o
CC build/bootloader/bootloader_support/src/esp_image_format.o
CC build/bootloader/bootloader_support/src/flash_partitions.o
CC build/bootloader/bootloader_support/src/flash_qio_mode.o
CC build/bootloader/bootloader_support/src/esp32/bootloader_sha.o
CC build/bootloader/bootloader_support/src/esp32/flash_encrypt.o
CC build/bootloader/bootloader_support/src/esp32/secure_boot.o
CC build/bootloader/bootloader_support/src/esp32/secure_boot_signatures.o
AR build/bootloader/bootloader_support/libbootloader_support.a
CC build/bootloader/log/log.o
AR build/bootloader/log/liblog.a
CC build/bootloader/spi_flash/spi_flash_rom_patch.o
AR build/bootloader/spi_flash/libspi_flash.a
CC build/bootloader/micro-ecc/micro-ecc/uECC.o
AR build/bootloader/micro-ecc/libmicro-ecc.a
CC build/bootloader/soc/esp32/cpu_util.o
make[2]: *** No rule to make target `esp32/gpio_periph.o', needed by `libsoc.a'.  Stop.
make[1]: *** [component-soc-build] Error 2
make: *** [/Users/username/esp/esp-idf/examples/get-started/hello_world/build/bootloader/bootloader.bin] Error 2
```

yene, 2019-05-12:
try `make clean`

0cca, 2019-05-13:
Thank you for your suggestion, yene.  Unfortunately, the problem still occurs after running `make clean`.  I was finally able to get through the compilation process by ignoring the errors and running `make flash` three times in succession. See below for abbreviated output.  I re-downloaded the latest esp-idf and started over from within the new examples directory, but it didn't help.

There was some discussion about needing to run `make clean` in [issue 712](https://github.com/espressif/esp-idf/issues/712) , but it's not clear to me that those solutions apply to this case.  I've updated esp-idf versions before and did not have any problems.

```
$ cd /Users/user/esp/esp-idf/examples/get-started/hello_world
$ make clean       # works fine
$ make menuconfig  # works fine

$ make flash       # fails at CC build/bootloader/soc/esp32/gpio_periph.o
Python requirements from /Users/user/esp/esp-idf/requirements.txt are satisfied.
GENCONFIG
App "hello-world" version: v4.0-dev-512-g6fd535c
...
make[2]: *** No rule to make target `esp32/gpio_periph.o', needed by `libsoc.a'.  Stop.
make[1]: *** [component-soc-build] Error 2
make: *** [/Users/user/esp/esp-idf/examples/get-started/hello_world/build/bootloader/bootloader.bin] Error 2

$ make flash       # now fails at CC build/soc/esp32/gpio_periph.o
...
make[1]: *** No rule to make target `esp32/gpio_periph.o', needed by `libsoc.a'.  Stop.
make: *** [component-soc-build] Error 2

$ make flash       # finally succeeds and successfully flashes

```

yene, 2019-05-13:
master seems broken, try latest release tag

```

cd ~/esp/esp-idf
git checkout tags/v3.1.4
git clean -df
git submodule update --recursive
```

0cca, 2019-05-14:
As noted in my original post, everything up through v4.0-dev-141-g106dc05-dirty works fine.  Just the last couple development versions are not working.  So, yes, v3.1.4 works as well.  I can develop on earlier versions of 4.0, but I wanted to note this problem now so that (hopefully) it gets fixed and I can develop using the latest version.  There are some new features that I would like to use.  Again, thank you for your suggestions and help on this issue.

bpietsch, 2019-05-16:
I am also seeing this same issue. It appears to be a missing dependency in the make files of some sort.

david-cermak, 2019-05-16:
Hi @0cca 

Thanks for reporting this. I can reproduce this issue on mac with make3.81. To resolve this you can use CMake build system or update you GNU make (4.0 works fine).

Interestingly removing the trailing `/` in https://github.com/espressif/esp-idf/blob/master/components/soc/component.mk#L3 seems to solve the issue with make 3.81 for me.

bpietsch, 2019-05-17:
Since there is an identified fix that does not involve changing the build system or make version, can that be pushed to the repo please?

0cca, 2019-05-17:
@david-cermak, thank you.  I confirmed that it works using CMake.  I'll check GNU make at a later time, but I'm up and running now.

david-cermak, 2019-05-19:
@bpietsch of course it will. Just keep this issue open so it can be tracked and closed upon merging the fix.

Have just noted CMake and new version of GNU make to encourage users to employ up to date tools where chances to fail are lower:)

closed 2019-05-20 by igrr