espressif/esp-idf #3460: Use Apache Activemq as a broker to communicate with esp32 via MQTT

GitOfLance, 2019-05-12:
### Environment
- Module or chip used: ESP32-WROOM-32
- IDF version : v4.0-dev-207-g76da10a66
- Build System: Make
- Compiler version : xtensa-esp32-elf-
- Operating System: Ubuntu 18.04
- Power Supply: USB

### Problem Description
Example:  [ssl_mutual_auth](https://github.com/espressif/esp-idf/tree/master/examples/protocols/mqtt/ssl_mutual_auth), the broker is **mosquitto**,and generate key and csr in **.pem** format using **OpenSSL**.
Finally use [Mosquitto test certificate signer](https://test.mosquitto.org/ssl/index.php) to get the client certificate and put the certificate into the program.
Now I have an **Apache ActiveMQ 5.15.9** broker server, I want my esp32 through ssl-mutual-auth for MQTT communication.
However, the information like this [CSDN blog](https://blog.csdn.net/kpp19920121/article/details/69662242) or [The Apache Tomcat](http://tomcat.apache.org/tomcat-5.5-doc/ssl-howto.html) I found on the Internet shows that the Apache Activemq server uses **keytool** to generate **.keystore** and **.truststore** files that are not **.pem** format files for ssl verification.

So, I want to ask two questions:
- Can Esp32 only use the `.pem` format certificate to complete ssl encryption? 
```
typedef struct {
    mqtt_event_callback_t event_handle;     /*!< handle for MQTT events */
    const char *host;                       /*!< MQTT server domain (ipv4 as string) */
    const char *uri;                        /*!< Complete MQTT broker URI */
    uint32_t port;                          /*!< MQTT server port */
    const char *client_id;                  /*!< default client id is ``ESP32_%CHIPID%`` where %CHIPID% are last 3 bytes of MAC address in hex format */
    const char *username;                   /*!< MQTT username */
    const char *password;                   /*!< MQTT password */
    const char *lwt_topic;                  /*!< LWT (Last Will and Testament) message topic (NULL by default) */
    const char *lwt_msg;                    /*!< LWT message (NULL by default) */
    int lwt_qos;                            /*!< LWT message qos */
    int lwt_retain;                         /*!< LWT retained message flag */
    int lwt_msg_len;                        /*!< LWT message length */
    int disable_clean_session;              /*!< mqtt clean session, default clean_session is true */
    int keepalive;                          /*!< mqtt keepalive, default is 120 seconds */
    bool disable_auto_reconnect;            /*!< this mqtt client will reconnect to server (when errors/disconnect). Set disable_auto_reconnect=true to disable */
    void *user_context;                     /*!< pass user context to this option, then can receive that context in ``event->user_context`` */
    int task_prio;                          /*!< MQTT task priority, default is 5, can be changed in ``make menuconfig`` */
    int task_stack;                         /*!< MQTT task stack size, default is 6144 bytes, can be changed in ``make menuconfig`` */
    int buffer_size;                        /*!< size of MQTT send/receive buffer, default is 1024 */
    const char *cert_pem;                   /*!< Pointer to certificate data in PEM format for server verify (with SSL), default is NULL, not required to verify the server */
    const char *client_cert_pem;            /*!< Pointer to certificate data in PEM format for SSL mutual authentication, default is NULL, not required if mutual authentication is not needed. If it is not NULL, also `client_key_pem` has to be provided. */
    const char *client_key_pem;             /*!< Pointer to private key data in PEM format for SSL mutual authentication, default is NULL, not required if mutual authentication is not needed. If it is not NULL, also `client_cert_pem` has to be provided. */
    esp_mqtt_transport_t transport;         /*!< overrides URI transport */
    int refresh_connection_after_ms;        /*!< Refresh connection after this value (in milliseconds) */
} esp_mqtt_client_config_t;
```
Can **.client_cert_pem** and **.client_key_pem**  only be pointers to certificate data in PEM format?

- Have you tried using `Apache Activemq` as a broker server? Any guidance? 
I can't find a solution. Could you do me a favour to have a look and give some advice.

chegewara, 2019-05-12:
https://stackoverflow.com/questions/652916/converting-a-java-keystore-into-pem-format

https://www.google.com/search?ei=OT3YXKSKJoKZkwXs477gAw&q=keytool+cert+format+to+pem&oq=keytool+certificate+format+to+pem&gs_l=psy-ab.1.0.33i22i29i30l2.1470.16514..24119...0.0..0.322.4060.0j24j1j1....2..0....1..gws-wiz.......0i71j0i67j0j0i19j0i22i30i19j0i22i30.am_6vjnMnNc

david-cermak, 2019-05-14:
Hi @GitOfLance 

Besides the useful links above, let me add some additional details:
* By default mbedtls supports also DER format for certificates (please see menuconfig->components->mbedtls->certificates->ReadAndParsePEM)

* Have just checked ActiveMQ with both server/client certificates, here are some notes

- configure ssl per: http://activemq.apache.org/how-do-i-use-ssl
- convert broker.ks to pem
`keytool -keystore broker.ks -exportcert -alias broker |  openssl x509 -inform der -outform pem -out  broker.pem`
- convert client ks to key format
```
keytool -importkeystore -srckeystore client.ks -destkeystore client.p12 -srcstoretype jks -deststoretype pkcs12
openssl pkcs12 -in client.p12 -out client.key
```
- convert clinet.ks to pem certificate
`keytool -keystore client.ks -exportcert -alias client |  openssl x509 -inform der -outform pem -out  client.pem`
- esp32 client config:
```
    const esp_mqtt_client_config_t mqtt_cfg = {
        .uri = "mqtts://client.address",
        .event_handle = mqtt_event_handler,
        .cert_pem = (const char *)server_cert_pem_start,
        .client_cert_pem = (const char *)client_cert_pem_start,
        .client_key_pem = (const char *)client_key_pem_start,
    };
```

GitOfLance, 2019-05-15:
@chegewara  @david-cermak 
Thank you very much for your suggestions, I will verify these methods later.
Thanks.

Alvin1Zhang, 2019-05-20:
@GitOfLance Hi, thanks for reporting the issue. Would you help share if any updates for the issue? Have these methods been tried? Thanks.

GitOfLance, 2019-05-21:
@Alvin1Zhang Hi,
I recently wrote the task code for the mqtt part. I lowered the priority of SSL encryption verification and put it in the final processing. I will post the result here.

GitOfLance, 2019-05-23:
@chegewara @david-cermak @ Alvin1Zhang 
I tried the above method, get the private key and certificate of the client and server (`.ks format`) through `keytools`, then import each other to join the trust list, 
and then convert the client's certificate, private key, and server's certificate into a `.pem` format file. These files are added to the program like this:
```
        .cert_pem = (const char *)server_cert_pem_start,
        .client_cert_pem = (const char *)client_cert_pem_start,
        .client_key_pem = (const char *)client_key_pem_start,
```
Finally, the MQTT connection is opened, but the connection fails:
[image]

So, I want to take a step back and try one-way authentication, client authentication server, so I refer to this example: [ssl](https://github.com/espressif/esp-idf/tree/master/examples/protocols/mqtt/ssl). When I use this command `openssl s_client -showcerts -connect 192.168.50.220:5500 </dev/null 2>/dev/null | openssl x509 -outform PEM >iot_eclipse_org.pem
` to generate a certificate and add it to the program to open an MQTT connection, the connection fails and the error message is the same as above.

david-cermak, 2019-05-23:
@GitOfLance the error message `CN doesn't match` indicates that the host name/address to connect to doesn't match the one in the certificate. Please regenerate your keys with correct CN (could even be your IP `192.168.50.220`)

chegewara, 2019-05-23:
I dont know how Apache Activemq is authenticating and how is using CN, but in AWS IoT for example CN in device certificate has to match AWS IoT thing name.

david-cermak, 2020-01-17:
@GitOfLance Any update? Have the test passed with the new keys matching the CN?

Alvin1Zhang, 2020-03-03:
@GitOfLance Thanks for reporting. Will close due to short of feedback, feel free to reopen with more details provided. Thanks.

closed 2020-03-03 by Alvin1Zhang