espressif/esp-idf #3462: Does modbus currently support TCP functionality?

jackyc58, 2019-05-13:
Does modbus currently support TCP functionality? Or can provide examples. Thanks

alisitsyn, 2019-05-13:
Hi jackyc58,

The current release f ESP-IDF does not support Modbus TCP. The work in progress version for Modbus TCP slave exists but is not ready for testing because related work has been postponed. Could you wait for official example in IDF release 4.1?

--
Alex

AkhileshThorat, 2019-05-13:
Hi @alisitsyn 

I assume the future versions will be supporting both modbus master/slave  for RTU and TCP. Can you please share WIP so that we can get familiar with API's and start initial development for the same

jackyc58, 2019-05-14:
Hi Alex, 

My current project is more urgent, I will try to modify the tcp server program to implement it. In addition, maybe I can also help you test tcp slave.

Regard and thanks,
Jacky.

AkhileshThorat, 2019-05-14:
> 
> 
> Hi Alex,
> 
> My current project is more urgent, I will try to modify the tcp server program to implement it. In addition, maybe I can also help you test tcp slave.
> 
> Regard and thanks,
> Jacky.

Hi @jackyc58 
Can you help me out with Modbus TCP porting on esp32?

jackyc58, 2019-05-14:
> > Hi Alex,
> > My current project is more urgent, I will try to modify the tcp server program to implement it. In addition, maybe I can also help you test tcp slave.
> > Regard and thanks,
> > Jacky.
> 
> Hi @jackyc58
> Can you help me out with Modbus TCP porting on esp32?

Hi @AkhileshThorat,

I am not familiar with the freemodbus program architecture, so I am modifying the ESP-IDF example "tcp_server". I can give it to you if you need it.

Regards,
Jacky.

AkhileshThorat, 2019-05-14:
> Hi @AkhileshThorat,
>
> I am not familiar with the freemodbus program architecture, so I am modifying the ESP-IDF example "tcp_server". I can give it to you if you need it.
>
> Regards,
> Jacky.

@jackyc58 how can you share the file? share the snippet via git or you can  mail me at akhilesh.thorat@gmail.com  :)

jackyc58, 2019-05-14:
> > > > Hi Alex,
> > > > My current project is more urgent, I will try to modify the tcp server program to implement it. In addition, maybe I can also help you test tcp slave.
> > > > Regard and thanks,
> > > > Jacky.
> > > 
> > > 
> > > Hi @jackyc58
> > > Can you help me out with Modbus TCP porting on esp32?
> > 
> > 
> > Hi @AkhileshThorat,
> > I am not familiar with the freemodbus program architecture, so I am modifying the ESP-IDF example "tcp_server". I can give it to you if you need it.
> > Regards,
> > Jacky.
> 
> @jackyc58 how can you share the file? share the snippet via git or you can mail me at [akhilesh.thorat@gmail.com](mailto:akhilesh.thorat@gmail.com) :)

[https://github.com/jackyc58/Modbus_TCP](https://github.com/jackyc58/Modbus_TCP)

AkhileshThorat, 2019-05-14:
> https://github.com/jackyc58/Modbus_TCP

Thank you @jackyc58

closed 2019-05-14 by jackyc58

jannikbecher, 2019-07-11:
is there any progress on this?

nated0g, 2020-01-20:
I'd love to see progress on this.  I can see the groundwork has been laid, and it looks good.

Is this still on the roadmap?  I can try to help but I can see that it has been planned out already.

Looks like things are left off here:
`xMBTCPPortInit( );`
I can't find a definition for this.

nated0g, 2020-01-20:
Maybe this was delayed due to update from tcpip_adapter to ESP-NETIF?

alisitsyn, 2020-01-21:
@nated0g, @jannikbecher,
Unfortunately the code is still is not ready for merge. Slave TCP code has been updated to use ESP-NETIF and supports simultaneous multi master connections. Still update and test Modbus master implementation. I will update you as soon as it is ready for review.

alisitsyn, 2020-07-24:
@nated0g, @jannikbecher,

I would like to inform you that Modbus TCP Master/Slave code has been merged and will be available at GitHub. There are example applications for master and slave with support of mDNS option to resolve slave IP addresses. The example test script can be used to check communication between master and slave over TCP (support Ethernet and WiFi phy interface).
Let me know if you have questions.

--
Alex

nated0g, 2020-07-28:
Excellent, I can't wait to try it out.  Thanks for the hard work :)

gecko242, 2020-07-31:
> I would like to inform you that Modbus TCP Master/Slave code has been merged and will be available at GitHub. There are example applications for master and slave with support of mDNS option to resolve slave IP addresses (support Ethernet and WiFi phy interface). The example test script can be used to check communication between master and slave over TCP.

@alisitsyn 

This is great news. Any idea when the code will be avaliable for testing?

alisitsyn, 2020-07-31:
@gecko242, 
I don't know exactly date when it will be merged to github. If it is not so urgent for you, I would  suggest to wait until final release.

gecko242, 2020-07-31:
@alisitsyn unfortunately we have a customer demo coming up, and having an early build of the modbus implemented would be a huge benefit. Is there any chance you are able to share a beta build at some point? Obviously I appreciate this may be buggy / undocumented, but it would be a huge help to us!

alisitsyn, 2020-08-03:
gecko242,

I understand you. I attach the complete Modbus TCP Master/Slave project to this post. It overrides the freemodbus component functionality. 
Before compile please make sure you have FMB_COMM_MODE_TCP_EN = y, in your sdkconfig, and enable MB_MDNS_IP_RESOLVER option in slave and master example to allow resolving of Modbus slave IP addresses using mDNS service. Follow the documentation for examples.
Let me know if you still have issues on how to configure the project.

[Mosbus_tcp_m_s_override.zip](https://github.com/espressif/esp-idf/files/5015148/Mosbus_tcp_m_s_override.zip)

gecko242, 2020-08-03:
@alisitsyn 

That is excellent, thankyou. I have smashed your example together with my project and got it running well. 

I should add, we only require input reads at this stage, but I will let you know if I find any issues. So far I have had luck with floats and ints too. 

Thanks again.

alisitsyn, 2020-08-03:
@gecko242,

Good to hear it works with your project. This is just example to explain how to use API, so you can use whatever you need in your project but the API supports all modbus spaces and types: unit8, uint16, floats, ascii and can be extended. 

Thank you for feedback.