espressif/esp-idf #3464: Sprintf

KonssnoK, 2019-05-13:
## Environment

- Development Kit:      [ESP32-DevKitC]
- Kit version : [v4]
- IDF version  **v4.0-dev-459-gba1ff1692**
- Build System:         [Make]
- Compiler version (crosstool-NG crosstool-ng-1.22.0-80-g6c4433a5)
- Operating System:     [Windows]
- Power Supply:         [USB]

## Problem Description

- Sprintf returning wrong value + printing wrong size.
- Testcase for sprintf is not testing anything apart from nominal case
- Sprintf is corrupting data of static variables

### Steps to repropduce

// It helps if you attach a picture of your setup/wiring here.

### Code to reproduce this issue

```cpp
// the code should be wrapped in the ```cpp tag so that it will be displayed better.
#include "esp_log.h"
#define MAX_DEVICES 20
children_t _my_network[MAX_DEVICES] = { 0 };
char _the_string[128] = { 0 }; //PUTTING THE STRING HERE IT'S OK FOR VARIABLES CONSISTENCY

typedef struct {
	uint8_t layer;
	uint16_t mac_coded;
	uint16_t parent_coded;
	bool exists;
} children_t;

void handle_periodic_task_tx()
{
		//static char _the_string[128] = { 0 }; //PUTTING THE STRING HERE WILL FUCK EVERYTHING UP
    		uint8_t idx = 0;
		//Do ancillary jobs
		if ((esp_timer_get_time() - lastDrawnUS) > 5000000U) {//Every 5 seconds
			//Draw the network
			ESP_LOGW(MESH_TAG, "Layer 0 means unknown!");
			for (int i = 0; i < CONFIG_MESH_MAX_LAYER; ++i) {

				idx = 0;

				for(int j = 0; j < MAX_DEVICES; ++j){
					//If this is the right layer and the node is active
					if(_my_network[j].exists && _my_network[j].layer == i){
						idx += sprintf(&_the_string[idx], "(%X<-%X) ", _my_network[j].parent_coded, _my_network[j].mac_coded);
					}
				}
				//Terminate string
				_the_string[idx] = 0;
				if(idx)
					ESP_LOGW(MESH_TAG, "layer %d %s %d", i, _the_string, idx);
			}
			lastDrawnUS = esp_timer_get_time();
		}
}

```
// If your code is longer than 30 lines, [GIST](https://gist.github.com) is preferred.

## Debug Logs

```
W (4986) mesh_root: <MESH_EVENT_ROUTING_TABLE_ADD>add 1, new:2
I (4986) mesh_root: <MESH_EVENT_CHILD_CONNECTED>aid:1, 30:ae:a4:d3:46:20
I (5736) mesh_root: Send to: 30:ae:a4:d3:46:20
I (5756) mesh: 1163[recv]cidx[0]30:ae:a4:d3:46:20, xseqno duplicate check failure, new xseqno:24, in:0, out:0, pending:0
I (5756) mesh: 1172[recv]cidx[0]30:ae:a4:d3:46:20 xseqno loss, current/new:0/24, in:0, out:0, pending:0
I (5766) mesh_root: RTT 1. Elapsed 16ms
I (6746) mesh_root: Send to: 30:ae:a4:d3:46:20
W (6756) mesh_root: Layer 0 means unknown!

#####################################################
######## LOOK HERE -> sprintf has changed the index to 4, when this is not possible, independently on the used structure.
W (6756) mesh_root: layer 2 (330 4
######## LOOK HERE
####################################################

I (6766) mesh_root: RTT 2. Elapsed 7ms
I (7756) mesh_root: Send to: 30:ae:a4:d3:46:20
I (7776) mesh_root: RTT 3. Elapsed 5ms
I (8766) mesh_root: Send to: 30:ae:a4:d3:46:20
I (8786) mesh_root: RTT 4. Elapsed 5ms
I (9776) mesh_root: Send to: 30:ae:a4:d3:46:20
I (9786) mesh_root: RTT 5. Elapsed 1ms
I (10786) mesh_root: Send to: 30:ae:a4:d3:46:20
I (10806) mesh_root: RTT 6. Elapsed 10ms
I (11796) mesh_root: Send to: 30:ae:a4:d3:46:20
W (11806) mesh_root: Layer 0 means unknown!
W (11806) mesh_root: layer 2 (3305<-4620) (3305<-4620) (3305<-4620)  39

```

igrr, 2019-05-13:
Is it possible that `idx` (an uint8_t) overflows? 13 characters per sprintf call times 20 (MAX_DEVICES) = 260, which is 256 + 4. This matches idx=4 in your log output. And since _the_string is only 128 bytes long, it will also be overflown, corrupting static data (if it is global) or the stack (if it is local). Recommend using snprintf and keeping track of the remaining space in the buffer.

KonssnoK, 2019-05-13:
i think you may be right, just found out a missing "break;" in the code when adding devices. 
This will fill the entire structure with the same value each time a child connects -> the draw function will try to draw devices on the same layer.
Got some ears to pull, will check better next time.

closed 2019-05-13 by KonssnoK