espressif/esp-idf #3465: Some thing wrong in python scripts

dinets, 2019-05-13:
I install Espressif toolchain for windows (without  MSYS2) and esp-idf sdk.
I successful buid hellow_world and blink examples, and flash it on ESP32.
All fine, but next step, I try config Eclipse.
I make changes in project... but I got error in python scripts....

`13:58:28 **** Build of configuration Default for project blink ****
Info: Configuration "Default" uses tool-chain "Cygwin GCC" that is unsupported on this system, attempting to build anyway.
python "C:\\Users\\dae\\esp\\esp-idf/tools/windows/eclipse_make.py" all 
Traceback (most recent call last):
  File "C:\Users\dae\esp\esp-idf/tools/windows/eclipse_make.py", line 43, in <module>
    main()
  File "C:\Users\dae\esp\esp-idf/tools/windows/eclipse_make.py", line 34, in main
    print("Running make in '%s'" % check_path(os.getcwd()))
  File "C:\Users\dae\esp\esp-idf/tools/windows/eclipse_make.py", line 23, in check_path
    winpath = subprocess.check_output(["cygpath", "-w", path]).strip()
  File "C:\Python27\lib\subprocess.py", line 212, in check_output
    process = Popen(stdout=PIPE, *popenargs, **kwargs)
  File "C:\Python27\lib\subprocess.py", line 390, in __init__
    errread, errwrite)
  File "C:\Python27\lib\subprocess.py", line 640, in _execute_child
    startupinfo)
WindowsError: [Error 2] 
"python C:\Users\dae\esp\esp-idf/tools/windows/eclipse_make.py all" terminated with exit code 1. Build might be incomplete.

13:58:28 Build Finished. 0 errors, 0 warnings. (took 365ms)`

P.S. with multi thread compile option I see additional args -j4 (eclipse_make.py all), but error same
P.P.S I use this toolchain [https://docs.espressif.com/projects/esp-idf/en/latest/get-started-cmake/windows-setup.html](url) without MSYS2

Dazza0, 2019-05-16:
Hi @dinets , by "without MSYS2" do you mean CMake?

Alvin1Zhang, 2019-05-20:
@dinets Hi, thanks for reporting the issue. Would you help share if any updates for the issue? by "without MSYS2" do you mean CMake? Thanks.

Alvin1Zhang, 2020-01-22:
@dinets Thanks for reporting. Do you have any updates on this issue? Thanks.

closed 2021-01-22 by projectgus