espressif/esp-idf #3467: C++17 build failure with  esp32-2019r1-rc2 toolchain

tgill880, 2019-05-14:
## Environment

- Development Kit: Adafruit HUZZAH32
- Module or chip used:  ESP32-WROOM-32
- IDF version:  n/a (esp32-2019r1-rc2 toolchain)
- Build System:             Make
- Compiler version: xtensa-esp32-elf-gcc (crosstool-NG esp32-2019r1-rc2) 8.2.0
- Operating System:     Linux
- Power Supply:            USB

## Problem Description
C++17 build failure when compiling C++17 header only network library that relies on `asio`.

I am attempting to port an open source, C++17 networking library, [chops-net-ip](https://github.com/connectivecpp/chops-net-ip), to the `ESP32`. The library builds on top of `asio`. I can get the `asio` sample apps to run correctly on my development board. I am using the `asio` `chat_server` project as starter code (to get the right headers for `asio`). When I add the `chops-net-ip` headers, I get a thicket of build errors, and the build fails.

C++17 is only used for the project source file.

`component.mk` (in main)
```
CXXFLAGS = -std=c++17
```

### Expected Behavior
Project builds.

### Actual Behavior
Project does not build.

### Steps to reproduce
See above

### Code to reproduce this issue

```cpp
// multiple system includes not listed
// asio project includes not listed
// chops-net-ip specific headers
#include "net_ip/net_ip.hpp"
#include "net_ip/basic_net_entity.hpp"
#include "net_ip/component/worker.hpp"
#include "net_ip/component/send_to_all.hpp"

// if I can get this code to compile, the rest of my development can continue
extern "C" void app_main()
{
    // standard startup code in asio chat_server.cpp not included
    chops::net::worker wk;
    wk.start();
    chops::net::net_ip server(wk.get_io_context());
    wk.stop();
}

```

## Debug Logs
Compiler output
```
Toolchain path: /home/thurman/esp/xtensa-esp32-elf/bin/xtensa-esp32-elf-gcc
WARNING: Toolchain version is not supported: esp32-2019r1-rc2
Expected to see version: crosstool-ng-1.22.0-80-g6c4433a
Please check ESP-IDF setup instructions and update the toolchain, or proceed at your own risk.
WARNING: Compiler version is not supported: 8.2.0
Expected to see version(s): 5.2.0
Please check ESP-IDF setup instructions and update the toolchain, or proceed at your own risk.
Python requirements from /home/thurman/esp/esp-idf/requirements.txt are satisfied.
WARNING: Toolchain version is not supported: esp32-2019r1-rc2
Expected to see version: crosstool-ng-1.22.0-80-g6c4433a
Please check ESP-IDF setup instructions and update the toolchain, or proceed at your own risk.
WARNING: Compiler version is not supported: 8.2.0
Expected to see version(s): 5.2.0
[... 814 lines cut ...]
# CONFIG_SPIFFS_CACHE_DBG is not set
# CONFIG_SPIFFS_CHECK_DBG is not set
# CONFIG_SPIFFS_TEST_VISUALISATION is not set
CONFIG_IP_LOST_TIMER_INTERVAL=120
CONFIG_TCPIP_LWIP=y
CONFIG_UNITY_ENABLE_FLOAT=y
CONFIG_UNITY_ENABLE_DOUBLE=y
# CONFIG_UNITY_ENABLE_COLOR is not set
CONFIG_UNITY_ENABLE_IDF_TEST_RUNNER=y
# CONFIG_UNITY_ENABLE_FIXTURE is not set
CONFIG_SUPPRESS_SELECT_DEBUG_OUTPUT=y
CONFIG_SUPPORT_TERMIOS=y
CONFIG_SEMIHOSTFS_MAX_MOUNT_POINTS=1
CONFIG_SEMIHOSTFS_HOST_PATH_MAX_LEN=128
# CONFIG_WL_SECTOR_SIZE_512 is not set
CONFIG_WL_SECTOR_SIZE_4096=y
CONFIG_WL_SECTOR_SIZE=4096
# CONFIG_LEGACY_INCLUDE_COMMON_HEADERS is not set

# Deprecated options for backward compatibility
CONFIG_TOOLPREFIX="xtensa-esp32-elf-"
CONFIG_PYTHON="python"
CONFIG_MAKE_WARN_UNDEFINED_VARIABLES=y
# End of deprecated options
```

igrr, 2019-05-14:
Please try CXXFLAGS += instead of =. You are removing -mlongcalls and other flags which are added by the build system to the default CXXFLAGS.

tgill880, 2019-05-14:
Yes, that fixed the problem. I also needed to enable exceptions.
```
CXXFLAGS += -std=c++17 -fexceptions
```
Many thanks!

igrr, 2019-05-14:
For enabling exceptions, please use a menuconfig option (under compiler config).

Aside from adding -fexceptions to CXXFLAGS, it enabled the required run-time support for exception handling in the early start up code. Without that, any exception thrown will lead to abort. See examples/system/cpp_exceptions.

tgill880, 2019-05-14:
Exceptions now enabled in `menuconfig`. Thanks.

I have now added all the application code to be ported to the `chops-net-ip` version of `chat_server`. A new compiler error is present.
```
In file included from /home/thurman/esp/esp-idf/components/asio/asio/asio/include/asio/ip/basic_endpoint.hpp:20,
                 from /home/thurman/esp/esp-idf/components/asio/asio/asio/include/asio/ip/tcp.hpp:24,
                 from /home/thurman/Projects/ESP32/chops_chat_server/components/chops-net-ip/include/net_ip/net_ip.hpp:30,
                 from /home/thurman/Projects/ESP32/chops_chat_server/main/chat_server.cpp:42:
/home/thurman/esp/esp-idf/components/asio/asio/asio/include/asio/ip/detail/endpoint.hpp: In member function 'bool chops::net::detail::io_common<IOT>::start_write_setup(const chops::const_shared_buffer&) [with IOT = chops::net::detail::tcp_io]':
/home/thurman/esp/esp-idf/components/asio/asio/asio/include/asio/ip/detail/endpoint.hpp:47:24: error: '<anonymous>.asio::ip::detail::endpoint::data_' may be used uninitialized in this function [-Werror=maybe-uninitialized]
     : data_(other.data_)
                        ^
cc1plus: some warnings being treated as errors
/home/thurman/esp/esp-idf/make/component_wrapper.mk:289: recipe for target 'chat_server.o' failed
make[1]: *** [chat_server.o] Error 1
/home/thurman/esp/esp-idf/make/project.mk:582: recipe for target 'component-main-build' failed
make: *** [component-main-build] Error 2
```
This error is triggered by the copy constructor of the `asio` `endpoint` helper class.
From: `/home/thurman/esp/esp-idf/components/asio/asio/asio/include/asio/ip/detail/endpoint.hpp`
```
// Copy constructor.
  endpoint(const endpoint& other)
    : data_(other.data_)
  {
  }
```
Of course, this code compiles fine with the latest g++ and clang++ on linux, macOS and Windows environments.

My various lame attempts to override the `-Werror=maybe-uninitialized` flag have failed:
`CXXFLAGS += -std=c++17 -Wno-error`
`CXXFLAGS := $(filter-out -Werrorr=maybe-uninitialized,$(CXXFLAGS))`
`CXXFLAGS := $(filter-out -Werror,$(CXXFLAGS))`
In `menuconfig`:
`[*] Disable new warnings introduced in GCC 6 - 8 `

Your prompt responses are greatly appreciated.

david-cermak, 2019-05-15:
Hi @tgill880 

Do you have to add https://github.com/connectivecpp/chops-net-ip to reproduce this build failure? It looks like simply compiling asio/using endpoints works under c++17.

Meantime, it might be possible to ignore the `maybe-uininialized` error with 
```
#pragma GCC diagnostic ignored "-Wmaybe-uninitialized"
``` in the code.

tgill880, 2019-05-15:
Yes. 

The `#pragma` suppresses the warning, and the code compiles and is uploaded to the ESP32.

Many thanks.

tgill880, 2019-05-15:
On further testing, it is clear that `chops-net-ip` and `asio` are not working well together in this enviromment. Since the build failure has been solved, I will close this ticket and open another if needed.

closed 2019-05-15 by tgill880