espressif/esp-idf #3468: ble_spp_client crash if send data on reconnection

moggiozzi, 2019-05-14:
## Environment

- Development Kit:      ESP32-DevKitV1
- Kit version (for WroverKit/PicoKit/DevKitC): v1
- Module or chip used:  ESP32-WROOM-32
- IDF version (run ``git describe --tags`` to find it): v4.0-dev-402-ga20d02b7f
- Build System:         CMake
- Compiler version (run ``xtensa-esp32-elf-gcc --version`` to find it): xtensa-esp32-elf-gcc.exe (crosstool-NG crosstool-ng-1.22.0-80-g6c4433a5) 5.2.0
- Operating System:     Windows
- Power Supply:         USB

## Problem Description
ble_spp_client crash if send data on reconnection

### Steps to repropduce
1. run ble_spp_server
2. run ble_spp_client
3. start sending data from client to server (hold down any key in terminal)
4. power off/on server module

## Debug Logs

```
rst:0x1 (POWERON_RESET),boot:0x13 (SPI_FAST_FLASH_BOOT)
configsip: 0, SPIWP:0xee
clk_drv:0x00,q_drv:0x00,d_drv:0x00,cs0_drv:0x00,hd_drv:0x00,wp_drv:0x00
mode:DIO, clock div:2
load:0x3fff0018,len:4
load:0x3fff001c,len:6356
load:0x40078000,len:11308
load:0x40080400,len:6700
entry 0x40080764
I (29) boot: ESP-IDF v4.0-dev-402-ga20d02b7f-dirty 2nd stage bootloader
I (29) boot: compile time 11:05:37
I (30) boot: Enabling RNG early entropy source...
I (35) boot: SPI Speed      : 40MHz
I (39) boot: SPI Mode       : DIO
I (43) boot: SPI Flash Size : 4MB
I (47) boot: Partition Table:
I (51) boot: ## Label            Usage          Type ST Offset   Length
I (58) boot:  0 nvs              WiFi data        01 02 00009000 00006000
I (66) boot:  1 phy_init         RF data          01 01 0000f000 00001000
I (73) boot:  2 factory          factory app      00 00 00010000 00100000
I (81) boot: End of partition table
I (85) esp_image: segment 0: paddr=0x00010020 vaddr=0x3f400020 size=0x20eec (134892) map
I (141) esp_image: segment 1: paddr=0x00030f14 vaddr=0x3ffbdb60 size=0x02b80 ( 11136) load
I (146) esp_image: segment 2: paddr=0x00033a9c vaddr=0x40080000 size=0x00400 (  1024) load
I (148) esp_image: segment 3: paddr=0x00033ea4 vaddr=0x40080400 size=0x0c16c ( 49516) load
I (177) esp_image: segment 4: paddr=0x00040018 vaddr=0x400d0018 size=0x71b08 (465672) map
I (341) esp_image: segment 5: paddr=0x000b1b28 vaddr=0x4008c56c size=0x05e44 ( 24132) load
I (362) boot: Loaded app from partition at offset 0x10000
I (362) boot: Disabling RNG early entropy source...
client start
Guru Meditation Error: Core  0 panic'ed (LoadProhibited). Exception was unhandled.
Core 0 register dump:
PC      : 0x400dc44a  PS      : 0x00060e30  A0      : 0x8008f124  A1      : 0x3ffd2f70
A2      : 0x00000000  A3      : 0x3ffba15c  A4      : 0x00060023  A5      : 0x3ffd3044
A6      : 0x00000000  A7      : 0x00000000  A8      : 0x3ffc8af4  A9      : 0x3ffd2f50
A10     : 0x00000046  A11     : 0x3ffc95ec  A12     : 0x00000000  A13     : 0x00000000
A14     : 0x3ffba348  A15     : 0x00000001  SAR     : 0x00000018  EXCCAUSE: 0x0000001c
EXCVADDR: 0x0000002a  LBEG    : 0x4000c349  LEND    : 0x4000c36b  LCOUNT  : 0xffffffff

ELF file SHA256: 0ce9f4c087c5ce3f1634266d8f42139ca2a425892cb9a62a4224d8cee1386ecf

Backtrace: 0x400dc44a:0x3ffd2f70 0x4008f121:0x3ffd2fb0

Rebooting...
```

## Other items if possible
[sdkconfig.zip](https://github.com/espressif/esp-idf/files/3176164/sdkconfig.zip)
[Pat5239.zip](https://github.com/espressif/esp-idf/files/3176173/Pat5239.zip)

moggiozzi, 2019-05-14:
The problem was in NULL pointer immediately after the connection.
Memory for db variable malloc in ESP_GATTC_CFG_MTU_EVT event:

    case ESP_GATTC_CFG_MTU_EVT:
        if(p_data->cfg_mtu.status != ESP_OK){
            break;
        }
        ESP_LOGI(GATTC_TAG,"+MTU:%d\n", p_data->cfg_mtu.mtu);
        spp_mtu_size = p_data->cfg_mtu.mtu;

        db = (esp_gattc_db_elem_t *)malloc(count*sizeof(esp_gattc_db_elem_t));`

I fix it by adding code:

					if (is_connect == true && db != NULL)
						esp_ble_gattc_write_char( spp_gattc_if,
                                              spp_conn_id,
                                              (db+SPP_IDX_SPP_DATA_RECV_VAL)->attribute_handle,
                                              event.size,
                                              temp,
                                              ESP_GATT_WRITE_TYPE_RSP,
                                              ESP_GATT_AUTH_REQ_NONE);

Weijian-Espressif, 2019-05-15:
@moggiozzi, Do you modify our spp client example?  when BLE disconection, we free db. So you can not send data when BLE disconnect

moggiozzi, 2019-05-15:
@Weijian-Espressif In ble_spp_client code no check db variable:
db != NULL

Weijian-Espressif, 2019-05-15:
we have checked  is_connect, when `is_connect ` is false, db is always NULL

moggiozzi, 2019-05-15:
@Weijian-Espressif Yes, but db still NULL right after ESP_GATTC_CONNECT_EVT(is_connect is true) until ESP_GATTC_CFG_MTU_EVT not occurs.

Weijian-Espressif, 2019-05-15:
@moggiozzi, You are right, tanks for your reminder. I will fix the bug as soon as possible.

closed 2019-05-15 by moggiozzi