espressif/esp-idf #3469: ble_spp_client example buffers uart data before connection established

moggiozzi, 2019-05-14:
## Environment
Development Kit: ESP32-DevKitV1
Kit version (for WroverKit/PicoKit/DevKitC): v1
Module or chip used: ESP32-WROOM-32
IDF version (run git describe --tags to find it): v4.0-dev-402-ga20d02b7f
Build System: CMake
Compiler version (run xtensa-esp32-elf-gcc --version to find it): xtensa-esp32-elf-gcc.exe (crosstool-NG crosstool-ng-1.22.0-80-g6c4433a5) 5.2.0
Operating System: Windows
Power Supply: USB
## Problem Description
ble_spp_client example buffers uart data before connection established

### Expected Behavior
When you press a key, previously entered data is sent.

### Actual Behavior
When you press a key sends the corresponding symbol.

### Steps to repropduce

1. run ble_spp_client
2. open and put  few symbol in terminal (press any key)
3. run ble_spp_server
4. Server terminal receive previously entered symbols(from step 2) when you press a key in client terminal.

## Other items if possible
[FixPat1678.zip](https://github.com/espressif/esp-idf/files/3176425/FixPat1678.zip)

How I can flush/reset/discard ble stack buffered data (by function esp_ble_gattc_write_char)?

Alex-Smart, 2019-05-14:
I'm having same issue (got it on server side - but behavior is the same). The only thing different is you have to send a couple of symbols from ble_spp_server while ble_spp_client is down. No issue with one symbol. 10 chars will cause it for sure.

Alex-Smart, 2019-05-15:
Here is the solution (worked well for server side)
in function
`static void gatts_profile_event_handler(esp_gatts_cb_event_t event, esp_gatt_if_t gatts_if, esp_ble_gatts_cb_param_t *param)`

at connect event handler I added uart input flush and queue reset

```
case ESP_GATTS_CONNECT_EVT:
    	{
    	    spp_conn_id = p_data->connect.conn_id;
    	    spp_gatts_if = gatts_if;
    	    is_connected = true;
    	    memcpy(&spp_remote_bda,&p_data->connect.remote_bda,sizeof(esp_bd_addr_t));
    	    // -------------------------------------------------------------------- 
    	    uart_flush_input(UART_NUM_0); //  ------------ uart input flush
    	    xQueueReset(spp_uart_queue); //   ------------ queue reset
    	    // -------------------------------------------------------------------- 
#ifdef SUPPORT_HEARTBEAT
    	    uint16_t cmd = 0;
            xQueueSend(cmd_heartbeat_queue,&cmd,10/portTICK_PERIOD_MS);
#endif
        	break;
    	}

```

Weijian-Espressif, 2019-05-15:
@moggiozzi , you can reset or clear uart buffer when BLE connect.

closed 2019-05-15 by moggiozzi