espressif/esp-idf #3470: Can't read SPIFFS, when i upload using mkspiffs & esptool.py

nopnop2002, 2019-05-14:
My Environment

	freeRTOS version:V8.2.0
	NEWLIB version:2.2.0
	lwIP version:2-0-3-255
	ESP-IDF version:v4.0-dev-512-g6fd535c98

My partitions_example.csv:
	$ cat partitions_example.csv
	# Name,   Type, SubType, Offset,  Size, Flags
	# Note: if you change the phy_init or app partition offset, make sure to change the offset in Kconfig.projbuild
	nvs,      data, nvs,     0x9000,  0x6000,
	phy_init, data, phy,     0xf000,  0x1000,
	factory,  app,  factory, 0x10000, 1M,
	storage,  data, spiffs,  ,        0xF0000,

Create File

  $ mkdir data
	$ echo "1234567890" > data/test
	$ cat data/test
	1234567890
	$ ls -l data/test
	-rw-rw-r-- 1 nop nop 11  5月 14 21:05 data/test

To pack a folder into 0xf0000(983,040) byte image:

	$ mkspiffs -c data -b 4096 -p 256 -s 0xf0000 spiffs.bin
	/test
	$ ls -l spiffs.bin
	-rw-rw-r-- 1 nop nop 983040  5月 14 21:06 spiffs.bin

To flash the image to ESP32 at offset 0x110000:

	$ python /home/nop/esp-idf/components/esptool_py/esptool/esptool.py --chip esp32 --port /dev/ttyUSB0 --baud 115200 write_flash -z 0x110000 spiffs.bin
	esptool.py v2.7-dev
	Serial port /dev/ttyUSB0
	Connecting........_
	Chip is ESP32D0WDQ6 (revision 1)
	Features: WiFi, BT, Dual Core, Coding Scheme None
	MAC: 30:ae:a4:24:bc:5c
	Uploading stub...
	Running stub...
	Stub running...
	Configuring flash size...
	Auto-detected Flash size: 4MB
	Compressed 983040 bytes to 1898...
	Wrote 983040 bytes (1898 compressed) at 0x00110000 in 0.2 seconds (effective 46262.1 kbit/s)...
	Hash of data verified.

	Leaving...
	Hard resetting via RTS pin...

Execute program:
File is exist.
Return value of fread() is not correct.
Return value of fgets() is not correct.

	I (47) boot: Partition Table:
	I (51) boot: ## Label            Usage          Type ST Offset   Length
	I (58) boot:  0 nvs              WiFi data        01 02 00009000 00006000
	I (66) boot:  1 phy_init         RF data          01 01 0000f000 00001000
	I (73) boot:  2 factory          factory app      00 00 00010000 00100000
	I (81) boot:  3 storage          Unknown data     01 82 00110000 000f0000
	I (88) boot: End of partition table

	I (0) cpu_start: App cpu up.
	I (234) heap_init: Initializing. RAM available for dynamic allocation:
	I (241) heap_init: At 3FFAE6E0 len 00001920 (6 KiB): DRAM
	I (247) heap_init: At 3FFB2ED0 len 0002D130 (180 KiB): DRAM
	I (253) heap_init: At 3FFE0440 len 00003AE0 (14 KiB): D/IRAM
	I (259) heap_init: At 3FFE4350 len 0001BCB0 (111 KiB): D/IRAM
	I (266) heap_init: At 40088ED8 len 00017128 (92 KiB): IRAM
	I (272) cpu_start: Pro cpu start user code
	I (290) cpu_start: Starting scheduler on PRO CPU.
	I (0) cpu_start: Starting scheduler on APP CPU.
	I (291) example: Initializing SPIFFS
	I (391) example: Partition size: total: 896321, used: 502
	I (391) example: fullPath=[/spiffs/test] fp=0x3ffae9e8
	I (391) example: d_name=test d_ino=0 d_type=1 fsize=11
	I (401) example: fread cb=0
	 K硤蓍晳]xample: fgets line=[┌
	I (451) example: SPIFFS unmounted

My source code:

	#include <stdio.h>
	#include <string.h>
	#include <sys/unistd.h>
	#include <sys/stat.h>
	#include "esp_err.h"
	#include "esp_log.h"
	#include "esp_system.h"
	#include "esp_vfs.h"
	#include "esp_spiffs.h"

	static const char *TAG = "example";

	static fpos_t SPIFFS_FileSize(char *fullPath) {
	    fpos_t fsize = 0;

	    FILE *fp = fopen(fullPath,"rb");
	    ESP_LOGI(TAG, "fullPath=[%s] fp=%p", fullPath, fp);
	    if (fp == NULL) return 0;
	    fseek(fp,0,SEEK_END);
	    fgetpos(fp,&fsize);
	    fclose(fp);
	    return fsize;
	}

	static void SPIFFS_Read(char *fullPath) {
	    FILE *f = fopen(fullPath,"r");
	    if (f == NULL) {
	        ESP_LOGE(TAG, "Failed to open file for reading");
	        return;
	    }
	        char buf[10];
	        int cb = fread(buf, 1, sizeof(buf), f);
	        ESP_LOGI(TAG, "fread cb=%d",cb);
	        fclose(f);
	}

	static void SPIFFS_Gets(char *fullPath) {
	    FILE *f = fopen(fullPath, "r");
	    if (f == NULL) {
	        ESP_LOGE(TAG, "Failed to open file for reading");
	        return;
	    }
	    char line[64];
	    fgets(line, sizeof(line), f);
	        ESP_LOGI(TAG, "fgets line=[%s]",line);
	    fclose(f);
	}

	static void SPIFFS_Directory(char * path) {
	    DIR* dir = opendir(path);
	        assert(dir != NULL);
	    while (true) {
	        struct dirent* pe = readdir(dir);
	        if (!pe) break;
	                char fullPath[32];
	                sprintf(fullPath, "%s/%s", path, pe->d_name);
	                int fsize = SPIFFS_FileSize(fullPath);
	        ESP_LOGI(TAG, "d_name=%s d_ino=%d d_type=%x fsize=%d", pe->d_name, pe->d_ino, pe->d_type, fsize);
	                SPIFFS_Read(fullPath);
	                SPIFFS_Gets(fullPath);
	    }
	    closedir(dir);
	}

	void app_main(void)
	{
	    ESP_LOGI(TAG, "Initializing SPIFFS");

	    esp_vfs_spiffs_conf_t conf = {
	      .base_path = "/spiffs",
	      .partition_label = NULL,
	      .max_files = 5,
	      .format_if_mount_failed = true
	    };

	    // Use settings defined above to initialize and mount SPIFFS filesystem.
	    // Note: esp_vfs_spiffs_register is an all-in-one convenience function.
	    esp_err_t ret = esp_vfs_spiffs_register(&conf);

	    if (ret != ESP_OK) {
	        if (ret == ESP_FAIL) {
	            ESP_LOGE(TAG, "Failed to mount or format filesystem");
	        } else if (ret == ESP_ERR_NOT_FOUND) {
	            ESP_LOGE(TAG, "Failed to find SPIFFS partition");
	        } else {
	            ESP_LOGE(TAG, "Failed to initialize SPIFFS (%s)", esp_err_to_name(ret));
	        }
	        return;
	    }

	    size_t total = 0, used = 0;
	    ret = esp_spiffs_info(NULL, &total, &used);
	    if (ret != ESP_OK) {
	        ESP_LOGE(TAG, "Failed to get SPIFFS partition information (%s)", esp_err_to_name(ret));
	    } else {
	        ESP_LOGI(TAG, "Partition size: total: %d, used: %d", total, used);
	    }

	    SPIFFS_Directory("/spiffs");

	    // All done, unmount partition and disable SPIFFS
	    esp_vfs_spiffs_unregister(NULL);
	    ESP_LOGI(TAG, "SPIFFS unmounted");
	}

igrr, 2019-05-14:
What is the output of mkspiffs --version?

Since you are using IDF v4.0-dev, you may try https://github.com/espressif/esp-idf/tree/master/examples/storage/spiffsgen which demonstrates SPIFFS image generation, integrated with the build system, without dependency on mkspiffs.

igrr, 2019-05-14:
Also please post the output of
`grep SPIFFS sdkconfig`

nopnop2002, 2019-05-14:
Thank you for QUICK replay.

It's mine.

	$ mkspiffs --version
	mkspiffs ver. 0.2.3-6-g983970e
	Build configuration name: generic
	SPIFFS ver. 0.3.7-5-gf5e26c4
	Extra build flags: (none)
	SPIFFS configuration:
	  SPIFFS_OBJ_NAME_LEN: 32
	  SPIFFS_OBJ_META_LEN: 0
	  SPIFFS_USE_MAGIC: 1
	  SPIFFS_USE_MAGIC_LENGTH: 1
	  SPIFFS_ALIGNED_OBJECT_INDEX_TABLES: 0

	$ grep SPIFFS sdkconfig
	CONFIG_SPIFFS_MAX_PARTITIONS=3
	CONFIG_SPIFFS_CACHE=y
	CONFIG_SPIFFS_CACHE_WR=y
	# CONFIG_SPIFFS_CACHE_STATS is not set
	CONFIG_SPIFFS_PAGE_CHECK=y
	CONFIG_SPIFFS_GC_MAX_RUNS=10
	# CONFIG_SPIFFS_GC_STATS is not set
	CONFIG_SPIFFS_PAGE_SIZE=256
	CONFIG_SPIFFS_OBJ_NAME_LEN=32
	CONFIG_SPIFFS_USE_MAGIC=y
	CONFIG_SPIFFS_USE_MAGIC_LENGTH=y
	CONFIG_SPIFFS_META_LENGTH=4
	CONFIG_SPIFFS_USE_MTIME=y
	# CONFIG_SPIFFS_DBG is not set
	# CONFIG_SPIFFS_API_DBG is not set
	# CONFIG_SPIFFS_GC_DBG is not set
	# CONFIG_SPIFFS_CACHE_DBG is not set
	# CONFIG_SPIFFS_CHECK_DBG is not set
	# CONFIG_SPIFFS_TEST_VISUALISATION is not set

nopnop2002, 2019-05-14:
I tested spiffs_create_partition_image.
It look to work correct

	I (0) cpu_start: App cpu up.
	I (234) heap_init: Initializing. RAM available for dynamic allocation:
	I (241) heap_init: At 3FFAE6E0 len 00001920 (6 KiB): DRAM
	I (247) heap_init: At 3FFB2ED0 len 0002D130 (180 KiB): DRAM
	I (253) heap_init: At 3FFE0440 len 00003AE0 (14 KiB): D/IRAM
	I (259) heap_init: At 3FFE4350 len 0001BCB0 (111 KiB): D/IRAM
	I (266) heap_init: At 40088ED8 len 00017128 (92 KiB): IRAM
	I (272) cpu_start: Pro cpu start user code
	I (290) cpu_start: Starting scheduler on PRO CPU.
	I (0) cpu_start: Starting scheduler on APP CPU.
	I (291) example: Initializing SPIFFS
	I (391) example: Partition size: total: 896321, used: 502
	I (391) example: fullPath=[/spiffs/test] fp=0x3ffae9e8
	I (391) example: d_name=test d_ino=0 d_type=1 fsize=11
	I (401) example: fread cb=10
	I (401) example: fgets line=[1234567890
	]
	I (451) example: SPIFFS unmounte

igrr, 2019-05-14:
Compile time configuration of your mkspiffs copy doesn't match the compile time configuration of your application:

SPIFFS_OBJ_META_LEN: 0
vs
CONFIG_SPIFFS_META_LENGTH=4

More information, please see https://github.com/igrr/mkspiffs/blob/master/README.md#spiffs-configuration

Or use spiffs_create_partition_image which is easier.

nopnop2002, 2019-05-14:
I rebuild mkspiffs, and I tested mkspiffs & esptool.py again.
It's work fine.
But i use spiffs_create_partition_image which is easier.

	I (0) cpu_start: App cpu up.
	I (234) heap_init: Initializing. RAM available for dynamic allocation:
	I (241) heap_init: At 3FFAE6E0 len 00001920 (6 KiB): DRAM
	I (247) heap_init: At 3FFB2ED0 len 0002D130 (180 KiB): DRAM
	I (253) heap_init: At 3FFE0440 len 00003AE0 (14 KiB): D/IRAM
	I (259) heap_init: At 3FFE4350 len 0001BCB0 (111 KiB): D/IRAM
	I (266) heap_init: At 40088ED8 len 00017128 (92 KiB): IRAM
	I (272) cpu_start: Pro cpu start user code
	I (290) cpu_start: Starting scheduler on PRO CPU.
	I (0) cpu_start: Starting scheduler on APP CPU.
	I (291) example: Initializing SPIFFS
	I (391) example: Partition size: total: 896321, used: 502
	I (391) example: fullPath=[/spiffs/test] fp=0x3ffae9e8
	I (391) example: d_name=test d_ino=0 d_type=1 fsize=11
	I (401) example: fread cb=10
	I (401) example: fgets line=[1234567890
	]
	I (451) example: SPIFFS unmounted

Tnank you.
Plase close this issues.

closed 2019-05-14 by Alvin1Zhang