espressif/esp-idf #3474: WiFi hangs when trying to connect.

arossoinsite, 2019-05-14:
## Environment

- Development Kit: none
- Module or chip used: ESP32-WROOM-32U
- IDF version: v3.1.3-2-g0d825c6e7
- Build System: Make
- Compiler version: 1.22.0-80-g6c4433a
- Operating System: Linux
- Power Supply: external 5V

## Problem Description

After reset or connection lost, WiFi is unable to connect and hangs in an unknown state or falls into a connect/disconnect loop.

### Expected Behavior
On system reset or disconnect event, and after calling `esp_wifi_connect()`, the module should always connect to the AP.
### Actual Behavior
Sometimes, on system reset or disconnect event, and after calling `esp_wifi_connect()`, the module does not connect to the AP and eithers hangs or falls into a connect/disconnect loop.

## Debug Logs
### Logs when it hangs
```
rst:0xc (SW_CPU_RESET),boot:0x13 (SPI_FAST_FLASH_BOOT)
configsip: 0, SPIWP:0xee
clk_drv:0x00,q_drv:0x00,d_drv:0x00,cs0_drv:0x00,hd_drv:0x00,wp_drv:0x00
mode:DIO, clock div:1
load:0x3fff0018,len:4
load:0x3fff001c,len:6156
load:0x40078000,len:8588
ho 0 tail 12 room 4
load:0x40080000,len:6544
entry 0x40080330
I (31) boot: ESP-IDF v3.1.3-2-g0d825c6e7 2nd stage bootloader
I (31) boot: compile time 11:23:13
I (31) boot: Enabling RNG early entropy source...
I (36) qio_mode: Enabling default flash chip QIO
I (42) boot: SPI Speed      : 80MHz
I (46) boot: SPI Mode       : QIO
I (50) boot: SPI Flash Size : 4MB
I (54) boot: Partition Table:
I (58) boot: ## Label            Usage          Type ST Offset   Length
I (65) boot:  0 nvs              WiFi data        01 02 00009000 00010000
I (72) boot:  1 otadata          OTA data         01 00 00019000 00002000
I (80) boot:  2 phy_init         RF data          01 01 0001b000 00001000
I (87) boot:  3 ota_0            OTA app          00 10 00020000 00150000
I (95) boot:  4 ota_1            OTA app          00 11 00170000 00150000
I (102) boot:  5 certs            unknown          fe 00 002c0000 00001000
I (110) boot:  6 apinfo           unknown          fe 00 002c1000 00001000
I (117) boot: End of partition table
I (122) esp_image: segment 0: paddr=0x00170020 vaddr=0x3f400020 size=0x3285c (206940) map
I (185) esp_image: segment 1: paddr=0x001a2884 vaddr=0x3ffb0000 size=0x02f98 ( 12184) load
I (188) esp_image: segment 2: paddr=0x001a5824 vaddr=0x40080000 size=0x00400 (  1024) load
I (192) esp_image: segment 3: paddr=0x001a5c2c vaddr=0x40080400 size=0x0a3e4 ( 41956) load
I (213) esp_image: segment 4: paddr=0x001b0018 vaddr=0x400d0018 size=0x996d8 (628440) map
I (377) esp_image: segment 5: paddr=0x002496f8 vaddr=0x4008a7e4 size=0x06a58 ( 27224) load
I (396) boot: Loaded app from partition at offset 0x170000
I (396) boot: Disabling RNG early entropy source...
I (396) cpu_start: Pro cpu up.
I (400) cpu_start: Starting app cpu, entry point is 0x400812b8
I (391) cpu_start: App cpu up.
I (411) heap_init: Initializing. RAM available for dynamic allocation:
I (417) heap_init: At 3FFAE6E0 len 00001920 (6 KiB): DRAM
I (423) heap_init: At 3FFC16A8 len 0001E958 (122 KiB): DRAM
I (430) heap_init: At 3FFE0440 len 00003BC0 (14 KiB): D/IRAM
I (436) heap_init: At 3FFE4350 len 0001BCB0 (111 KiB): D/IRAM
I (442) heap_init: At 4009123C len 0000EDC4 (59 KiB): IRAM
I (449) cpu_start: Pro cpu start user code
I (18) cpu_start: Starting scheduler on PRO CPU.
I (0) cpu_start: Starting scheduler on APP CPU.
I (58) system_api: Base MAC address is not set, read default base MAC address from BLK0 of EFUSE
[... 299 lines cut ...]
I (270) wifi: Init static rx buffer num: 10
I (280) wifi: Init dynamic rx buffer num: 32
I (280) wifi: Set ps type: 0

I (290) Intelligence: Loading certs ok.
I (290) Intelligence: Intelligence config ok.
I (490) phy: phy_version: 4008, c9ae59f, Jan 25 2019, 16:54:06, 0, 2
I (490) wifi: mode : sta (24:0a:c4:17:1b:a8)
I (500) system_api: Base MAC address is not set, read default base MAC address from BLK0 of EFUSE
I (500) Wi-Fi: wifi_set_static_ip() ok.
I (510) Wi-Fi: wifi_set_dns_server() ok.
I (510) Wi-Fi: wifi_set_dns_server() ok.
I (520) Wi-Fi: wifi_set_dns_server() ok.
I (520) system_api: Base MAC address is not set, read default base MAC address from BLK0 of EFUSE
I (760) wifi: n:2 0, o:1 0, ap:255 255, sta:2 0, prof:1
I (1330) wifi: state: init -> auth (b0)
I (1330) wifi: state: auth -> assoc (0)
I (1340) wifi: state: assoc -> run (10)
I (1360) wifi: connected with insite2.4, channel 2
I (1360) wifi: pm start, type: 0

I (1360) Network Manager: Waiting IP...
I (1360) event: sta ip: 192.168.1.218, mask: 255.255.255.0, gw: 192.168.1.254
I (1360) Network Manager: Got IP!
```

negativekelvin, 2019-05-15:
It looks like it is trying to connect on different channels 1,2,4. What is your AP setup?

arossoinsite, 2019-05-15:
Well, the place where the ESP's are is a little big, so, there are various AP's configured on different channels, that probably explains why it tries to connect on different channels.

DietmarHoch, 2019-05-16:
same problem with one AP, see the log. nothing happens after last line...

I (0) cpu_start: App cpu up.
I (384) heap_init: Initializing. RAM available for dynamic allocation:
I (391) heap_init: At 3FFAE6E0 len 00001920 (6 KiB): DRAM
I (397) heap_init: At 3FFB8EA0 len 00027160 (156 KiB): DRAM
I (403) heap_init: At 3FFE0440 len 00003AE0 (14 KiB): D/IRAM
I (409) heap_init: At 3FFE4350 len 0001BCB0 (111 KiB): D/IRAM
I (416) heap_init: At 4008FCC4 len 0001033C (64 KiB): IRAM
I (422) cpu_start: Pro cpu start user code
I (104) cpu_start: Starting scheduler on PRO CPU.
I (0) cpu_start: Starting scheduler on APP CPU.
I (172) wifi: wifi driver task: 3ffc080c, prio:23, stack:3584, core=0
I (172) wifi: wifi firmware version: 7c00966
I (172) wifi: config NVS flash: enabled
I (182) wifi: config nano formating: disabled
I (182) system_api: Base MAC address is not set, read default base MAC address from BLK0 of EFUSE
I (192) system_api: Base MAC address is not set, read default base MAC address from BLK0 of EFUSE
I (232) wifi: Init dynamic tx buffer num: 32
I (232) wifi: Init data frame dynamic rx buffer num: 32
I (232) wifi: Init management frame dynamic rx buffer num: 32
I (242) wifi: Init management short buffer num: 32
I (242) wifi: Init static rx buffer size: 1600
I (242) wifi: Init static rx buffer num: 10
I (252) wifi: Init dynamic rx buffer num: 32
I (362) phy: phy_version: 4100, 2a5dd04, Jan 23 2019, 21:00:07, 0, 0
I (362) wifi: mode : sta (30:ae:a4:fe:08:d8)
I (362) wifi station: wifi_init_sta finished.
I (372) wifi station: connect to ap SSID:xxxxxxxxxxx password:xxxxxxxxxxx
I (492) wifi: new:<6,0>, old:<1,0>, ap:<255,255>, sta:<6,0>, prof:1
I (492) wifi: state: init -> auth (b0)
I (492) wifi: state: auth -> init (8a0)
I (492) wifi: new:<6,0>, old:<6,0>, ap:<255,255>, sta:<6,0>, prof:1

camiloperezv, 2019-05-17:
I have a similar issue using 3.1.3 release. The device neither get the IP from router nor send a disconnection event or any event

DietmarHoch, 2019-05-21:
any updates???
Issue is very important for my project...

negativekelvin, 2019-05-21:
> I (1098) wifi: state: assoc -> init (**9c0**)
> I (32920) wifi: state: auth -> init (**200**)
> I (492) wifi: state: auth -> init (**8a0**)

It would help to know what these codes mean @liuzfesp

liuzfesp, 2019-05-22:
Hi @zhangyanjiaoesp please help.

zhangyanjiaoesp, 2019-05-22:
@DietmarHoch Which IDF version do you use?

zhangyanjiaoesp, 2019-05-22:
> I have a similar issue using 3.1.3 release. The device neither get the IP from router nor send a disconnection event or any event

@camiloperezv Do you have mulit APs with the same ssid also?

camiloperezv, 2019-05-22:
@zhangyanjiaoesp Yes, I have at least 3 AP with the same SSID but they are in different channels because they are set to AUTO

DietmarHoch, 2019-05-22:
**@zhangyanjiaoesp** 
i am on the master branch
v4.0-dev-584-g3bf56cdd1-dirty

zhangyanjiaoesp, 2019-05-23:
@DietmarHoch Please use the wifi lib to test, I think it can resolve your peoblem. 
[GH3474_master.zip](https://github.com/espressif/esp-idf/files/3210203/GH3474_master.zip)

zhangyanjiaoesp, 2019-05-23:
@arossoinsite Please check your IDF commit ID, I can't find `v3.1.3-2-g0d825c6e7`. And I find there are two commit ID in your log: `v3.1.3-2-g0d825c6e7`, `v3.1.3-2-g796b6fff8`. Both of them can't be found.

arossoinsite, 2019-05-23:
@zhangyanjiaoesp IDF commit is v3.1.2-99-gcf5dbadf4, but we add the MQTT client as another commit, so that's why the commit I gave previously is different.

DietmarHoch, 2019-05-23:
> @DietmarHoch Please use the wifi lib to test, I think it can resolve your peoblem.
> [GH3474_master.zip](https://github.com/espressif/esp-idf/files/3210203/GH3474_master.zip)

tried out this fix, 50 connects without a problem so far, I think this fix my problem.

here are a trace with a possible reconnect after fail:

_LED init
STA start
I (535) wifi: new:<6,0>, old:<1,0>, ap:<255,255>, sta:<6,0>, prof:1
I (535) wifi: state: init -> auth (b0)
I (545) wifi: state: auth -> init (8a0)
I (545) wifi: new:<6,0>, old:<6,0>, ap:<255,255>, sta:<6,0>, prof:1
I (545) wifi: new:<6,0>, old:<6,0>, ap:<255,255>, sta:<6,0>, prof:1
I (555) wifi: state: init -> auth (b0)
I (555) wifi: state: auth -> init (8a0)
I (555) wifi: new:<6,0>, old:<6,0>, ap:<255,255>, sta:<6,0>, prof:1
STA disconnected
STA disconnected
I (2745) wifi: new:<6,0>, old:<6,0>, ap:<255,255>, sta:<6,0>, prof:1
I (2745) wifi: state: init -> auth (b0)
I (2745) wifi: state: auth -> assoc (0)
I (2755) wifi: state: assoc -> run (10)
I (2795) wifi: connected with home-sweet-home, channel 6, bssid = e0:28:6d:76:30:fc
I (2795) wifi: pm start, type: 1

I (3745) tcpip_adapter: sta ip: 192.168.178.29, mask: 255.255.255.0, gw: 192.168.178.1
WiFI ready_

zhangyanjiaoesp, 2019-05-24:
@DietmarHoch 

> could you please include this in the master branch and give me an update when done?
We will merge it into the master, and give you update later.

zhangyanjiaoesp, 2019-05-24:
> @zhangyanjiaoesp IDF commit is v3.1.2-99-gcf5dbadf4, but we add the MQTT client as a local commit, so that's why the commit I gave previously is different.

Please use the wifi lib to test, thanks. 
[GH3474_v3.1.zip](https://github.com/espressif/esp-idf/files/3214829/GH3474_v3.1.zip)

arossoinsite, 2019-05-28:
@zhangyanjiaoesp I wil try and tell you if it works fine in a couple of days.

closed 2019-05-29 by igrr

arossoinsite, 2019-05-29:
@zhangyanjiaoesp It is possible to have these bugfix for release 3.2?

zhangyanjiaoesp, 2019-05-30:
> @igrr @zhangyanjiaoesp Do you know if these is fixed on release 3.2? If not, it is possible to have these bugfix for that release?

@arossoinsite Can this fix solve your problem? This bugfix has been merged into master and release/v3.3, and it will be merged into v3.2 and v3.1 later. 
By the way, you found this problem while testing another problem(https://github.com/espressif/esp-idf/issues/3323), Will the original problem still occur with the wifi lib I provided ?

arossoinsite, 2019-06-04:
@zhangyanjiaoesp I've just released a new firmware for the products with the lib provided, I'm watching how it behaves, the problems arise about once a week, so probably at the end of this week I will have some news.

zhangyanjiaoesp, 2019-06-11:
@arossoinsite Do you have any updates?

arossoinsite, 2019-06-11:
@zhangyanjiaoesp Almost two weeks have passed without the problem occurring, so it seems this fixes my problem.

arossoinsite, 2019-06-25:
@zhangyanjiaoesp It seems the issue is gone, about 3 weeks without problems. The only board that disconnected was because it's power supply failed.

camiloperezv, 2019-06-25:
Yeah, my issue is gone too, thanks a lot.

q7777777hk, 2019-11-17:
I am experiencing same problem in 3.3
After the last line, nothing happened after the last line. No IP from router nor any disconnection event 

I (680333) wifi: new:<1,0>, old:<1,0>, ap:<255,255>, sta:<1,0>, prof:1
I (680333) wifi: state: init -> auth (b0)
I (680333) wifi: state: auth -> assoc (0)
I (680333) wifi: state: assoc -> run (10)
I (680353) wifi: connected with xxxxxxx, channel 1, HT20, bssid = 0a:8d:db:6e:f8:20
I (680483) wifi: pm start, type: 1
_[0;32mI (680493) STA: Connected_[0m

arossoinsite, 2019-11-18:
In our project we implement a kind of sentinel task, independent of Wi-Fi and system event loop tasks, that receives event notifications from the Wi-Fi event handler to reset a timer, if the timer expires, the system reboot, but if receives the got IP event, stops the timer. If a disconnect or lost IP event arrives, the timer is enabled again.

q7777777hk, 2019-11-19:
> In our project we implement a kind of sentinel task, independent of Wi-Fi and system event loop tasks, that receives event notifications from the Wi-Fi event handler to reset a timer, if the timer [...]

Problem is I got no event at all after the "STA: Connected", no got IP, no disconnect, nor lost IP

zhangyanjiaoesp, 2019-11-19:
@q7777777hk 
Thanks for reporting the issue. Would you please help provide more details as suggested in the [issue template](https://github.com/espressif/esp-idf/blob/master/docs/issue_template.md)? Information like elf, sdk configuration, backtrace, log outputs, commit ID, hardware and etc. would help us debug further. Thanks.

q7777777hk, 2019-11-19:
> @q7777777hk
> Thanks for reporting the issue. Would you please help provide more details as suggested in the [issue template](https://github.com/espressif/esp-idf/blob/master/docs/issue_template.md)? [...]

Thanks for reminding, below is the detail information

My hardware is : ESP32-WROOM-32D
SDK configuration
https://drive.google.com/file/d/12vQOLFqxpMnvkQzHrierQmvmu_v82tqy/view?usp=sharing
Log output
https://drive.google.com/file/d/1Ddr5w6kQ5V1MqmbIw6pQj9G0W--PychH/view?usp=sharing
Commit log
https://drive.google.com/file/d/1JV0FZ-H5SRCXtP2MvcWFt8SCtepdkx3B/view?usp=sharing

I am using 2 routers, firstly my ESP32 connects to "MYNET". This router is always ok.
Then I switch the wifi connection to my 2nd router "ABC_Devices"  (Meraki MR42). It can connect but it never return IP not disconnection event. Tried a number of time but same situation

(Sorry I cannot provide the elf file, as it contains too much information of my customer)

zhangyanjiaoesp, 2019-11-19:
Are `MYNET` and `ABC_Devices` the two routers ssid? Both of them are `Meraki MR42`?

zhangyanjiaoesp, 2019-11-19:
What are the encryption methods of the two routers?

negativekelvin, 2019-11-19:
@q7777777hk are you sure this isn't just dhcp problem? Have you tried disabling CONFIG_LWIP_DHCP_DOES_ARP_CHECK?

zhangyanjiaoesp, 2019-11-20:
@q7777777hk Can you provide some packet captures when this issue happens.

negativekelvin, 2019-11-20:
@q7777777hk did you try with CONFIG_LWIP_DHCP_DOES_ARP_CHECK disabled?  It is affected by other things besides reserved addresses.

q7777777hk, 2019-11-20:
@negativekelvin , I will disable this in our next release, but since this problem in found in customer environment, I need to wait for their reply to get the result

q7777777hk, 2019-11-20:
@zhangyanjiaoesp , not possible to do so, as this problem is found in customer environment. They will not help us to capture packets