espressif/esp-idf #3475: NimBLE - how to cleanly close BLE stack

bfriedkin, 2019-05-15:
Hello -

I have been using the ESP32/NimBLE preview branch and have had good success getting our BLE central and peripheral apps running with the NimBLE APIs. Everything I've tried works as expected. It is also great that the stack requires less RAM compared to the Bluedroid implementation. In my testing we get back about 29 KB RAM.

One issue we have bumped into is not knowing how to cleanly shutdown the ESP32+NimBLE stack. All the example apps loop forever with no apparent cleanup. In our case, we need the ability to shutdown BLE while the app is running to conserve energy and RAM. Then on-demand we re-initialize BLE to perform certain functions.

The APIs to terminate the ESP32/Bluedroid stack are well documented and we use them. I didn't see anything similar for the ESP32/NimBLE stack. Could you please let us know the calls required to shutdown the ESP32/NimBLE stack, such that the stack can be restarted at some point later on-demand? For reference, here are the calls we currently use to initialize the NimBLE stack for a BLE server:

```
esp_nimble_hci_and_controller_init();
nimble_port_init();

ble_hs_cfg.reset_cb = nimble_on_reset;
ble_hs_cfg.sync_cb = nimble_on_sync;
ble_hs_cfg.gatts_register_cb = nimble_on_register;

ble_svc_gatt_init();
nimble_port_freertos_init(ble_host_task);

void nimble_host_task(void *param)
{
	nimble_port_run();
}

void ble_host_task(void *param)
{
	nimble_host_task(param);
}
```

Best regards,
Brian

dhrishi, 2019-05-16:
@bfriedkin Thanks for the overall feedback. 
I understand that disabling and then enabling Bluetooth is a valid use case. There is an API to stop the host `ble_hs_stop()` in the NimBLE code, but currently there seems to be no provision to stop and restart different ports (FreeRTOS, Linux etc.) it supports. We will soon be working on this.

dhrishi, 2019-06-05:
@bfriedkin Please pull the latest `feature/nimble-preview` branch and update the NimBLE submodule too. We have added the support to cleanly close the NimBLE stack. Please refer the documentation here and try it out: https://github.com/espressif/esp-idf/blob/127caf3a04c603b8c289a67d36849c3ef9f269d7/components/nimble/esp-hci/include/esp_nimble_hci.h#L100

bfriedkin, 2019-06-05:
@dhrishi -

Thank you for the heads-up. I was able to confirm that the APIs work as expected.

In the `esp_nimble_hci.h` file, I think there's a typo here:

```
 *      ret = esp_nimble_hci_and_controller_deinit();
 *      if (ret != ESP_OK) {
            ESP_LOGE(TAG, "esp_nimble_hci_and_controller_init() failed with error: %d", ret);
 *      }
```
Inside the `ESP_LOGE`, I believe the string should be `esp_nimble_hci_and_controller_deinit()`.

Regards,
Brian

closed 2019-06-05 by bfriedkin

dhrishi, 2019-06-06:
@bfriedkin Oops. I will fix it. Thanks!

TheSnowFall, 2024-01-12:
Hi @dhrishi ,

Thanks for supporting us with amazing jobs. I am stuck in such a situation that I used the example of esp-idf **blehr** and simply modified the write & notify operation. Nimble Bluetooth initialization was as it is. There is no problem with data transmission. But the problem is I can not completely clean the BLE task which gives me lots of errors. then I dug down the problem and found that maybe I am not getting the sequences of things I need to deinitialize/stop to clean everything about Nimble.  can you help me out?
my code part of initializing ble:

```
        nimble_port_init();                        // 3 - Initialize the host stack
	ble_svc_gap_init();     // 4 - Initialize NimBLE configuration - gap service
	ble_svc_gatt_init();   // 4 - Initialize NimBLE configuration - gatt service
	ble_gatts_count_cfg(gatt_svcs); // 4 - Initialize NimBLE configuration - config gatt services
	ble_gatts_add_svcs(gatt_svcs); // 4 - Initialize NimBLE configuration - queues gatt services.
	ble_hs_cfg.sync_cb = ble_app_on_sync;      // 5 - Initialize application
	nimble_port_freertos_init(host_task);
```

I hope I get proper guidance from you (could be a proper documentation/link etc).

rahult-github, 2024-01-12:
Hi @TheSnowFall ,

Please refer to [this](https://github.com/espressif/esp-idf/blob/master/examples/bluetooth/nimble/blecent/main/main.c#L866) example code that shows to initialise and deinitialise stack .

TheSnowFall, 2024-01-12:
Hi @rahult-github ,
Thanks for answering. It is really helpful for me to have an answer so quickly. I am so delighted that someone is here to help me. Because I was feeling so helpless.
here is my code to stop nimble that I used in my code:
```
void nim_ble_stop(void){

	int ret = nimble_port_stop();
	if (ret == 0) {
	     nimble_port_deinit();

	     ret = esp_nimble_deinit();
	     if (ret != ESP_OK) {
	         ESP_LOGE(BLE_TAG, "esp_nimble_hci_and_controller_deinit() failed with error: %d", ret);
	     }
	}
}
```

Then here is my implementation of the nim_ble_stop() code :
```
if(JSON_packet_parser((char*) cb_ble_buffer)==ESP_OK){
		ESP_LOGI(BLE_TAG, "NIM BLE Stoping");

		nim_ble_stop();
		ESP_LOGI(BLE_TAG, "NIM BLE Stopped");
		vTaskDelay(20000/portTICK_PERIOD_MS);
		ESP_LOGI(BLE_TAG, "NIM BLE re_starting");
		xTaskCreate(nim_BLE_init, "BT", 2048, NULL, 10, NULL);

	}
```
The serial terminal output:
```
I (26046) BLE: NIM BLE Stoping
I (26056) NimBLE: GAP procedure initiated: stop advertising.

I (26066) NimBLE: GAP procedure initiated: terminate connection; conn_handle=1 hci_reason=19
```
seems that my log print in code ESP_LOGI(BLE_TAG, "NIM BLE re_starting"); does not execute i.e the program is stuck inside nim_ble_stop(). I need help regarding this issue. Can you please?

rahult-github, 2024-01-12:
Hi @TheSnowFall ,

   Are you invoking esp_nimble_deinit() inside your  nim_ble_stop function ?. If yes, then you should not. 

 As pointed in the code , you will need to invoke  nimble_port_stop and nimble_port_deinit for  stopping stack and  nimble_port_init and nimble_port_freertos_init to restart the stack. No other functions are needed to be invoked. 
 
 You can try the blecent example once with the flag enabled to check the working .

TheSnowFall, 2024-01-12:
Hello @rahult-github ,
Thanks for giving me a way to explore.

I mentioned in my 1st comment about my appmain() function. At the end of the app main() code I added:

	```
for(; ;){
			if(BLE_reconnect_flag == 1){
				nimble_port_stop();
				nimble_port_deinit();

				vTaskDelay(100);

				        rc = nimble_port_init();
				        if (rc != ESP_OK) {
				            ESP_LOGI(BLE_TAG, "Failed to init nimble %d ", rc);
				            break;
				        }

				        nimble_port_freertos_init(blecent_host_task);
				        BLE_reconnect_flag=0;
				break;

			}

	     vTaskDelay(20);
		}
```
I am controlling the reinitializing with BLE_reconnect_flag.
Deinitialization is ok but before reinitiating serial monitor says an error:

```

I (14147) NimBLE: GAP procedure initiated: stop advertising.

I (14157) NimBLE: GAP procedure initiated: terminate connection; conn_handle=1 hci_reason=19

I (34147) BLE: NIM BLE re_starting
I (34147) GAP: BLE GAP EVENT DISCONNECTED
E (34147) NimBLE: ble_hs_stop_terminate_timeout_cb,0 connection(s) still up 

[nimble]Nimble started
W (34157) BLE_INIT: invalid controller state
E (34157) BLE_INIT: controller init failed

E (34167) BLE: BLE PORT Initilized Failed
assertion:ev->event
line:107,function:npl_freertos_event_deinit

assert failed: npl_freertos_event_deinit npl_os_freertos.c:107 (0)
Core  0 register dump:
MEPC    : 0x40380608  RA      : 0x40388e60  SP      : 0x3fcdd8a0  GP      : 0x3fcaeed0
0x40380608: panic_abort at C:/Espressif/frameworks/esp-idf-v5.1.1/components/esp_system/panic.c:452

0x40388e60: __ubsan_include at C:/Espressif/frameworks/esp-idf-v5.1.1/components/esp_system/ubsan.c:313

TP      : 0x3fcc5f9c  T0      : 0x37363534  T1      : 0x7271706f  T2      : 0x33323130
S0/FP   : 0x00000085  S1      : 0x00000001  A0      : 0x3fcdd8dc  A1      : 0x3fcaf9d5
A2      : 0x00000001  A3      : 0x00000029  A4      : 0x00000001  A5      : 0x3fcb1000
A6      : 0x0000000c  A7      : 0x76757473  S2      : 0x00000009  S3      : 0x3fcdd9f2
S4      : 0x3fcaf9d4  S5      : 0x00000000  S6      : 0x00000000  S7      : 0x00000000
S8      : 0x00000000  S9      : 0x00000000  S10     : 0x00000000  S11     : 0x00000000
T3      : 0x6e6d6c6b  T4      : 0x6a696867  T5      : 0x66656463  T6      : 0x62613938
MSTATUS : 0x00001881  MTVEC   : 0x40380001  MCAUSE  : 0x00000007  MTVAL   : 0x00000000
0x40380001: _vector_table at ??:?

MHARTID : 0x00000000

Stack memory:
3fcdd8a0: 0x3fcdd9bc 0x0000006b 0x3c045678 0x4038d1fa 0x3fcafcd4 0x3c045678 0x3fcafcb8 0x3c045305
[... 40 lines cut ...]

0x4038b530: vPortExitCritical at C:/Espressif/frameworks/esp-idf-v5.1.1/components/freertos/FreeRTOS-Kernel/portable/riscv/port.c:339

3fcddaa0: 0x0000000a 0x3fcdd270 0x00005442 0x00000000 0x3fcddb78 0x00cddb78 0x00000000 0x3fcdda60
3fcddac0: 0x0000000a 0x00000000 0x00000000 0x00000000 0x00000000 0x3fcb2d24 0x3fcb2d8c 0x3fcb2df4
3fcddae0: 0x00000000 0x00000000 0x00000001 0x00000000 0x00000000 0x00000000 0x4203d01a 0x00000000
0x4203d01a: _cleanup_r at /builds/idf/crosstool-NG/.build/HOST-x86_64-w64-mingw32/riscv32-esp-elf/src/newlib/newlib/libc/stdio/findfp.c:229

3fcddb00: 0x00000000 0x00000000 0x00000000 0x00000000 0x00000000 0x00000000 0x00000000 0x00000000
3fcddb20: 0x00000000 0x00000000 0x00000000 0x00000000 0x00000000 0x00000000 0x00000000 0x00000000
3fcddb40: 0x00000000 0x00000000 0x00000000 0x00000000 0x00000000 0x00000000 0x00000000 0x00000000
3fcddb60: 0x00000000 0x00000000 0x00000000 0x00000000 0x00000000 0x00000000 0x00000000 0x00000000
3fcddb80: 0x00000000 0x00000000 0x00000000 0x00000000 0x00000000 0x00000000 0x00000000 0x00000000
3fcddba0: 0x00000000 0x00000000 0x00000000 0x00000000 0x00000000 0x00000000 0x00000000 0x00000000
3fcddbc0: 0x00000000 0x3f000000 0x00000101 0x3fcdcb84 0x3fcdcb84 0x00000000 0x3fcb2d24 0x3fcb2d8c
3fcddbe0: 0x3fcb2df4 0x00000000 0x00000000 0x00000001 0x00000000 0x00000000 0x00000000 0x4203d01a
0x4203d01a: _cleanup_r at /builds/idf/crosstool-NG/.build/HOST-x86_64-w64-mingw32/riscv32-esp-elf/src/newlib/newlib/libc/stdio/findfp.c:229

3fcddc00: 0x00000000 0x00000000 0x00000000 0x00000000 0x00000000 0x00000000 0x00000000 0x00000000
3fcddc20: 0x00000000 0x00000000 0x00000000 0x00000000 0x00000000 0x00000000 0x00000000 0x00000000
3fcddc40: 0x00000000 0x00000000 0x00000000 0x00000000 0x00000000 0x00000000 0x00000000 0x00000000
3fcddc60: 0x00000000 0x00000000 0x00000000 0x00000000 0x00000000 0x00000000 0x00000000 0x00000000
3fcddc80: 0x00000000 0x00000000 0x00000000 0x00000000 0x00000000 0x00000000 0x00000000 0x00000000

```

TheSnowFall, 2024-01-12:
Hi @rahult-github ,

Thanks for the help. It is working great.
I was just looping myself with the api's . Now I know how to do it. i appreciate your help.