espressif/esp-idf #3476: timer queue overflow

4infantasy, 2019-05-15:
Hello, everyone, 

        I got the following log

I (29366286) APP: Sequence = 87798
{"Device":"","Protocol":"","Sequence":"87798","Information":{}}
E (29375546) MQTT_CLIENT: Error write data or timeout, written len = 0, errno=0
D (29375546) MQTT_CLIENT: Reconnect after 10000 ms
I (29375546) APP: MQTT_EVENT_DISCONNECTED
D (29393866) esp_timer: timer queue overflow
D (29395266) esp_timer: timer queue overflow
D (29397306) esp_timer: timer queue overflow
D (29400456) esp_timer: timer queue overflow

        And the TIMER code was :

esp_timer_create_args_t timerhandle_arg = {
	.callback  = &TimerHandle,				
        .arg         = NULL,							
        .name     = "TEST_TIMER"			
};

ESP_ERROR_CHECK( esp_timer_create(&timerhandle_arg, &timer_handle) );
ESP_ERROR_CHECK( esp_timer_start_periodic(timer_handle, 800000 ) );

        How can I solve it?

Thank you!

## Environment

- Kit version (for WroverKit/PicoKit/DevKitC): [v4]
- Module or chip used:  [ESP32-WROOM-32]
- IDF version : 
    v4.0-dev-311-g70eda3d
- Build System:         [Make]
- Compiler version :
    xtensa-esp32-elf-gcc (crosstool-NG crosstool-ng-1.22.0-80-g6c4433a) 5.2.0
- Operating System:     [Linux]
- Power Supply:         [USB]

I (459) cpu_start: Application information:
I (464) cpu_start: Project name:     APP
I (469) cpu_start: App version:      v4.0-dev-311-g70eda3d-dirty
I (475) cpu_start: Compile time:     May  6 2019 16:55:08
I (482) cpu_start: ELF file SHA256:  93fd1142c4aa26e9...
I (488) cpu_start: ESP-IDF:          v4.0-dev-311-g70eda3d-dirty
I (494) cpu_start: Starting app cpu, entry point is 0x400810e8

I (218) APP: [APP] Startup..
I (228) APP: [APP] Free memory: 270132 bytes
I (228) APP: [APP] Free Stack Space: 2440
I (238) APP: [APP] IDF version: v4.0-dev-311-g70eda3d-dirty

david-cermak, 2020-01-29:
Hi @4infantasy 

The log message "timer queue overflow" indicates that the timer interrupt triggered and signalled to a timer task to handle the callbacks, but unsuccessfully as so the timer task wasn't able to receive the signal. This means that the timer task seems to be blocked for some reason. Might the callback take too much time to finish, longer that the configured period for example?

Alvin1Zhang, 2020-09-09:
Thanks for reporting, will close due to short of feedback, feel free to reopen with more details. Thanks.

closed 2020-09-09 by Alvin1Zhang