espressif/esp-idf #3478: I2S example not building

thepian, 2019-05-15:
## Environment

- Development Kit:      [none]
- Module or chip used:  [ESP32-A1S]
- IDF version (run ``git describe --tags`` to find it): 
    v4.0-dev-512-g6fd535c98
- Build System:         Make
- Compiler version:
    xtensa-esp32-elf-gcc (crosstool-NG crosstool-ng-1.22.0-80-g6c4433a) 5.2.0
- Operating System:     macOS
- Power Supply:         USB

## Problem Description

I'm trying to compile examples from the SDK.

### Expected Behavior

Built firmware

### Actual Behavior

Fails to build bootloader

### Steps to repropduce

1. switch to latest master SDK and env
2. cd examples/periphials/i2s
3. make

// It helps if you attach a picture of your setup/wiring here.

### Code to reproduce this issue

Plain example

## Debug Logs

including /Volumes/Projects/esp-idf/components/app_update/Makefile.projbuild...
including /Volumes/Projects/esp-idf/components/bootloader/Makefile.projbuild...
including /Volumes/Projects/esp-idf/components/bootloader_support/Makefile.projbuild...
including /Volumes/Projects/esp-idf/components/efuse/Makefile.projbuild...
including /Volumes/Projects/esp-idf/components/esp32/Makefile.projbuild...
including /Volumes/Projects/esp-idf/components/esptool_py/Makefile.projbuild...
including /Volumes/Projects/esp-idf/components/mbedtls/Makefile.projbuild...
including /Volumes/Projects/esp-idf/components/nghttp/Makefile.projbuild...
including /Volumes/Projects/esp-idf/components/partition_table/Makefile.projbuild...
including /Volumes/Projects/esp-idf/components/spiffs/Makefile.projbuild...
including /Volumes/Projects/esp-idf/components/ulp/Makefile.projbuild...
including /Volumes/Projects/esp-idf/components/unity/Makefile.projbuild...
python /Volumes/Projects/esp-idf/tools/check_python_dependencies.py
Python requirements from /Volumes/Projects/esp-idf/requirements.txt are satisfied.
Building partitions from /Volumes/Projects/esp-idf/components/partition_table/partitions_singleapp.csv...
python /Volumes/Projects/esp-idf/components/partition_table/gen_esp32part.py -q  --flash-size "2MB" --offset 0x8000  /Volumes/Projects/esp-idf/components/partition_table/partitions_singleapp.csv /Volumes/Projects/esp-idf/examples/peripherals/i2s/build/partitions_singleapp.bin
PROJECT_PATH= COMPONENT_DIRS= /Applications/Xcode.app/Contents/Developer/usr/bin/make -C /Volumes/Projects/esp-idf/components/bootloader/subproject V=1 BUILD_DIR_BASE=/Volumes/Projects/esp-idf/examples/peripherals/i2s/build/bootloader TEST_COMPONENTS= TESTS_ALL= EXCLUDE_COMPONENTS= /Volumes/Projects/esp-idf/examples/peripherals/i2s/build/bootloader/bootloader.bin
including /Volumes/Projects/esp-idf/components/esptool_py/Makefile.projbuild...
including /Volumes/Projects/esp-idf/components/bootloader_support/Makefile.projbuild...
including /Volumes/Projects/esp-idf/components/bootloader/subproject/main/Makefile.projbuild...
including /Volumes/Projects/esp-idf/components/efuse/Makefile.projbuild...
/Applications/Xcode.app/Contents/Developer/usr/bin/make -C /Volumes/Projects/esp-idf/examples/peripherals/i2s/build/bootloader/efuse -f /Volumes/Projects/esp-idf/make/component_wrapper.mk COMPONENT_MAKEFILE=/Volumes/Projects/esp-idf/components/efuse/component.mk COMPONENT_NAME=efuse component_project_vars.mk
Building component project variables list /Volumes/Projects/esp-idf/examples/peripherals/i2s/build/bootloader/efuse/component_project_vars.mk
/Applications/Xcode.app/Contents/Developer/usr/bin/make -C /Volumes/Projects/esp-idf/examples/peripherals/i2s/build/bootloader/main -f /Volumes/Projects/esp-idf/make/component_wrapper.mk COMPONENT_MAKEFILE=/Volumes/Projects/esp-idf/components/bootloader/subproject/main/component.mk COMPONENT_NAME=main component_project_vars.mk
Building component project variables list /Volumes/Projects/esp-idf/examples/peripherals/i2s/build/bootloader/main/component_project_vars.mk
/Applications/Xcode.app/Contents/Developer/usr/bin/make -C /Volumes/Projects/esp-idf/examples/peripherals/i2s/build/bootloader/soc -f /Volumes/Projects/esp-idf/make/component_wrapper.mk COMPONENT_MAKEFILE=/Volumes/Projects/esp-idf/components/soc/component.mk COMPONENT_NAME=soc component_project_vars.mk
Building component project variables list /Volumes/Projects/esp-idf/examples/peripherals/i2s/build/bootloader/soc/component_project_vars.mk
/Applications/Xcode.app/Contents/Developer/usr/bin/make -C /Volumes/Projects/esp-idf/examples/peripherals/i2s/build/bootloader/micro-ecc -f /Volumes/Projects/esp-idf/make/component_wrapper.mk COMPONENT_MAKEFILE=/Volumes/Projects/esp-idf/components/bootloader/subproject/components/micro-ecc/component.mk COMPONENT_NAME=micro-ecc component_project_vars.mk
Building component project variables list /Volumes/Projects/esp-idf/examples/peripherals/i2s/build/bootloader/micro-ecc/component_project_vars.mk
/Applications/Xcode.app/Contents/Developer/usr/bin/make -C /Volumes/Projects/esp-idf/examples/peripherals/i2s/build/bootloader/spi_flash -f /Volumes/Projects/esp-idf/make/component_wrapper.mk COMPONENT_MAKEFILE=/Volumes/Projects/esp-idf/components/spi_flash/component.mk COMPONENT_NAME=spi_flash component_project_vars.mk
Building component project variables list /Volumes/Projects/esp-idf/examples/peripherals/i2s/build/bootloader/spi_flash/component_project_vars.mk
/Applications/Xcode.app/Contents/Developer/usr/bin/make -C /Volumes/Projects/esp-idf/examples/peripherals/i2s/build/bootloader/log -f /Volumes/Projects/esp-idf/make/component_wrapper.mk COMPONENT_MAKEFILE=/Volumes/Projects/esp-idf/components/log/component.mk COMPONENT_NAME=log component_project_vars.mk
[... 43 lines cut ...]
rm -f liblog.a
xtensa-esp32-elf-ar cru liblog.a ./log.o   
Target 'component-log-build' responsible for '/Volumes/Projects/esp-idf/examples/peripherals/i2s/build/bootloader/log/liblog.a'
/Applications/Xcode.app/Contents/Developer/usr/bin/make -C /Volumes/Projects/esp-idf/examples/peripherals/i2s/build/bootloader/spi_flash -f /Volumes/Projects/esp-idf/make/component_wrapper.mk COMPONENT_MAKEFILE=/Volumes/Projects/esp-idf/components/spi_flash/component.mk COMPONENT_NAME=spi_flash build
xtensa-esp32-elf-gcc -std=gnu99 -Og -ggdb -Wno-frame-address -ffunction-sections -fdata-sections -fstrict-volatile-bitfields -mlongcalls -nostdlib -Wall -Werror=all -Wno-error=unused-function -Wno-error=unused-but-set-variable -Wno-error=unused-variable -Wno-error=deprecated-declarations -Wextra -Wno-unused-parameter -Wno-sign-compare -Wno-old-style-declaration -I /Volumes/Projects/esp-idf/components/esp32/include -I /Volumes/Projects/esp-idf/components/esp_rom/include -I /Volumes/Projects/esp-idf/components/esp_common/include -I /Volumes/Projects/esp-idf/components/xtensa/include -I /Volumes/Projects/esp-idf/components/xtensa/esp32/include -D BOOTLOADER_BUILD=1 -I/Volumes/Projects/esp-idf/examples/peripherals/i2s/build/bootloader/../include -DESP_PLATFORM -D IDF_VER=\"v4.0-dev-512-g6fd535c98\" -MMD -MP -DESP_PLATFORM -D IDF_VER=\"v4.0-dev-512-g6fd535c98\" -MMD -MP   -D_GNU_SOURCE -DGCC_NOT_5_2_0=0 -DMBEDTLS_CONFIG_FILE='"mbedtls/esp_config.h"' -DHAVE_CONFIG_H -DUNITY_INCLUDE_CONFIG_H  -D_GNU_SOURCE -DGCC_NOT_5_2_0=0 -I /Volumes/Projects/esp-idf/components/spi_flash/include -I /Volumes/Projects/esp-idf/components/bootloader_support/include -I /Volumes/Projects/esp-idf/components/bootloader_support/include_bootloader -I /Volumes/Projects/esp-idf/components/log/include -I /Volumes/Projects/esp-idf/components/bootloader/subproject/components/micro-ecc/micro-ecc -I /Volumes/Projects/esp-idf/components/soc/esp32/include -I /Volumes/Projects/esp-idf/components/soc/include -I /Volumes/Projects/esp-idf/components/bootloader/subproject/main/include -I /Volumes/Projects/esp-idf/components/efuse/esp32/include -I /Volumes/Projects/esp-idf/components/efuse/include -I /Volumes/Projects/esp-idf/examples/peripherals/i2s/build/bootloader/include  -I . -c /Volumes/Projects/esp-idf/components/spi_flash/spi_flash_rom_patch.c -o spi_flash_rom_patch.o
echo "/Volumes/Projects/esp-idf/components/spi_flash/./spi_flash_rom_patch.c:" >> spi_flash_rom_patch.d
rm -f libspi_flash.a
xtensa-esp32-elf-ar cru libspi_flash.a spi_flash_rom_patch.o   
Target 'component-spi_flash-build' responsible for '/Volumes/Projects/esp-idf/examples/peripherals/i2s/build/bootloader/spi_flash/libspi_flash.a'
/Applications/Xcode.app/Contents/Developer/usr/bin/make -C /Volumes/Projects/esp-idf/examples/peripherals/i2s/build/bootloader/micro-ecc -f /Volumes/Projects/esp-idf/make/component_wrapper.mk COMPONENT_MAKEFILE=/Volumes/Projects/es [... cut ...] se/include -I /Volumes/Projects/esp-idf/examples/peripherals/i2s/build/bootloader/include  -I esp32 -c /Volumes/Projects/esp-idf/components/soc/esp32/cpu_util.c -o esp32/cpu_util.o
echo "/Volumes/Projects/esp-idf/components/soc/esp32/cpu_util.c:" >> esp32/cpu_util.d
make[2]: *** No rule to make target `esp32/gpio_periph.o', needed by `libsoc.a'.  Stop.
make[1]: *** [component-soc-build] Error 2
make: *** [/Volumes/Projects/esp-idf/examples/peripherals/i2s/build/bootloader/bootloader.bin] Error 2

jcabad100, 2019-05-25:
I am working with ($ git describe --tags) v4.0-dev-562-g2b301f53e ..if i try to build anything with entering "make" as command in the project directory i get "$bash: make: command not found" ...instead i manage to build and flash the getstarted examples with commands "idf.py build" and "idf.py -p COM8 flash" ...but trying to build the I2S example i get a complete different error than you have" ...  -c C:/Users/DELL/esp/esp-idf/components/hd44780/hd4
4780.c
C:/Users/DELL/esp/esp-idf/components/hd44780/hd44780.c: In function 'write_nibble':
C:/Users/DELL/esp/esp-idf/components/hd44780/hd44780.c:48:29: error: implicit declaration of function 'ets_delay_u
s' [-Werror=implicit-function-declaration]
 #define toggle_delay() do { ets_delay_us(DELAY_TOGGLE); } while (0)
                             ^
C:/Users/DELL/esp/esp-idf/components/hd44780/hd44780.c:66:9: note: in expansion of macro 'toggle_delay'
         toggle_delay();
         ^
C:/Users/DELL/esp/esp-idf/components/hd44780/hd44780.c: At top level:
cc1.exe: warning: unrecognized command line option '-Wno-frame-address'
cc1.exe: some warnings being treated as errors
[674/826] Building C object esp-idf/freemodbus/CMakeFil...serial_master/modbus_controller/mbc_serial_master.c.obj
ninja: build stopped: subcommand failed.
ninja failed with exit code 1"

...the above error appear in more than one examples i try to build ... e.g. I2C etc...

koobest, 2019-07-19:
Hi, have you solved this issue?

Alvin1Zhang, 2020-01-22:
@thepian Thanks for reporting, will close due to short of feedback, feel free to reopen if the issue still happens. Thanks.

closed 2020-01-22 by Alvin1Zhang