espressif/esp-idf #3479: https_server multiple socket/ssl handshake errors/warnings

no1seman, 2019-05-15:
## Environment

- Development Kit: ESP32-Wrover-Kit
- Kit version (for WroverKit/PicoKit/DevKitC): v4.1
- Module or chip used:  ESP32-WROVER-B
- IDF version (run ``git describe --tags`` to find it): v4.0-dev-521-g5ffc9f2d7
- Build System:  CMake
- Compiler version (run ``xtensa-esp32-elf-gcc --version`` to find it): 1.22.0-80-g6c4433a
- Operating System:     Linux
- Power Supply: USB|external 5V & Battery

## Problem Description
With https_server example got mutiple handshake & socket warnings/errors (see log below) when using Google Chrome  74.0.3729.157 (Windows x64) & Microsoft Edge 44.17763.1.0 (Windows x64), while on Firefox 66.0.5 (Windows x64) there is no such errors except missing favicon.ico

### Expected Behavior
No any errrors with session, sockets and so on

### Actual Behavior
Mutiple warnings & errors

### Steps to repropduce

Just use vanilla https_server example and try to connect to device via https with different browsers; Chrome, Firefox, MS Edge

### Code to reproduce this issue
/esp-idf/examples/protocols/https_server/ from latest master

## Debug Logs

```
W (52582) esp_https_server: fail to SSL_accept - handshake error
W (52582) httpd: httpd_accept_conn: session creation failed
W (52592) httpd: httpd_server: error accepting new connection
E (52592) httpd: httpd_server: error in select (9)
W (52602) httpd_sess: httpd_sess_delete_invalid: Closing invalid socket 57
W (54532) esp_https_server: fail to SSL_accept - handshake error
W (54532) httpd: httpd_accept_conn: session creation failed
W (54532) httpd: httpd_server: error accepting new connection
E (54542) httpd: httpd_server: error in select (9)
W (54542) httpd_sess: httpd_sess_delete_invalid: Closing invalid socket 58
W (60362) esp_https_server: fail to SSL_accept - handshake error
W (60362) httpd: httpd_accept_conn: session creation failed
W (60372) httpd: httpd_server: error accepting new connection
W (62002) esp_https_server: fail to SSL_accept - handshake error
W (62002) httpd: httpd_accept_conn: session creation failed
W (62002) httpd: httpd_server: error accepting new connection
E (62012) httpd: httpd_server: error in select (9)
W (62012) httpd_sess: httpd_sess_delete_invalid: Closing invalid socket 59
W (62022) httpd_sess: httpd_sess_delete_invalid: Closing invalid socket 60
W (66212) esp_https_server: fail to SSL_accept - handshake error
W (66212) httpd: httpd_accept_conn: session creation failed
W (66222) httpd: httpd_server: error accepting new connection
W (67852) esp_https_server: fail to SSL_accept - handshake error
W (67852) httpd: httpd_accept_conn: session creation failed
W (67852) httpd: httpd_server: error accepting new connection
I (69952) APP: Free memory: 181432, Minimum free memory: 165892
E (69962) httpd: httpd_server: error in select (9)
W (69962) httpd_sess: httpd_sess_delete_invalid: Closing invalid socket 61
W (69962) httpd_sess: httpd_sess_delete_invalid: Closing invalid socket 62
W (70112) httpd_uri: httpd_uri: URI '/favicon.ico' not found
W (70112) httpd_txrx: httpd_resp_send_err: 404 Not Found - This URI does not exist
W (272662) esp_https_server: fail to SSL_accept - handshake error
W (272662) httpd: httpd_accept_conn: session creation failed
W (272672) httpd: httpd_server: error accepting new connection
W (274292) esp_https_server: fail to SSL_accept - handshake error
W (274302) httpd: httpd_accept_conn: session creation failed
W (274302) httpd: httpd_server: error accepting new connection
E (276462) httpd: httpd_server: error in select (9)
W (276462) httpd_sess: httpd_sess_delete_invalid: Closing invalid socket 57
W (276462) httpd_sess: httpd_sess_delete_invalid: Closing invalid socket 58
I (276482) APP: Free memory: 184684, Minimum free memory: 165708
W (277062) httpd_uri: httpd_uri: URI '/favicon.ico' not found
W (277062) httpd_txrx: httpd_resp_send_err: 404 Not Found - This URI does not exist
W (282152) esp_https_server: fail to SSL_accept - handshake error
W (282152) httpd: httpd_accept_conn: session creation failed
W (282162) httpd: httpd_server: error accepting new connection
W (283812) esp_https_server: fail to SSL_accept - handshake error
W (283812) httpd: httpd_accept_conn: session creation failed
W (283812) httpd: httpd_server: error accepting new connection
E (285952) httpd: httpd_server: error in select (9)
W (285952) httpd_sess: httpd_sess_delete_invalid: Closing invalid socket 60
W (285952) httpd_sess: httpd_sess_delete_invalid: Closing invalid socket 61
I (285972) APP: Free memory: 181096, Minimum free memory: 163940

```

## Other items if possible

anurag-kar, 2019-05-17:
@no1seman This is because we are using a self signed certificate causing a SSL handshake error, and then chrome internally sends multiple GET requests to the same URL, so each request causes an SSL handshake error.

One way to stop these warning would be to generate your own `rootCA.pem`, and import that to chrome. Then generate a new set of `prvtkey.pem` and `cacert.pem` using the `rootCA.pem`

anurag-kar, 2019-05-17:
Though looking at this:
```
E (62012) httpd: httpd_server: error in select (9)
```
I believe the socket for which SSL handshake failed, and is closed (and hence invalid), is being added to internal socket database, later causing error in select. I'll look into this.

Thanks for reporting!

no1seman, 2019-05-17:
@anurag-kar,
thank you for quick answer.

1.  Yes, I know that using non valid certificates if not good idea, but it's a way test application in all circumstances. Case of using non-valid certificate is real, because often IoT devices is used only in local networks and there are not many who use certificates signed by commercial CA for such a devices . In that case what if someone need to connect to device without installing root CA or installation of root CA is impossible due to user rights restrictions?
2. Error is changing if CONFIG_HTTPD_ERR_RESP_NO_DELAY = n
3. Also, there are some difference in server's behavour when using different browsers (for example Chrome & Firefox, see attached logs for Chrome & Firefox: 
[chrome.zip](https://github.com/espressif/esp-idf/files/3191175/chrome.zip)
[firefox.zip](https://github.com/espressif/esp-idf/files/3191176/firefox.zip)).
First handshake error in both cases occurs when browsers shows warning of invalid certificate and promts user to continue or to skip, but in case of Chrome if choose countinue there are some erorors after (see chrome.log). That erros seems to be not good for production. 
Also I think It will be useful  to realize logic (may be config flag in httpd) like 'ngx_http_limit_conn_module' to limit number of connections from one IP/client. This may solve problems with resources:
- only one socket per client (number of sockets is limited and as now 4 sockets needed to establish new connection);
- reduce memory consumption (less connections - less memory needed for SSL structures) ;
- increase performance of SSL connection (quote from official documentation: _...The initial session setup can take about two seconds, or more with slower clock speeds or more verbose logging...._ - it's slow enougth).

anurag-kar, 2019-05-19:
@no1seman We will soon merge a fix to the issue relating to the error in select on SSL handshake failure. I am attaching a patch that you may use till then : [httpd_err_sel.patch.txt](https://github.com/espressif/esp-idf/files/3194968/httpd_err_sel.patch.txt)

>  Case of using non-valid certificate is real, because often IoT devices is used only in local networks

I agree. The possible ways to deal with this, that I can think off are:
* If you are using a browser (like chrome / firefox) to access webpages on the HTTPS server on your device IoT, the straight forward way would be to simply ignore the warning messages (which adds a security exception for the IP)
* If using `curl` is sufficient for your use case, then using that with the `--insecure` option will skip certificate validation
* Use plain HTTP server along with content level security, like `protocomm_httpd` with security 1. But that will be much different from your regular client - server interaction

> CONFIG_HTTPD_ERR_RESP_NO_DELAY = n

Could you please use the above patch and try again? If it still happens, please post the log.

> there are some difference in server's behavour when using different browsers

It is a known fact, that chrome sends multiple GET requests to the same resource, probably to speed up things. This is useful when the server and client are on different networks, with different possible routes between them. But for our case this doesn't help, as they are on the same network, infact it worsens things as multiple connection requests will cause multiple SSL handshakes, slowing things down. I have looked about, but can't seem to find a way to disable this in chrome. Also, if security exception is added for the IP, it still fails with SSL handshake error on atleast one of the multiple connection requests, which is quite puzzling.

> limit number of connections from one IP/client. This may solve problems with resources

Thanks. Will consider having this config option. Will be quite useful.

> only one socket per client  (number of sockets is limited and as now 4 sockets needed to establish new connection)

Server uses only 1 socket per connection and not 4. The server internally uses up 3 sockets (1 TCP listening socket + 2 UDP control sockets) when it is started. But afterwords, for every new connection only one socket is created per connection.

> reduce memory consumption / increase performance of SSL connection

We are working in that direction, by having an alternate transport layer security implementation for the HTTPS server. Hopefully we'll have something better soon.

no1seman, 2019-05-19:
@anurag-kar, thank you for your answer, but I have to disagree with one of your statements, cause my my tests shows that server<->browser connection uses 2 sockets.

If you add a simple persistant session hander to https example, smth like that:

void http_get_remote_ip(httpd_req_t *req, user_ctx_t *user_ctx)
{
    int s = httpd_req_to_sockfd(req);
    struct sockaddr_in6 addr_in;
    socklen_t addrlen = sizeof(addr_in);
    if (lwip_getpeername(s, (struct sockaddr *)&addr_in, &addrlen) != -1)
    {
        user_ctx->ip.addr = addr_in.sin6_addr.un.u32_addr[3];
        user_ctx->port = addr_in.sin6_port;
    }
    else
    {
        user_ctx->ip.addr = 0;
        user_ctx->port = 0;
        ESP_LOGE(tag, "Error getting peer's IP/port");
    }
}

static void http_create_session(httpd_req_t *req)
{
    if (req->sess_ctx == NULL)
    {
        req->sess_ctx = malloc(sizeof(user_ctx_t));
        req->free_ctx = http_delete_session;
        user_ctx_t *ctx_data = (user_ctx_t *)req->sess_ctx;
        ctx_data->timestamp = esp_log_timestamp();
        ctx_data->id = 0;
        http_get_remote_ip(req, ctx_data);

        char *buf = NULL;
        if (http_get_hdr_value(req, "Accept-Encoding", &buf) == ESP_OK)
        {
            if (strstr(buf, "gzip"))
            {
                ctx_data->accept_zip = true;
            }
            else
            {
                ctx_data->accept_zip = false;
            }
        }
        else
        {
            ctx_data->accept_zip = false;
        }
        safe_free(buf);

        if (http_get_hdr_value(req, "Cookie", &buf) == ESP_OK)
        {
            char *token = NULL;
            token = buf;

            while ((*token != '\0') && (*token != '='))
            {
                token++;
            }
            if ((*token != '\0') && (*token == '=')) token++;
            if (*token == '+')
            {
                while ((*token != '\0') && (*token == '+'))
                {
                    token++;
                }
            }
            printf("Token from Cookie: '%s'\n", token);
            
            ctx_data->token = pvPortMalloc(strlen(token) + 1);
            strcpy(ctx_data->token, token);
        }
        else
        {
            ctx_data->token = NULL;
        }
        safe_free(buf);

        ctx_data->username = NULL;
        ctx_data->password = NULL;

        ESP_LOGI(tag, "allocating new session for remote: %s:%d", inet_ntoa(ctx_data->ip), ntohs(ctx_data->port));
    }
}

you will see that in any case: https/https - server uses **2** sockets, here is log for my app:
I (78444) HTTP: allocating new session for remote: 192.168.3.155:53043
I (83834) HTTP: allocating new session for remote: 192.168.3.155:53044
and also you can see wireshark log too:
[image]
*.155 - it's my windows host from where i made request from Chrome browser to esp32 by https and *.222 it's esp32 device. 
In that case I used Non-valid self-signet certificate.
With http - server also uses **2** sockets.
If it so, I don't understand what 'httpd_config_t->max_open_sockets' limits? If I set it to 1 nothing works at all, beacuse 2nd socket goes to retrasmission cycle:
[image]

anurag-kar, 2019-05-19:
@no1seman It's using up two sockets because chrome chooses to create 2 connections. So again 1 socket per connection (and not 2). As per the wireshark capture, it can be seen that chrome sends 2 'Client Hello`s, each with different socket numbers. So it is asking for 2 TLS connections. Notice the `Key Exchange` is also happening twice, once for each TLS connection. I'm not sure why it does so, but I believe that it sends one GET request for the URI, and opens other parallel connections for future use, to speed things up.

You won't see the same behavior with firefox or curl. You can try logging those with wireshark. Even if you try with some other server (like apache running on your localhost), you are likely to see the same behavior with chrome. So I would stick with my statement, that **only 1 socket is used per TLS connection**. So, if the browser opens 2 TLS connections at the same time, then the server would use up 2 sockets.

> I don't understand what 'httpd_config_t->max_open_sockets' limits? If I set it to 1 nothing works at all, beacuse 2nd socket goes to retrasmission cycle (on Chrome) / And on Firefox everything works fine.

I'm not sure why the TCP re-transmission is happening for you. I tried with `max_open_sockets' set to 1 and It seems to work fine for me on chrome. When` max_open_sockets' is set to 1 only one socket is allowed at a time. So if chrome tries to create two parallel connections, it will close one of them and keep the other open. If by chance it closes the connection for the GET request, the page won't load on your browser.

anurag-kar, 2019-05-19:
@no1seman I was able to reproduce the TCP re-transmissions, but for that I had to set `httpd.lru_purge_enable = false`, apart from `httpd.max_open_sockets = 1`.

Could you confirm if you are using the same configuration?

no1seman, 2019-05-20:
@anurag-kar,

1. After applying patch - select(9) error is gone
2. In any case http/https, Chrome/Firefox, valid/Invalid certificate/key 2 sockets used (httpd.lru_purge_enable = false, apart from httpd.max_open_sockets = 1 - this combination of settings proves that it so, cause my static content of 5 different resources (html, ico, js, css, json) is not downloading) 
Note: on nginx with limit_conn addr 1; also used 2 sockets! ))))
3. Handshake errors have no place to be if I set MAX CPU frequency greater than 80 Mhz if https used. So the only question left - the performance of establishing SSL connection, it would be better if the perfomance will be greater and SSL connection latency - lower. Honestly it's not clear why different SSL libs used in different components in esp-idf: mbedtls, openssl, libsoidum. Seems that not all of them uses hardware crypto capabilities of CPU?

anurag-kar, 2019-05-20:
@no1seman Thanks for the info

> After applying patch - select(9) error is gone

Glad that it worked. An internal merge request is in progress for this fix.

> httpd.lru_purge_enable = false, apart from httpd.max_open_sockets = 1 / this combination of settings proves that it so, cause my static content of 5 different resources (html, ico, js, css, json) is not downloading

The content is not loading and TCP re-transmissions are happening because of `httpd.lru_purge_enable = false`. Setting this to true should work.

If you set it to false, browser will keep a persistent socket open, and since `httpd.max_open_sockets = 1`, connection requests will fail for any further transactions.

closed 2019-06-05 by igrr