espressif/esp-idf #3480: MQTT Memory leak

shamtastic, 2019-05-15:
## Environment

- Development Kit:      [ESP32-Sparkfun thing]
- ESP-IDF V3.2
- Build System:         [Make]
- Compiler version 5.2.0
- Operating System:     [Windows]
- Power Supply:         [USB]

## Problem Description
Ive been noticing memory leak issue with the built-in MQTT library. Im stress testing it with 500bytes and 1KB payload sizes. after a number of messages (around 200 for 1k and 400 for 500bytes) heap memory is almost completely consumed and the device crashes

### Expected Behavior

MQTT_EXAMPLE: Free memory after recieving: 22340 bytes
MQTT_EXAMPLE: Free memory after publishing: 22340 bytes

### Actual Behavior
I (306062) MQTT_EXAMPLE: Free memory after recieving: 22340 bytes
I (306072) MQTT_EXAMPLE: Free memory after publishing: 17804 bytes
I (306872) MQTT_EXAMPLE: Free memory after recieving: 21768 bytes
I (306892) MQTT_EXAMPLE: Free memory after publishing: 17232 bytes
I (307492) MQTT_EXAMPLE: Free memory after recieving: 21196 bytes

### Steps to repropduce

- vary the payload size to (1000+), and period to 500 to expedite the process
- run the attached code
- Publish a message to ESP32/Start
- whenever a message is received to ESP32/data then publish to ESP32/ack

### Code to reproduce this issue

https://github.com/shamtastic/ESP32_MQTT

## Debug Logs

```
I (37192) MQTT_EXAMPLE: MQTT_EVENT_DATA
TOPIC=ESP32/start
DATA={"frequency":500,"payload":5000,"qos":0}
I (37192) MQTT_EXAMPLE: Free memory after recieving: 223876 bytes
I (37192) MQTT_EXAMPLE: Start_Code, msg_id=0
I (37202) MQTT_EXAMPLE: init data is 500 5000 0 SSSSSSSSSSSSSSSSSSSSSSSSSSSSSSSSSSSSSSSSSSSSSSSSSSSSSSSSSSSSSSSSSSSSSS
SSSSSSSSSSSSSSSSSSSSSSSSSSSSSSSSSSSSSSSSSSSSSSSSSSSSSSSSSSSSSSSSSSSSSSSSSSSSSSSSSSSSSSSSSSSSSSSSSSSSSSSSSSSSSSSSSSSSSS
SSSSSSSSSSSSSSSSSSSSSSSSSSSSSSSSSSSSSSSSSSSSSSSSSSSSSSSSSSSSSSSSSSSSSSSSSSSSSSSSSSSSSSSSSSSSSSSSSSSSSSSSSSSSSSSSSSSSSS
SSSSSSSSSSSSSSSSSSSSSSSSSSSSSSSSSSSSSSSSSSSSSSSSSSSSSSSSSSSSSSSSSSSSSSSSSSSSSSSSSSSSSSSSSSSSSSSSSSSSSSSSSSSSSSSSSSSSSS
SSSSSSSSSSSSSSSSSSSSSSSSSSSSSSSSSSSSSSSSSSSSSSSSSSSSSSSSSSSSSSSSSSSSSSSSSSSSSSSSSSSSSSSSSSSSSSSSSSSSSSSSSSSSSSSSSSSSSS
SSSSSSSSSSSSSSSSSSSSSSSSSSSSSSSSSSSSSSSSSSSSSSSSSSSSSSSSSSSSSSSSSSSSSSSSSSSSSSSSSSSSSSSSSSSSSSSSSSSSSSSSSSSSSSSSSSSSSS
SSSSSSSSSSSSSSSSSSSSSSSSSSSSSSSSSSSSSSSSSSSSSSSSSSSSSSSSSSSSSSSSSSSSSSSSSSSSSSSSSSSSSSSSSSSSSSSSSSSSSSSSSSSSSSSSSSSSSS
SSSSSSSSSSSSSSSSSSSSSSSSSSSSSSSSSSSSSSSSSSSSSSSSSSSSSSSSSSSSSSSSSSSSSSSSSSSSSSSSSSSSSSSSSSSSSSSSSSSSSSSSSSSSSSSSSSSSSS
SSSSSSSSSSSSSSSSSSSSSSSSSSSSSSSSSSSSSSSSSSSSSSSSSSSSSSSSSSSSSSSSSSSSSSSSSSSSSSSSSSSSSSSSSSSSSSSSSSSSSSSSSSSSSSSSSSSSSS
SSSSSSSSSSSSSSSSSSSSSSSSSSSSSSSSSSSSSSSSSSSSSSSSSSSSSSSSSSSSSSSSSSSSSSSSSSSSSSSSSSSSSSSSSSSSSSSSSSSSSSSSSSSSSSSSSSSSSS
SSSSSSSSSSSSSSSSSSSSSSSSSSSSSSSSSSSSSSSSSSSSSSSSSSSSSSSSSSSSSSSSSSSSSSSSSSSSSSSSSSSSSSSSSSSSSSSSSSSSSSSSSSSSSSSSSSSSSS
SSSSSSSSSSSSSSSSSSSSSSSSSSSSSSSSSSSSSSSSSSSSSSSSSSSSSSSSSSSSSSSSSSSSSSSSSSSSSSSSSSSSSSSSSSSSSSSSSSSSSSSSSSSSSSSSSSSSSS
SSSSSSSSSSSSSSSSSSSSSSSSSSSSSSSSSSSSSSSSSSSSSSSSSSSSSSSSSSSSSSSSSSSSSSSSSSSSSSSSSSSSSSSSSSSSSSSSSSSSSSSSSSSSSSSSSSSSSS
SSSSSSSSSSSSSSSSSSSSSSSSSSSSSSSSSSSSSSSSSSSSSSSSSSSSSSSSSSSSSSSSSSSSSSSSSSSSSSSSSSSSSSSSSSSSSSSSSSSSSSSSSSSSSSSSSSSSSS
SSSSSSSSSSSSSSSSSSSSSSSSSSSSSSSSSSSSSSSSSSSSSSSSSSSSSSSSSSSSSSSSSSSSSSSSSSSSSSSSSSSSSSSSSSSSSSSSSSSSSSSSSSSSSSSSSSSSSS
SSSSSSSSSSSSSSSSSSSSSSSSSSSSSSSSSSSSSSSSSSSSSSSSSSSSSSSSSSSSSSSSSSSSSSSSSSSSSSSSSSSSSSSSSSSSSSSSSSSSSSSSSSSSSSSSSSSSSS
SSSSSSSSSSSSSSSSSSSSSSSSSSSSSSSSSSSSSSSSSSSSSSSSSSSSSSSSSSSSSSSSSSSSSSSSSSSSSSSSSSSSSSSSSSSSSSSSSSSSSSSSSSSSSSSSSSSSSS
SSSSSSSSSSSSSSSSSSSSSSSSSSSSSSSSSSSSSSSSSSSSSSSSSSSSSSSSSSSSSSSSSSSSSSSSSSSSSSSSSSSSSSSSSSSSSSSSSSSSSSSSSSSSSSSSSSSSSS
SSSSSSSSSSSSSSSSSSSSSSSSSSSSSSSSSSSSSSSSSSSSSSSSSSSSSSSSSSSSSSSSSSSSSSSSSSSSSSSSSSSSSSSSSSSSSSSSSSSSSSSSSSSSSSSSSSSSSS
SSSSSSSSSSSSSSSSSSSSSSSSSSSSSSSSSSSSSSSSSSSSSSSSSSSSSSSSSSSSSSSSSSSSSSSSSSSSSSSSSSSSSSSSSSSSSSSSSSSSSSSSSSSSSSSSSSSSSS
SSSSSSSSSSSSSSSSSSSSSSSSSSSSSSSSSSSSSSSSSSSSSSSSSSSSSSSSSSSSSSSSSSSSSSSSSSSSSSSSSSSSSSSSSSSSSSSSSSSSSSSSSSSSSSSSSSSSSS
SSSSSSSSSSSSSSSSSSSSSSSSSSSSSSSSSSSSSSSSSSSSSSSSSSSSSSSSSSSSSSSSSSSSSSSSSSSSSSSSSSSSSSSSSSSSSSSSSSSSSSSSSSSSSSSSSSSSSS
SSSSSSSSSSSSSSSSSSSSSSSSSSSSSSSSSSSSSSSSSSSSSSSSSSSSSSSSSSSSSSSSSSSSSSSSSSSSSSSSSSSSSSSSSSSSSSSSSSSSSSSSSSSSSSSSSSSSSS
SSSSSSSSSSSSSSSSSSSSSSSSSSSSSSSSSSSSSSSSSSSSSSSSSSSSSSSSSSSSSSSSSSSSSSSSSSSSSSSSSSSSSSSSSSSSSSSSSSSSSSSSSSSSSSSSSSSSSS
SSSSSSSSSSSSSSSSSSSSSSSSSSSSSSSSSSSSSSSSSSSSSSSSSSSSSSSSSSSSSSSSSSSSSSSSSSSSSSSSSSSSSSSSSSSSSSSSSSSSSSSSSSSSSSSSSSSSSS
SSSSSSSSSSSSSSSSSSSSSSSSSSSSSSSSSSSSSSSSSSSSSSSSSSSSSSSSSSSSSSSSSSSSSSSSSSSSSSSSSSSSSSSSSSSSSSSSSSSSSSSSSSSSSSSSSSSSSS
SSSSSSSSSSSSSSSSSSSSSSSSSSSSSSSSSSSSSSSSSSSSSSSSSSSSSSSSSSSSSSSSSSSSSSSSSSSSSSSSSSSSSSSSSSSSSSSSSSSSSSSSSSSSSSSSSSSSSS
[... 373 lines cut ...]

0x400d398a: deliver_publish at C:/Users/Shams/Desktop/esp32_win32_msys2_environment_and_toolchain-20181001/msys32/home
/Shams/esp/esp-idf/components/mqtt/esp-mqtt/mqtt_client.c:951

0x400d3b49: mqtt_process_receive at C:/Users/Shams/Desktop/esp32_win32_msys2_environment_and_toolchain-20181001/msys32
/home/Shams/esp/esp-idf/components/mqtt/esp-mqtt/mqtt_client.c:951

0x400d3d08: esp_mqtt_task at C:/Users/Shams/Desktop/esp32_win32_msys2_environment_and_toolchain-20181001/msys32/home/S
hams/esp/esp-idf/components/mqtt/esp-mqtt/mqtt_client.c:951

0x4008ce69: vPortTaskWrapper at C:/Users/Shams/Desktop/esp32_win32_msys2_environment_and_toolchain-20181001/msys32/hom
e/Shams/esp/esp-idf/components/freertos/port.c:403

Rebooting...
ets Jun  8 2016 00:22:57

rst:0xc (SW_CPU_RESET),boot:0x13 (SPI_FAST_FLASH_BOOT)
configsip: 0, SPIWP:0xee
clk_drv:0x00,q_drv:0x00,d_drv:0x00,cs0_drv:0x00,hd_drv:0x00,wp_drv:0x00
mode:DIO, clock div:2
load:0x3fff0018,len:4
load:0x3fff001c,len:6476
load:0x40078000,len:10180
load:0x40080400,len:6640
entry 0x40080760

negativekelvin, 2019-05-15:
Memory leaks and cjson go together

shamtastic, 2019-05-15:
> Memory leaks and cjson go together 👫

Do you recommend another JSON library i should go with instead of cJSON?

negativekelvin, 2019-05-15:
No you just have to track all your cjson objects carefully and delete/free them

shamtastic, 2019-05-15:
Just figured it out. The leak is cJSON_Print() that im using. Apparently it allocates memory whenever called and that needs to be deallocated.

closed 2019-05-15 by shamtastic

law-ko, 2022-10-27:
@shamtastic , there still seemed to be some leaks in `cJSON_Print()` even after freeing the char pointer. Do you have this issue?
I have created an [issue](https://github.com/DaveGamble/cJSON/issues/715) at cJSON as well.