espressif/esp-idf #3482: No rule to make phy_init_data

KaeLL, 2019-05-15:
Anyone else gets this error from time to time as well?
`make: *** No rule to make target '~/Documents/IDF/esp-idf/components/esp32/phy_init_data.h', needed by '~Documents/VSCode/esp32_learning/build/phy_init_data.o'.  Stop.`

I was getting this when I first tried to build the app this morning. I replaced my esp-idf with the latest snapshot of the master branch right after, and I'm still getting this.

[sdkconfig](https://gist.github.com/KaeLL/f1075d651aa8601a8f16a49586a85a0c)

igrr, 2019-05-15:
What is the IDF_PATH set to? (`echo $IDF_PATH`). If it starts with `~`, try using `$HOME` when setting it, instead: `export IDF_PATH=$HOME/Documents/IDF/esp-idf`.

KaeLL, 2019-05-15:
nvmd@nvmd-W35xSS-370SS:~$ printenv IDF_PATH
/home/nvmd/Documents/IDF/esp-idf
nvmd@nvmd-W35xSS-370SS:~$ cat .profile | grep IDF_PATH
export IDF_PATH="$HOME/Documents/IDF/esp-idf"

KaeLL, 2019-05-15:
So I found the problem.
It appears to be an issue with the `CONFIG_ESP32_PHY_INIT_DATA_IN_PARTITION` option in the sdkconfig.
To reproduce it, just enable it, and compile your app against the master branch of esp-idf. And yes, I do have an phy partition in my partition table, which is why I think this might be a real issue.

Here's the partitions file
```
# Name
nvs,data,nvs,0x9000,0x4000
otadata,data,ota,0xd000,0x2000
phy_init,data,phy,0xf000,0x1000
ota_0,app,ota_0,0x10000,1600K
ota_1,app,ota_1,,1600K
```

KaeLL, 2019-05-29:
Any update on this issue?

suda-morris, 2019-06-12:
@KaeLL Thank you, we're fixing this bug, will push to master ASAP.

adam-esp, 2019-06-15:
What should I do to make the compilation successful? When will you fix it? A lot of confusion has arisen with these changes.

KaeLL, 2019-06-16:
> What should I do to make the compilation successful? When will you fix it? A lot of confusion has arisen with these changes.

[Disable it](https://docs.espressif.com/projects/esp-idf/en/latest/api-reference/kconfig.html#config-esp32-phy-init-data-in-partition)

adam-esp, 2019-06-16:
Thanks, now he's compiling. What consequences have the ability to disable this setting?

closed 2019-06-26 by igrr