espressif/esp-idf #3483: NIMBLE: assertion "evbuf != NULL" failed: file ".../esp_nimble_hci.c", line 356, function: host_rcv_pkt

no1seman, 2019-05-15:
## Environment

- Development Kit:      ESP32-Wrover-Kit
- Kit version (for WroverKit/PicoKit/DevKitC): v4.1
- Module or chip used:  ESP32-WROVER-B
- IDF version (run ``git describe --tags`` to find it):  v3.3-beta1-452-gfefb2fda2 (latest feature/nimble-preview branch)
- Build System:  CMake
- Compiler version (run ``xtensa-esp32-elf-gcc --version`` to find it): 1.22.0-80-g6c4433a
- Operating System:     Linux
- Power Supply:         USB|external 5V & Battery

## Problem Description
Making tests with NIMBLE (logging all scanned data in BLE_GAP_EVENT_DISC event handler) and a little modyfied nimble/blecent example hangs with:
assertion "evbuf != NULL" failed: file "/home/user/esp/esp-mdf/esp-idf/components/nimble/esp-hci/src/esp_nimble_hci.c", line 356, function: host_rcv_pkt
abort() was called at PC 0x400d40fb on core 0
0x400d40fb: __assert_func at /Users/ivan/e/newlib_xtensa-2.2.0-bin/newlib_xtensa-2.2.0/xtensa-esp32-elf/newlib/libc/stdlib/../../../.././newlib/libc/stdlib/assert.c:63 (discriminator 8)
Backtrace: 0x4008c8f8:0x3ffb5d00 0x4008cb29:0x3ffb5d20 0x400d40fb:0x3ffb5d40 0x400ef869:0x3ffb5d70 0x40109fd2:0x3ffb5d90 0x401001b6:0x3ffb5db0 0x4001791d:0x3ffb5dd0 0x40100799:0x3ffb5df0 0x40100925:0x3ffb5e30 0x40017573:0x3ffb5e50 0x40019d11:0x3ffb5e70 0x40055b4d:0x3ffb5e90 0x400ff3e3:0x3ffb5eb0 0x400ff8e1:0x3ffb5ed0 0x4008cd65:0x3ffb5f00
0x4008c8f8: invoke_abort at /home/user/esp/esp-mdf/esp-idf/components/esp32/panic.c:707
0x4008cb29: abort at /home/user/esp/esp-mdf/esp-idf/components/esp32/panic.c:707
0x400d40fb: __assert_func at /Users/ivan/e/newlib_xtensa-2.2.0-bin/newlib_xtensa-2.2.0/xtensa-esp32-elf/newlib/libc/stdlib/../../../.././newlib/libc/stdlib/assert.c:63 (discriminator 8)
0x400ef869: host_rcv_pkt at /home/user/esp/esp-mdf/esp-idf/components/nimble/esp-hci/src/esp_nimble_hci.c:225
0x40109fd2: vhci_send at ??:?
0x401001b6: r_eif_send at ??:?
0x40100799: hci_tx_start at hci_tl.c:?
0x40100925: hci_tx_done at hci_tl.c:?
0x400ff3e3: r_rw_schedule at ??:?
0x400ff8e1: btdm_controller_task at ??:?
0x4008cd65: vPortTaskWrapper at /home/user/esp/esp-mdf/esp-idf/components/freertos/port.c:403

### Expected Behavior
No assertions
### Actual Behavior
Assert happenning
### Steps to repropduce

https://github.com/no1seman/nimble_0_test

### Code to reproduce this issue

https://github.com/no1seman/nimble_0_test

## Debug Logs
See attached file for full log
## Other items if possible
[log.zip](https://github.com/espressif/esp-idf/files/3183854/log.zip)

dhrishi, 2019-05-16:
@no1seman Looks like you have enabled duplicate advertising reports in the blecent code. I have fixed the handling which caused this issue seen from your logs. Please pull the latest `feature/nimble-preview` branch and test it out.

no1seman, 2019-05-16:
@dhrishi,
thanks for quick answer, error in gone, but now test app (https://github.com/no1seman/nimble_0_test) just stops to scan after some time (with 2 beacons after 36 minutes, and with 8 beacons to scan - after 18 minutes). Seems that there's an infinite cycle somewhere or deadlock. Modified app here:  Here is the end of the log:

1 st log:
I (2181179) NimBLE_BLE_CENT: MAC: b8:27:eb:c0:7c:c3, Discoverable: Discoverable, Mode: Only LE, Data: 
02 01 06 03 03 aa fe 17 16 aa fe 00 ec f7 82 6d a6 bc 5b 71 e0 89 3e 67 74 37 42 66 63 00 00 
I (2181199) NimBLE_BLE_CENT: MAC: b8:27:eb:c0:7c:c3, Name: 0000124mfzd
I (2181199) NimBLE_BLE_CENT: MAC: 6f:65:9d:ef:ab:83, Discoverable: Discoverable, Mode: LE & BR/EDR, Data: 02 01 1a 0a ff 4c 00 10 05 03 18 79 34 09 
I (2181369) NimBLE_BLE_CENT: MAC: 00:15:a3:00:0b:74, Discoverable: Discoverable, Mode: LE & BR/EDR, Data: 
02 01 02 03 03 e0 ff 09 ff 0e 05 74 0b 00 a3 15 00 
I (2181379) NimBLE_BLE_CENT: MAC: 00:15:a3:00:0b:74, Name: UM25C
I (2181389) NimBLE_BLE_CENT: MAC: 00:15:a3:00:0b:74, Discoverable: Discoverable, Mode: LE & BR/EDR, Data:  02 01 02 03 03 e0 ff 09 ff 0e 05 74 0b 00 a3 15 00 
I (2181399) NimBLE_BLE_CENT: MAC: 00:15:a3:00:0b:74, Name: UM25C
I (2181409) NimBLE_BLE_CENT: MAC: 00:15:a3:00:0b:74, Discoverable: Discoverable, Mode: LE & BR/EDR, Data:  02 01 02 03 03 e0 ff 09 ff 0e 05 74 0b 00 a3 15 00 
I (2181419) NimBLE_BLE_CENT: MAC: 00:15:a3:00:0b:74, Name: UM25C
<----- no any output after, but also theres no any warnings, errors, asserts or crash.

2nd log:
I (1019411) NimBLE_BLE_CENT: MAC: 80:7d:3a:b9:96:7a, Discoverable: Discoverable, Mode: Only LE, Data: 
02 01 06 1a ff 4c 00 02 15 f7 82 6d a6 4f a2 4e 98 80 24 bc 5b 71 e0 89 3e 5f 9c cb 92 c9 
I (1019431) NimBLE_BLE_CENT: MAC: 80:7d:3a:b9:96:7a, Name: 1e2szwhyt4

I (1019431) NimBLE_BLE_CENT: MAC: 80:7d:3a:b9:96:7a, Discoverable: Discoverable, Mode: Only LE, Data: 
02 01 06 1a ff 4c 00 02 15 f7 82 6d a6 4f a2 4e 98 80 24 bc 5b 71 e0 89 3e 5f 9c cb 92 c9 
I (1019451) NimBLE_BLE_CENT: MAC: 80:7d:3a:b9:96:7a, Name: 1e2szwhyt4

I (1019461) NimBLE_BLE_CENT: MAC: 80:7d:3a:b9:97:96, Discoverable: Discoverable, Mode: Only LE, Data: 
02 01 06 1a ff 4c 00 02 15 f7 82 6d a6 4f a2 4e 98 80 24 bc 5b 71 e0 89 3e 5f 9c cb 92 c9 
I (1019481) NimBLE_BLE_CENT: MAC: 80:7d:3a:b9:97:96, Name: 1e2szwhz10

I (1019481) NimBLE_BLE_CENT: MAC: 00:15:a3:00:0b:74, Discoverable: Discoverable, Mode: LE & BR/EDR, Data: 
02 01 02 03 03 e0 ff 09 ff 0e 05 74 0b 00 a3 15 00 
I (1021031) NimBLE_BLE_CENT: Internal memory: Free: 284300 bytes, Allocated: 46580 bytes, Mininum free: 278720 bytes, Largest free block: 113804 bytes
I (1041031) NimBLE_BLE_CENT: Internal memory: Free: 284300 bytes, Allocated: 46580 bytes, Mininum free: 278720 bytes, Largest free block: 113804 bytes
I (1061031) NimBLE_BLE_CENT: Internal memory: Free: 284300 bytes, Allocated: 46580 bytes, Mininum free: 278720 bytes, Largest free block: 113804 bytes
I (1081031) NimBLE_BLE_CENT: Internal memory: Free: 284300 bytes, Allocated: 46580 bytes, Mininum free: 278720 bytes, Largest free block: 113804 bytes
<----- no any scans output after, but also theres no any warnings, errors, asserts or crash.

P.S. You are right, i've set no filter & no whitelist & enable duplicates to make a load test of NIMBLE itself cause such a tests shows overall stability of stack in general. I'm very disapointed about stability of BlueDroid stack and looking for the ability to move to NIMBLE.

dhrishi, 2019-05-16:
@no1seman I am trying your app and it is up and running for an hour now. Is there anything else in the setup that is required? 

Also, can you please raise a new issue with the requested information and close this. Please see the log below:

```
I (3620536) NimBLE_BLE_CENT: MAC: xx:xx:xx:xx:xx:xx, Discoverable: Discoverable, Mode: LE & BR/EDR, Data: 
02 01 02 0d 09 48 6f 6e 6f 72 20 39 20 4c 69 74 65 
I (3620536) NimBLE_BLE_CENT: MAC: yy:yy:yy:yy:yy:yy, Discoverable: Discoverable, Mode: LE & BR/EDR, Data: 
02 01 02 06 09 50 69 78 65 6c
I (3620576) NimBLE_BLE_CENT: Internal memory: Free: 284304 bytes, Allocated: 46328 bytes, Mininum free: 278740 bytes, Largest free block: 113804 bytes
```

dhrishi, 2019-05-16:
@no1seman  I can understand that it is not very convenient but I cannot comment on the exact timeline of merging. Currently, we are in the process of QA for NimBLE.

Please close this issue

closed 2019-05-16 by no1seman

ksiahpar, 2023-10-06:
i had this problem, i solve it by removing (commented by semi column) this library (NimBLE) from platformio.ini: ;h2zero/NimBLE-Arduino@^1.4.1 then move the library folder from .\myProject\.pio\libdeps\esp32dev to .\myProject\lib. then edite the function static int host_rcv_pkt(uint8_t *data, uint16_t len) in file: .\myProject\lib\NimBLE-Arduino\src\nimble\esp_port\esp-hci\src\esp_nimble_hci.c in line 407 as below:

        if (data[1] == BLE_HCI_EVCODE_HW_ERROR)
        {
            // assert(0);
            return 0; // add it by  kianoosh
        }