espressif/esp-idf #3484: Suggestion: makefile target to print toolchain version

MitchBradley, 2019-05-15:
My project uses esp-idf and I have Makefile scripting to automatically install esp-idf from github and to download and install the xtensa toolchain.  At present, my Makefile hardcodes the toolchain version and when esp-idf changes to expect a different toolchain, I get warnings and have to manually update to the new toolchain.  It would be nice if there were an easy way to ask esp-idf which version it expects, so I could automate the process of installing the right one.  Something like "make print-toolchain-version".

It would be even better if esp-idf were to automatically install the toolchain, but I understand how fiddly that can be in the face of different build environments, so the print target would be an acceptable intermediate step.

negativekelvin, 2019-05-16:
https://github.com/espressif/esp-idf/blob/ae5df5d6ce7c9abd9c8c4731ced96031c29f7fc1/tools/toolchain_versions.mk

MitchBradley, 2019-05-17:
Thanks.  I had missed that file because I had an old copy of the tree and forgot to look at the new makefiles after I updated it.  By including that file in my Makefile the toolchain installation is now automated.

MitchBradley, 2019-05-17:
Uh, that file does not appear in the v3.1.4 release archive file.  To save time, I switched from cloning the repo to grabbing the release archive, only to discover that the version stuff is buried in make/project.mk in the release version.

negativekelvin, 2019-05-17:
Yes it only exists in 3.2 and up.  You could grep it out of project.mk in 3.1.

tom-borcin, 2021-03-25:
Hello @MitchBradley, we are closing this issue. Information you are looking for is present in tools/tools.json in all maintained releases.

If you think this issue should be open, please let me know. Thank you!

MitchBradley, 2021-03-25:
Any suggestions for how to access the tools.json info from a Makefile, other than by using a sed or awk script that would be sensitive to formatting changes in tools.json ?

dobairoland, 2021-04-14:
@MitchBradley The best way to access the content of `tools.json` is indirectly through `tools/idf_tools.py`. You can invoke this tool even from a makefile. You can download the toolchain with it and generate the paths which should be exported to use the tools. Please check out the `download`, `export` and other subcommands of `tools/idf_tools.py`. Examples for using this tool can be found in `install.sh` and `export.sh` files. All of these tools are part of the cmake toolchain which is in preview state for ESP-IDF v3.3 and not yet available in v3.2. These tools are used by default since v4.0.

If you are using the make build system with ESP-IDF v3.x then the toolchain versions are available in `tools/toolchain_versions.mk` as it was already discussed in this issue. 

At this moment only ESP-IDF v3.3 is maintained from the v3.x series. But only bugfixes are backported there. So I'm afraid if you missing some `*.mk` files from v3.1 or v3.2 then adding them there is no longer possible. 

So I don't think anything else could be done to resolve this "issue". I'm closing it but feel free to ask further questions here.

closed 2021-04-14 by dobairoland