espressif/esp-idf #3487: Console example - linenoise in PuTTY loops in white chars garbage

bsaid, 2019-05-16:
## Environment

- Development Kit:     ESP32-DevKitC
- Module or chip used:  ESP32-WROOM-32
- IDF version (run ``git describe --tags`` to find it): v4.0-dev-546-gad3b820e7
- Build System: Make
- Compiler version (run ``xtensa-esp32-elf-gcc --version`` to find it): crosstool-ng-1.22.0-80-g6c4433a5
- Operating System: Windows + Linux
- Power Supply: USB

## Problem Description

When the console example starts in PuTTY or in gnome-terminal and prints the prompt, then I press any character key, and then the terminal starts moving the cursor randomly and prints some white chars in infinite loop.

The wrong behavior occurs inside linenoiseEdit function:
https://github.com/espressif/esp-idf/blob/master/components/console/linenoise/linenoise.c : line 702

### Steps to repropduce

1. Run examples/system/console
2. Connect your ESP32 to terminal with ESC chars support (eg. PuTTY, screen, ...)
3. Press any character key after the prompt is displayed.

## Debug Logs

1. What ESP32 sends after boot to the terminal (normal behavior):

00000000 65 74 73 20 4A 75 6E 20 20 38 20 32 30 31 36 20    |ets Jun  8 2016 |
00000010 30 30 3A 32 32 3A 35 37 0D 0A 0D 0A 72 73 74 3A    |00:22:57....rst:|
00000020 30 78 31 20 28 50 4F 57 45 52 4F 4E 5F 52 45 53    |0x1 (POWERON_RES|
00000030 45 54 29 2C 62 6F 6F 74 3A 30 78 31 33 20 28 53    |ET),boot:0x13 (S|
00000040 50 49 5F 46 41 53 54 5F 46 4C 41 53 48 5F 42 4F    |PI_FAST_FLASH_BO|
00000050 4F 54 29 0D 0A 66 6C 61 73 68 20 72 65 61 64 20    |OT)..flash read |
00000060 65 72 72 2C 20 31 30 30 30 0D 0A 65 74 73 5F 6D    |err, 1000..ets_m|
00000070 61 69 6E 2E 63 20 33 37 31 20 0D 0A 65 74 73 20    |ain.c 371 ..ets |
00000080 4A 75 6E 20 20 38 20 32 30 31 36 20 30 30 3A 32    |Jun  8 2016 00:2|
00000090 32 3A 35 37 0D 0A 0D 0A 72 73 74 3A 30 78 31 30    |2:57....rst:0x10|
000000A0 20 28 52 54 43 57 44 54 5F 52 54 43 5F 52 45 53    | (RTCWDT_RTC_RES|
000000B0 45 54 29 2C 62 6F 6F 74 3A 30 78 31 33 20 28 53    |ET),boot:0x13 (S|
000000C0 50 49 5F 46 41 53 54 5F 46 4C 41 53 48 5F 42 4F    |PI_FAST_FLASH_BO|
000000D0 4F 54 29 0D 0A 63 6F 6E 66 69 67 73 69 70 3A 20    |OT)..configsip: |
000000E0 30 2C 20 53 50 49 57 50 3A 30 78 65 65 0D 0A 63    |0, SPIWP:0xee..c|
000000F0 6C 6B 5F 64 72 76 3A 30 78 30 30 2C 71 5F 64 72    |lk_drv:0x00,q_dr|
00000100 76 3A 30 78 30 30 2C 64 5F 64 72 76 3A 30 78 30    |v:0x00,d_drv:0x0|
00000110 30 2C 63 73 30 5F 64 72 76 3A 30 78 30 30 2C 68    |0,cs0_drv:0x00,h|
00000120 64 5F 64 72 76 3A 30 78 30 30 2C 77 70 5F 64 72    |d_drv:0x00,wp_dr|
00000130 76 3A 30 78 30 30 0D 0A 6D 6F 64 65 3A 44 49 4F    |v:0x00..mode:DIO|
00000140 2C 20 63 6C 6F 63 6B 20 64 69 76 3A 32 0D 0A 6C    |, clock div:2..l|
00000150 6F 61 64 3A 30 78 33 66 66 66 30 30 31 38 2C 6C    |oad:0x3fff0018,l|
00000160 65 6E 3A 34 0D 0A 6C 6F 61 64 3A 30 78 33 66 66    |en:4..load:0x3ff|
00000170 66 30 30 31 63 2C 6C 65 6E 3A 31 31 30 30 0D 0A    |f001c,len:1100..|
00000180 6C 6F 61 64 3A 30 78 34 30 30 37 38 30 30 30 2C    |load:0x40078000,|
00000190 6C 65 6E 3A 39 32 33 32 0D 0A 6C 6F 61 64 3A 30    |len:9232..load:0|
000001A0 78 34 30 30 38 30 34 30 30 2C 6C 65 6E 3A 36 34    |x40080400,len:64|
000001B0 31 32 0D 0A 65 6E 74 72 79 20 30 78 34 30 30 38    |12..entry 0x4008|
000001C0 30 36 61 38 0D 0A 0D 0A 54 68 69 73 20 69 73 20    |06a8....This is |
000001D0 61 6E 20 65 78 61 6D 70 6C 65 20 6F 66 20 45 53    |an example of ES|
000001E0 50 2D 49 44 46 20 63 6F 6E 73 6F 6C 65 20 63 6F    |P-IDF console co|
000001F0 6D 70 6F 6E 65 6E 74 2E 0D 0A 54 79 70 65 20 27    |mponent...Type '|
00000200 68 65 6C 70 27 20 74 6F 20 67 65 74 20 74 68 65    |help' to get the|
00000210 20 6C 69 73 74 20 6F 66 20 63 6F 6D 6D 61 6E 64    | list of command|
00000220 73 2E 0D 0A 55 73 65 20 55 50 2F 44 4F 57 4E 20    |s...Use UP/DOWN |
00000230 61 72 72 6F 77 73 20 74 6F 20 6E 61 76 69 67 61    |arrows to naviga|
00000240 74 65 20 74 68 72 6F 75 67 68 20 63 6F 6D 6D 61    |te through comma|
00000250 6E 64 20 68 69 73 74 6F 72 79 2E 0D 0A 50 72 65    |nd history...Pre|
00000260 73 73 20 54 41 42 20 77 68 65 6E 20 74 79 70 69    |ss TAB when typi|
00000270 6E 67 20 63 6F 6D 6D 61 6E 64 20 6E 61 6D 65 20    |ng command name |
00000280 74 6F 20 61 75 74 6F 2D 63 6F 6D 70 6C 65 74 65    |to auto-complete|
00000290 2E 0D 0A 1B 5B 36 6E 65 74 73 20 4A 75 6E 20 20    |....[6nets Jun  |
000002A0 38 20 32 30 31 36 20 30 30 3A 32 32 3A 35 37 0D    |8 2016 00:22:57.|
[... 290 lines cut ...]
00000FD0 1B 5B 33 43 0D 1B 5B 30 4B 1B 5B 31 41 0D 1B 5B    |.[3C..[0K.[1A..[|
00000FE0 30 4B 1B 5B 30 3B 33 32 6D 65 73 70 33 32 3E 20    |0K.[0;32mesp32> |
00000FF0 1B 5B 30 6D 68 13 11 13 11 13 11 13 11 13 11 13    |.[0mh...........|
00001000 11 13 11 13 11 13 11 13 11 13 11 13 11 13 11 13    |................|
00001010 11 13 11 13 11 13 11 13 11 13 11 13 11 13 11 13    |................|
00001020 11 13 11 13 11 13 11 13 11 13 11 13 11 13 11 13    |................|
00001030 11 13 11 13 11 13 0D 1B 5B 34 43 0D 1B 5B 30 4B    |........[4C..[0K|
00001040 1B 5B 31 41 0D 1B 5B 30 4B 1B 5B 30 3B 33 32 6D    |.[1A..[0K.[0;32m|
00001050 65 73 70 33 32 3E 20 1B 5B 30 6D 68 13 11 13 11    |esp32> .[0mh....|
00001060 13 11 13 11 13 11 13 11 13 11 13 11 13 11 13 11    |................|
00001070 13 11 13 11 13 11 13 11 13 11 13 11 13 11 13 11    |................|
00001080 13 11 13 11 13 11 13 11 13 11 13 11 13 11 13 11    |................|
00001090 13 11 13 11 13 11 13 11 13 11 13 11 13 11 0D 1B    |................|
(...)

3. What terminal sends to ESP32:

00000000 13 11 13 11 13 11 13 11 13 11 13 11 13 11 13 11    |................|
00000010 13 11 13 11 13 11 13 11 1B 5B 32 32 3B 31 52 13    |.........[22;1R.|
00000020 11 1B 5B 32 32 3B 31 52 13 11 1B 5B 32 32 3B 38    |..[22;1R...[22;8|
00000030 52 68 13 11 13 11 13 11 13 11 13 11 13 11 13 11    |Rh..............|
00000040 13 11 13 11 13 11 13 11 13 11 13 11 13 11 13 11    |................|
00000050 13 11 13 11 13 11 13 11 13 11 13 11 13 11 13 11    |................|
00000060 13 11 13 11 13 11 13 11 13 11 13 11 13 11 13 11    |................|
(... infinitely repeats ...)

KonstantinKondrashov, 2019-05-17:
Hi @bsaid! I work with this console example right now and use PuTTY, but I do not see any issue with it. It works fine for me. Probably you have special settings or something else. Could you clarify that?
Please provide your sdkconfig file.

[image]

bsaid, 2019-05-19:
And the browsing history and auto-complete is really working, too? Well, I will perform a completely clean install to see if the problem persists. Thank you @KonstantinKondrashov

KonstantinKondrashov, 2019-05-20:
> And the browsing history and auto-complete is really working, too?

Yes, the browsing history and auto-complete work very well. 

FYI, There is [one option in Kconfig](https://github.com/espressif/esp-idf/blob/master/examples/system/console/main/Kconfig.projbuild#L3-L9) which enable the browsing history. This option turned on by default.

Please provide your sdkconfig to see which options you use.

Alvin1Zhang, 2019-05-23:
@bsaid Hi, thanks for reporting the issue. Would you please help share your sdkconfig to see which options you use? Thanks.

bsaid, 2019-05-24:
I tried it with clean Windows installation, clean msys32 and latest esp-idf, all configuration is kept in default. The issue is still present. I tried to manually filter chars 0x11 and 0x13 on the serial line and it helped, but this is not a solution. My sdkconfig is here (.txt added manually):
[sdkconfig.txt](https://github.com/espressif/esp-idf/files/3215518/sdkconfig.txt)

bsaid, 2019-05-24:
[image]

KonstantinKondrashov, 2019-05-29:
Hi @bsaid! I tried again to reproduce your issue I used your sdkconfig file but can not see this behavior. You wrote that "terminal sends to ESP32: 13 11 13 11 ......" - I don't know why the Putty sends it to the ESP32. These bytes mean [XOFF and XON](https://en.wikipedia.org/wiki/Software_flow_control).

I noticed that your console example has some changes (v4.0-dev-546-gad3b820e7-**dirty**). May they lead to this behavior? If you use original example do you also see this behavior?

bsaid, 2019-05-29:
XOFF and XON? I know what is the problem. We can enable/disable XON/XOFF control inside PuTTY configuration (see attached image). When I changed it to "None", the Console example was **working normally**.
The XON/XOFF was unfortunately enabled by default in my PuTTY installation. The same configuration change is probably needed in [screen](https://www.gnu.org/software/screen/) application under Linux.

[image]

closed 2019-05-29 by bsaid