espressif/esp-idf #3489: ble_ibeacon_example not compiling esp_gap_ble_api.c

ASL07, 2019-05-16:
When I try to compile the example ble_ibeacon_example in mode IBEACON_RECEIVER, the source file esp_gap_ble_api.c  is not included in the compilation. It compiles well with mode IBEACON_SENDER.

This is the compilation output for IBEACON_RECEIVER. Hope you can help me troubleshoot this
```
/Users/developer/esp2/esp-idf/examples/bluetooth/ble_ibeacon/build/main/libmain.a(ibeacon_demo.o):(.literal.esp_gap_cb+0x3c): undefined reference to `esp_ble_gap_start_scanning'
/Users/developer/esp2/esp-idf/examples/bluetooth/ble_ibeacon/build/main/libmain.a(ibeacon_demo.o): In function `esp_gap_cb':
/Users/developer/esp2/esp-idf/examples/bluetooth/ble_ibeacon/main/ibeacon_demo.c:80: undefined reference to `esp_ble_gap_start_scanning'
collect2: error: ld returned 1 exit status
make: *** [/Users/developer/esp2/esp-idf/examples/bluetooth/ble_ibeacon/build/ble_ibeacon_demo.elf] Error 1

```

And this is the content of my sdkconfig file:
```
#
# Automatically generated file; DO NOT EDIT.
# Espressif IoT Development Framework Configuration
#

#
# SDK tool configuration
#
CONFIG_TOOLPREFIX="xtensa-esp32-elf-"
CONFIG_PYTHON="python"
CONFIG_MAKE_WARN_UNDEFINED_VARIABLES=y

#
# Bootloader config
#
CONFIG_LOG_BOOTLOADER_LEVEL_NONE=
CONFIG_LOG_BOOTLOADER_LEVEL_ERROR=
CONFIG_LOG_BOOTLOADER_LEVEL_WARN=
CONFIG_LOG_BOOTLOADER_LEVEL_INFO=y
CONFIG_LOG_BOOTLOADER_LEVEL_DEBUG=
CONFIG_LOG_BOOTLOADER_LEVEL_VERBOSE=
CONFIG_LOG_BOOTLOADER_LEVEL=3
CONFIG_BOOTLOADER_VDDSDIO_BOOST_1_8V=
CONFIG_BOOTLOADER_VDDSDIO_BOOST_1_9V=y
CONFIG_BOOTLOADER_FACTORY_RESET=
CONFIG_BOOTLOADER_APP_TEST=

#
# Security features
#
CONFIG_SECURE_SIGNED_APPS_NO_SECURE_BOOT=
CONFIG_SECURE_BOOT_ENABLED=
CONFIG_FLASH_ENCRYPTION_ENABLED=

#
# Serial flasher config
#
CONFIG_ESPTOOLPY_PORT="/dev/ttyUSB0"
CONFIG_ESPTOOLPY_BAUD_115200B=y
CONFIG_ESPTOOLPY_BAUD_230400B=
CONFIG_ESPTOOLPY_BAUD_921600B=
CONFIG_ESPTOOLPY_BAUD_2MB=
CONFIG_ESPTOOLPY_BAUD_OTHER=
CONFIG_ESPTOOLPY_BAUD_OTHER_VAL=115200
CONFIG_ESPTOOLPY_BAUD=115200
CONFIG_ESPTOOLPY_COMPRESSED=y
CONFIG_FLASHMODE_QIO=
CONFIG_FLASHMODE_QOUT=
CONFIG_FLASHMODE_DIO=y
CONFIG_FLASHMODE_DOUT=
CONFIG_ESPTOOLPY_FLASHMODE="dio"
CONFIG_ESPTOOLPY_FLASHFREQ_80M=
CONFIG_ESPTOOLPY_FLASHFREQ_40M=y
CONFIG_ESPTOOLPY_FLASHFREQ_26M=
CONFIG_ESPTOOLPY_FLASHFREQ_20M=
CONFIG_ESPTOOLPY_FLASHFREQ="40m"
[... 721 lines cut ...]
CONFIG_SPIFFS_CHECK_DBG=
CONFIG_SPIFFS_TEST_VISUALISATION=

#
# tcpip adapter
#
CONFIG_IP_LOST_TIMER_INTERVAL=120

#
# Virtual file system
#
CONFIG_SUPPRESS_SELECT_DEBUG_OUTPUT=y

#
# Wear Levelling
#
CONFIG_WL_SECTOR_SIZE_512=
CONFIG_WL_SECTOR_SIZE_4096=y
CONFIG_WL_SECTOR_SIZE=4096

```

- [*] I have read the documentation [ESP-IDF Programming Guide](https://docs.espressif.com/projects/esp-idf/en/latest/) and the issue is not addressed there.
- [*] I have updated my IDF branch (master or release) to the latest version and checked that the issue is present there.
- [*] I have searched the issue tracker for a similar issue and not found a similar issue.

dhrishi, 2019-05-17:
@ASL07 Please provide more details about the branch (and the head commit id) you are using. 
Also, check if the IDF_PATH is correct. 
The compilation for this example seems to work fine for both modes  (IBEACON_SENDER as well as IBEACON_RECEIVER) on latest master (ad3b820e701c3ef0803b045b5a2c5ef19630fb0b)

Weijian-Espressif, 2019-05-20:
@ASL07 , I don't reproduce this issue, my IDF commit is 6cdfde094d2c8b5d3b58ad59e86786300a69307c
 
```
zhiweijian:ble_ibeacon zhiweijian$ 
zhiweijian:ble_ibeacon zhiweijian$ make -j4
Toolchain path: /Users/zhiweijian/esp/xtensa-esp32-elf/bin/xtensa-esp32-elf-gcc
WARNING: Toolchain version is not supported: crosstool-ng-1.22.0-61-gab8375a
Expected to see version: crosstool-ng-1.22.0-80-g6c4433a
Please check ESP-IDF setup instructions and update the toolchain, or proceed at your own risk.
Compiler version: 5.2.0
Python requirements from /Users/zhiweijian/Desktop/esp-idf/requirements.txt are satisfied.

App "ble_ibeacon_demo" version: v4.0-dev-549-g6cdfde094-dirty
WARNING: Toolchain version is not supported: crosstool-ng-1.22.0-61-gab8375a
Expected to see version: crosstool-ng-1.22.0-80-g6c4433a
Please check ESP-IDF setup instructions and update the toolchain, or proceed at your own risk.
To flash all build output, run 'make flash' or:
python /Users/zhiweijian/Desktop/esp-idf/components/esptool_py/esptool/esptool.py --chip esp32 --port /dev/tty.SLAB_USBtoUART --baud 921600 --before default_reset --after hard_reset write_flash -z --flash_mode dio --flash_freq 40m --flash_size detect 0x1000 /Users/zhiweijian/Desktop/esp-idf/examples/bluetooth/ble_ibeacon/build/bootloader/bootloader.bin 0x10000 /Users/zhiweijian/Desktop/esp-idf/examples/bluetooth/ble_ibeacon/build/ble_ibeacon_demo.bin 0x8000 /Users/zhiweijian/Desktop/esp-idf/examples/bluetooth/ble_ibeacon/build/partitions_singleapp.bin
zhiweijian:ble_ibeacon zhiweijian$ git log
commit 6cdfde094d2c8b5d3b58ad59e86786300a69307c (HEAD -> master)
Merge: ad3b820e7 015922f8d
Author: Angus Gratton <angus@espressif.com>
Date:   Thu May 16 19:55:47 2019 +0800

    Merge branch 'bugfix/protocomm_ble_128bit_uuid' into 'master'
    
    protocomm_ble : Fix support for custom service UUIDs
    
    See merge request idf/esp-idf!4978

```

ASL07, 2019-05-20:
Sorry, my fault.

My ESP_IDF path was pointing to an older version of esp-idf

I fixed that and it works using the latest master.

Thanks for your help

Alvin1Zhang, 2019-05-20:
@ASL07 Thanks for reporting the issue. Feel free to reopen the issue if it still exists. Thanks.

closed 2019-05-20 by Alvin1Zhang