espressif/esp-idf #3490: Mocking hardware dependencies? IDFGH-1177)

coolcodehere, 2019-05-16:
What's the recommended way to mock hardware dependencies?

If there isn't one is there a good way to get the IDF to work with ceedling?

Alvin1Zhang, 2019-05-17:
Thanks for reporting the issue. Would you please help provide more details as suggested in the [issue template](https://github.com/espressif/esp-idf/blob/master/docs/issue_template.md)? Thanks. @coolcodehere

coolcodehere, 2019-05-17:
Development Kit: ESP32-DevKitC
Module or chip used: ESP32-WROOM-32
IDF version: v3.2-dev-1148-g96cd3b75c
Build System: Make
Operating System: macOS

Here's all relevant info from the template. I really just need to figure out how to get either cmock or ceedling to work with the IDF.

Alvin1Zhang, 2019-05-21:
[#3501](https://github.com/espressif/esp-idf/issues/3501)

coolcodehere, 2019-05-21:
> [#3501](https://github.com/espressif/esp-idf/issues/3501)

Alvin I posted that one too.

Alvin1Zhang, 2019-05-21:
@coolcodehere Thanks and understand, therefore I referenced them.

deltalejo, 2019-10-20:
I wrote an example on how test doubles can be integrated into a project using the ESP-IDF build system, maybe it can be useful.
Here is the link: https://gitlab.com/deltalejo/esp32-unity-fff-demo

0xjakob, 2020-09-10:
@coolcodehere @deltalejo We're currently integrating cmock as an IDF-component. Soon, there will also be examples of IDF-unit tests using cmock to mock specific components. CMock actually includes Unity as submodule and both are part of ThrowTheSwitch. It also seems a bit less verbose when using it compared to FFF (looked at it briefly).

mc18g13, 2022-01-22:
This is a few years old now but could be useful for others.
I created an [example project](https://github.com/mc18g13/esp-idf-example) using cmock based on [nvs_page_test](https://github.com/espressif/esp-idf/tree/master/components/nvs_flash/host_test/nvs_page_test).
You could use this structure to mock esp-idf components as well as your own.

0xjakob, 2022-01-24:
@mc18g13 Looks great! You can actually re-use the original component as your mock component name, just make sure that it's added later to the build system, e.g. via `list(APPEND EXTRA_COMPONENT_DIRS "$ENV{SOURCE_DIRECTORY}/mocks/my_component_dependency")`. Then *override* the original component in the mock component as described in the [IDF build system documentation](https://docs.espressif.com/projects/esp-idf/en/latest/esp32/api-guides/build-system.html?highlight=build%20system#multiple-components-with-the-same-name). This is how we mock stuff in ESP-IDF. Makes it a bit easier. But thanks for the guidelines!

Btw. we have other unit tests using cmock in IDF in the [cxx experimental component host tests](https://github.com/espressif/esp-idf/tree/master/examples/cxx/experimental/experimental_cpp_component/host_test), the [esp_event host tests](https://github.com/espressif/esp-idf/tree/master/components/esp_event/host_test) and the [mqtt host tests](https://github.com/espressif/esp-idf/tree/master/components/mqtt/host_test).

0xjakob, 2022-01-24:
@coolcodehere We have guidelines on [how to use CMock in ESP-IDF](https://docs.espressif.com/projects/esp-idf/en/latest/esp32/api-guides/unit-tests.html#mocks) now. I'll close this issue then. If any questions come up, feel free to re-open the issue or create a new one.

closed 2022-01-24 by 0xjakob