espressif/esp-idf #3491: CMAKE Build fails on component folder

bmakovecki, 2019-05-16:
## Environment

- Development Kit:   ESP32-DevKitC
- IDF version : v4.0-dev-546-gad3b820e7
- Build System: CMake
- Compiler version: xtensa-esp32-elf-gcc (crosstool-NG crosstool-ng-1.22.0-80-g6c4433a5) 5.2.0
- Operating System:     Windows
- Project to build: [nvs_value_test.zip](https://github.com/espressif/esp-idf/files/3188481/nvs_value_test.zip)

## Problem Description
Build process fails during compilation:
In file included from ../components/test/test.cpp:1:0:
../components/test/include/test.h:3:17: fatal error: nvs.h: No such file or directory

### Expected Behavior

Build successful. Same as building with make.

### Steps to reproduce

1. idf.py build => fails
2. make app => succeed

### Code to reproduce this issue

[nvs_value_test.zip](https://github.com/espressif/esp-idf/files/3188481/nvs_value_test.zip)

projectgus, 2019-05-17:
In the CMake build system, components need to declare their dependencies (apart from some common dependencies).

If you add this line before register_component() in component/test/CMakeLists.txt, it will build:

```
set(COMPONENT_REQUIRES nvs_flash)
````

(Note: There is also COMPONENT_PRIV_REQUIRES if a header is only used to build the component, but not needed to include that component's public headers. Full details here: https://docs.espressif.com/projects/esp-idf/en/latest/api-guides/build-system-cmake.html#when-writing-a-component

closed 2019-05-17 by projectgus

TimeWarDoctor, 2020-03-08:
Made a copy of esp-idf\components\esp-eth and named the folder ethernet and it would build.