espressif/esp-idf #3492: Please provide toolchain with c++ RTTI enabled

bpietsch, 2019-05-17:
I would like to request that you please provide a version of the toolchain that has C++ RTTI enabled by default. RTTI is an essential part of most C++ development and having to build a custom toolchain is cumbersome. Thank you!

chanibal, 2019-05-20:
I wouldn't say that RTTI is essential to most of C++ development, nevertheless, ESP-IDF supports it, just configure your compiler.

Create a `component.mk` file in your main directory and in all components (if you use them) are with the contents:
```
CPPFLAGS += -frtti
```

Checked with:
```
class Person
  {
    public:
       virtual ~Person() {}
  };

 class Employee : public Person
 {
 };

#include <esp_log.h>
#include <typeinfo>
extern "C" void app_main(void) {
   Person person;
   Employee employee;
   Person* ptr = &employee;
   Person& ref = employee;
   // The string returned by typeid::name is implementation-defined
   ESP_LOGI("RTTI", "%s", typeid(person).name());   // Person (statically known at compile-time)
   ESP_LOGI("RTTI", "%s", typeid(employee).name()); // Employee (statically known at compile-time)
   ESP_LOGI("RTTI", "%s", typeid(ptr).name());      // Person* (statically known at compile-time)
   ESP_LOGI("RTTI", "%s", typeid(*ptr).name());     // Employee (looked up dynamically at run-time because it is the dereference of a pointer to a polymorphic class)
   ESP_LOGI("RTTI", "%s", typeid(ref).name());      // Employee (references can also be polymorphic)
}
```

bpietsch, 2019-05-20:
The compiler supports it, yes, but the standard libraries included in the pre-built toolchain do not.

bpietsch, 2019-05-30:
And, actually, it appears the cmake build system does not, in fact, allow me to specify -frtti... this entry in build.cmake appears to not be overridable:

    list(APPEND cxx_compile_options "-std=gnu++11"
                                    "-fno-rtti")

igrr, 2019-06-05:
Closing this as a duplicate of an older issue, https://github.com/espressif/esp-idf/issues/1684. To receive further updates, please follow the link and click the "Subscribe" button. Thanks.

closed 2019-06-05 by igrr