espressif/esp-idf #3493: project.mk msys path issue on windows

akotowski, 2019-05-18:
Yesterday I got a project using arduino components to compile in Eclipse after  implementing the fix suggested in "Eclipse on Windows shows Errors due to failing to find all header paths (build succeeds in background) (IDFGH-1059) #17" by @dsptech.  
I created a new project by copying & editing the working project.  I still had some compile warnings but compiling was working well.

Today after rebooting my computer, on the same project I now get:
...
C:/Users/Andreas/eclipse-workspace/WM/build/main\libmain.a(WMLoRa.o):(.data.ssid+0x0): multiple definition of `ssid'
C:/Users/Andreas/eclipse-workspace/WM/build/main\libmain.a(WM.o):(.data.ssid+0x0): first defined here
**collect2.exe: error: ld returned 1 exit status
make: *** [C:\msys32\home\Andreas\esp\esp-idf/make/project.mk:461: /c/Users/Andreas/eclipse-workspace/WM/build/WM.elf] Error 1**
"make -j4 all" terminated with exit code 2. Build might be incomplete.

It looks like an msys path on my Windows 10, esp-idf 3.2 system. The error occurs in Eclipse as well as running esp-idf in an msys window.

Any help appreciated.
A Kotowski

dsptech, 2019-05-18:
Hi, as I already said, the fix is not tested on the last esp-idf framework.
Could you try to remove the fix from the project makefile ?

akotowski, 2019-05-19:
The fix is not the problem, it is still working well.  It must be the structure of my program, I swapped a different version of the source code which has multiple files in the "main" directory.

dsptech, 2019-05-19:
Hi,
It seems to me that the two variables "ssid" are declared in two differerent sources (WM.c and WMLoRa.c ?) without the "static" qualifier, or "ssid" is declared in a common shared header without the "extern" keyword.  
Are you declaring such variable ? Is the project cloned correctly ? Could you try a "make clean" and "make menuconfig" (changing any option)  on the cloned project ?

akotowski, 2019-05-20:
@dsptech 
You are right "extern" declarations were missing.  I just did not know how to decode the error message.
Thank You

closed 2019-05-20 by akotowski