espressif/esp-idf #3494: Is there a memory leak in mqtt tcp?

Feng573795502, 2019-05-18:
```
   while(1)
    {
        if((event_bits & SEND_DATA) == SEND_DATA)
        {
            err_cnt = wifi_mqtt_init();
            //err_cnt = 5;
        }
        if((event_bits & MQTT_SEND) == MQTT_SEND)
        {
            mqtt_send_data();
            err_cnt = 5;
        }
        else if((event_bits & CONN_ERR) == CONN_ERR)
        {
            err_cnt++;
            // esp_mqtt_client_stop(mqtt_client);
            // esp_mqtt_client_start(mqtt_client);
        }
        else if((event_bits & CLOSE_WIRELESS) == CLOSE_WIRELESS)
        {
            err_cnt = 0;
            esp_mqtt_client_destroy(mqtt_client);
            wireless_off_wifi();
        }
        if(err_cnt >= 5)
        {
            TRACE_LOG("TEST", "关掉wifi");
            err_cnt = 0;
            esp_mqtt_client_stop(mqtt_client);
            esp_mqtt_client_destroy(mqtt_client);
            //esp_mqtt_client_destroy(mqtt_client);
            wireless_off_wifi();
        }
        xTaskNotifyWait(0x00, 0xffffffff, &event_bits, portMAX_DELAY);
    }

static uint8_t wifi_mqtt_init(void)
{
    uint32_t event_bits;
    uint8_t res = 0;
    /* 启动wifi发送数据 */
    wifi_start();
    xTaskNotifyWait(0x00, 0xffffffff, &event_bits, 25000);
    if((event_bits & 0x01) != 0x01)
    {
        res = 5;
    }
    else
    {
        wireless_read_signal();
        mqtt_app_start();
        if(strlen((char *)system_get_mqtt_srv_addr()) !=0 && strlen((char *)system_get_mqtt_srv_port()) != 0)
        {
            sprintf((char *)mqtt_url, "mqtt://%s:%s", system_get_mqtt_srv_addr(), system_get_mqtt_srv_port());
        }
        else 
        {
            sprintf((char *)mqtt_url, "mqtt://mq.tlink.io:1883");
            
        }
        esp_mqtt_client_set_uri(mqtt_client, (char *)mqtt_url);   
        esp_mqtt_client_start(mqtt_client);
    }
    return res;
}

static void wireless_off_wifi(void)
{
    wireless_info.wifi_on = 0;

    esp_wifi_stop();
    //esp_wifi_deinit();

    vTaskDelay(100);
    pwr_arg_halt(PWR_WIFI_BIT);
}
```

LOG 
`

[18:19:57.546]收←◆E (152261) WAKE: [APP] Free memory: 102888 bytes

[18:19:58.457]收←◆E (153174) PWR: exit sleep
E (153183) wifi: 启动连接
E (153194) WAKE: [APP] Free memory: 101564 bytes

[18:19:59.480]收←◆E (154195) WAKE: [APP] Free memory: 101276 bytes

[18:20:00.482]收←◆E (155197) WAKE: [APP] Free memory: 101564 bytes

[18:20:00.748]收←◆E (155458) wifi: 获取IP地址成功:
E (155460) WIFI: 信号强度-63

E (155462) wifi: other event id:7

[18:20:00.964]收←◆E (155661) MQTT: {"times":"2019-05-18 18:20:02","sensorDatas":[{"flag":"1","value":27.67},{"flag":"2","value":76.54},{"flag":"3","value":0.00},{"flag":"4","value":0.00},{"flag":"5","value":4031},{"flag":"6","switcher":"1"},{"flag":"7","value":25}]}

[18:20:01.484]收←◆E (156199) WAKE: [APP] Free memory: 91044 bytes

[18:20:01.962]收←◆E (156679) TEST: 关掉wifi

[18:20:02.486]收←◆E (157201) WAKE: [APP] Free memory: 90752 bytes

[18:20:03.488]收←◆E (158203) WAKE: [APP] Free memory: 102772 bytes

[18:20:04.490]收←◆E (159205) WAKE: [APP] Free memory: 102776 bytes

[18:20:49.534]收←◆E (204250) WAKE: [APP] Free memory: 102776 bytes

[18:20:50.535]收←◆E (205251) WAKE: [APP] Free memory: 102776 bytes

[18:20:51.536]收←◆E (206252) WAKE: [APP] Free memory: 102776 bytes

[18:20:52.537]收←◆E (207253) WAKE: [APP] Free memory: 102776 bytes

[18:20:53.538]收←◆E (208254) WAKE: [APP] Free memory: 102776 bytes

[18:20:54.539]收←◆E (209255) WAKE: [APP] Free memory: 102776 bytes

[18:20:55.540]收←◆E (210256) WAKE: [APP] Free memory: 102484 bytes

[18:20:56.541]收←◆E (211257) WAKE: [APP] Free memory: 102776 bytes

[18:20:57.541]收←◆E (212258) WAKE: [APP] Free memory: 102776 bytes

[18:20:58.452]收←◆E (213171) PWR: exit sleep
E (213180) wifi: 启动连接
E (213191) WAKE: [APP] Free memory: 101156 bytes

[18:20:59.476]收←◆E (214192) WAKE: [APP] Free memory: 101444 bytes

[18:21:00.478]收←◆E (215194) WAKE: [APP] Free memory: 101444 bytes

[18:21:00.713]收←◆E (215425) wifi: 获取IP地址成功:
E (215427) WIFI: 信号强度-65

E (215430) wifi: other event id:7

[18:21:00.780]收←◆E (215478) MQTT: {"times":"2019-05-18 18:21:02","sensorDatas":[{"flag":"1","value":27.66},{"flag":"2","value":76.47},{"flag":"3","value":0.00},{"flag":"4","value":0.00},{"flag":"5","value":4031},{"flag":"6","switcher":"1"},{"flag":"7","value":25}]}

[18:21:01.480]收←◆E (216196) WAKE: [APP] Free memory: 90796 bytes

[18:21:01.776]收←◆E (216495) TEST: 关掉wifi

[18:21:02.482]收←◆E (217198) WAKE: [APP] Free memory: 91092 bytes

[18:21:03.484]收←◆E (218200) WAKE: [APP] Free memory: 102676 bytes

[18:21:04.485]收←◆E (219202) WAKE: [APP] Free memory: 102396 bytes

[18:21:05.486]收←◆E (220203) WAKE: [APP] Free memory: 102680 bytes

[18:21:06.487]收←◆E (221204) WAKE: [APP] Free memory: 102680 bytes

[18:21:07.488]收←◆E (222205) WAKE: [APP] Free memory: 102388 bytes

[18:21:08.489]收←◆E (223206) WAKE: [APP] Free memory: 102676 bytes

[18:21:09.490]收←◆E (224207) WAKE: [APP] Free memory: 102680 bytes

[18:21:10.491]收←◆E (225208) WAKE: [APP] Free memory: 102684 bytes

[18:21:11.492]收←◆E (226209) WAKE: [APP] Free memory: 102684 bytes

[18:21:12.493]收←◆E (227210) WAKE: [APP] Free memory: 102684 bytes

[18:21:13.494]收←◆E (228211) WAKE: [APP] Free memory: 102684 bytes

[18:21:14.495]收←◆E (229212) WAKE: [APP] Free memory: 102684 bytes

[18:21:15.496]收←◆E (230213) WAKE: [APP] Free memory: 102684 bytes

[18:21:58.449]收←◆E (273168) PWR: exit sleep
E (273177) wifi: 启动连接
E (273189) WAKE: [APP] Free memory: 101192 bytes

[18:21:59.473]收←◆E (274190) WAKE: [APP] Free memory: 101364 bytes

[18:22:00.475]收←◆E (275192) WAKE: [APP] Free memory: 101072 bytes

[18:22:00.680]收←◆E (275393) wifi: 获取IP地址成功:
E (275395) WIFI: 信号强度-56

E (275397) wifi: other event id:7

[18:22:00.760]收←◆E (275461) MQTT: {"times":"2019-05-18 18:22:02","sensorDatas":[{"flag":"1","value":27.62},{"flag":"2","value":76.43},{"flag":"3","value":0.00},{"flag":"4","value":0.00},{"flag":"5","value":4031},{"flag":"6","switcher":"1"},{"flag":"7","value":25}]}

[18:22:01.477]收←◆E (276194) WAKE: [APP] Free memory: 91004 bytes

[18:22:01.758]收←◆E (276478) TEST: 关掉wifi

[18:22:02.479]收←◆E (277196) WAKE: [APP] Free memory: 91004 bytes

[18:22:03.480]收←◆E (278198) WAKE: [APP] Free memory: 102328 bytes

[18:22:04.482]收←◆E (279200) WAKE: [APP] Free memory: 102620 bytes`

**One minute to enter the reminder once SEND_DATA enters the mqtt sending stage  There is no memory leak problem only when starting wifi and stopping wifi. If mqtt is started, there will be a memory leak, but mqtt is already using esp_mqtt_client_destroy();**

projectgus, 2019-05-21:
Hi @Feng573795502 ,

When you close a TCP socket, some memory is still used while it says in the TIME_WAIT state. After a minute or two the socket will be cleaned up and the memory is freed. If you leave your project running for a few minutes, does the memory usage stabilise?

Angus

Feng573795502, 2019-05-21:
> 您好@ Feng573795502，
> 
> 当您关闭TCP套接字时，仍然会使用某些内存，而套接字仍处于TIME_WAIT状态。一两分钟后，套接字将被清理并释放内存。如果让项目运行几分钟，内存使用情况是否稳定？
> 
> 安格斯

No, no, the device will use MQTT once every time, the memory will be reduced in turn until the remaining memory is insufficient to restart automatically, so it is very troublesome

david-cermak, 2019-05-22:
Hi @Feng573795502 

I don't think there's a leak in MQTT, at least not in this obvious scenario. I've updated your code to a simple loop which connects, publishes and destroys in turn repeatedly:
```
        if ((event_bits & SEND_DATA) == SEND_DATA) {
            wifi_mqtt_init();
            ESP_LOGW(TAG, "[After init]: %d bytes", esp_get_free_heap_size());
            event_bits = MQTT_SEND;
        } else if ((event_bits & MQTT_SEND) == MQTT_SEND) {
            mqtt_send_data();
            ESP_LOGW(TAG, "[After publish]: %d bytes", esp_get_free_heap_size());
            event_bits = CLOSE_WIRELESS;
        } else if ((event_bits & CLOSE_WIRELESS) == CLOSE_WIRELESS) {
            esp_mqtt_client_destroy(mqtt_client);
            ESP_LOGW(TAG, "[After destroy]: %d bytes", esp_get_free_heap_size());
            event_bits = SEND_DATA;
        }
```
and getting quite consistent results after several hours of executing
```
W (5512690) MQTT_EXAMPLE: [After destroy]: 225072 bytes
W (5513690) MQTT_EXAMPLE: [After init]: 220604 bytes
W (5514690) MQTT_EXAMPLE: [After publish]: 218300 bytes
*** after 2 hours ***
W (6253190) MQTT_EXAMPLE: [After destroy]: 225188 bytes
```

Also I've noticed that you're sending JSON data, so expecting using cJSON library. Don't accidentally forget to free the buffer after `cJSON_Print`? Is it possible to send a constant text just for the sake of test to see if the memory still leaks?

Feng573795502, 2019-05-23:
> 您好@ Feng573795502
> 
> 我认为MQTT没有泄漏，至少在这个明显的情况下没有。我已经将您的代码更新为一个简单的循环，它反复连接，发布和销毁：
> 
> ```
>         if ((event_bits & SEND_DATA) == SEND_DATA) {
>             wifi_mqtt_init();
>             ESP_LOGW(TAG, "[After init]: %d bytes", esp_get_free_heap_size());
>             event_bits = MQTT_SEND;
>         } else if ((event_bits & MQTT_SEND) == MQTT_SEND) {
>             mqtt_send_data();
>             ESP_LOGW(TAG, "[After publish]: %d bytes", esp_get_free_heap_size());
>             event_bits = CLOSE_WIRELESS;
>         } else if ((event_bits & CLOSE_WIRELESS) == CLOSE_WIRELESS) {
>             esp_mqtt_client_destroy(mqtt_client);
>             ESP_LOGW(TAG, "[After destroy]: %d bytes", esp_get_free_heap_size());
>             event_bits = SEND_DATA;
>         }
> ```
> 
> 并在执行几个小时后得到相当一致的结果
> 
> ```
> W (5512690) MQTT_EXAMPLE: [After destroy]: 225072 bytes
> W (5513690) MQTT_EXAMPLE: [After init]: 220604 bytes
> W (5514690) MQTT_EXAMPLE: [After publish]: 218300 bytes
> *** after 2 hours ***
> W (6253190) MQTT_EXAMPLE: [After destroy]: 225188 bytes
> ```
> 
> 另外我注意到你正在发送JSON数据，所以期待使用cJSON库。不小心忘记释放缓冲后`cJSON_Print`？是否可以仅为了测试而发送一个常量文本，以查看内存是否仍然泄漏？

> Hi @Feng573795502
>
> I don't think there's a leak in MQTT, at least not in this obvious scenario. I've updated your code to a simple loop which connects, publishes and destroys in turn repeatedly: [...]

`    data_time = chip_get_time();
    sprintf(mqtt_payload, "{\"times\":\"20%d-%02d-%02d %02d:%02d:%02d\""
			                      ",\"sensorDatas\":[{\"flag\":\"1\",\"value\":%.2f},{\"flag\":\"2\",\"value\":%.2f},{\"flag\":\"3\",\"value\":%.2f},"
		"{\"flag\":\"4\",\"value\":%.2f},{\"flag\":\"5\",\"value\":%d},{\"flag\":\"6\",\"switcher\":\"%d\"},{\"flag\":\"7\",\"value\":%d}]}",
    data_time->year, 
    data_time->month,
    data_time->day,
    data_time->hour,
    data_time->min,
    data_time->sec,
    sensor_get_value(SENSOR_NUM_1),
    sensor_get_value(SENSOR_NUM_2),
    sensor_get_value(SENSOR_NUM_3),
    sensor_get_value(SENSOR_NUM_4),
    sensor_get_battery(),
    gpio_get_level(GPIO_NUM_34),
    wireless_get_signal()
    );
    TRACE_LOG("MQTT", "%s", mqtt_payload);
    esp_mqtt_client_publish(mqtt_client, (char *)system_get_mqtt_topic(), mqtt_payload, strlen(mqtt_payload), 0, 0);
    
    if(mqtt_ack_detect())   /* 等待发送完成耶 */
    {
        system_set_pending_num(system_get_pending_num() + 1);
        return ;
    }`
I don't use cjson, because the data is relatively simple, I don't want to use memory allocation all the direct allocation of strings. I will continue to find the next question, thank you.

Alvin1Zhang, 2019-05-23:
@Feng573795502 Hi, thanks for reporting the issue. Has your issue been fixed? If you have other question, please create a new issue. Thanks.

lettierimarco, 2019-05-28:
Hi, i've noticed the memory leak too.
To reproduce i connect to mqtt server via ssl.
I used the example mqtt ssl to reproduce

[BEFORE CONNECT MQTT] Free memory: 281024 bytes
[AFTER CONNECT MQTT] Free memory: 245080 bytes
*REBOOT THE ROUTER TO MAKE RECONNECT*
[BEFORE CONNECT MQTT] Free memory: 285624 bytes
[AFTER CONNECT MQTT] Free memory: 244396 bytes
*REBOOT THE ROUTER TO MAKE RECONNECT*
[BEFORE CONNECT MQTT] Free memory: 284924 bytes
[AFTER CONNECT MQTT] Free memory: 244280 bytes

david-cermak, 2019-05-31:
Hi @lettierimarco , 

Does running the ssl example exhaust entire memory if you periodically reset your router? It should not, the logs you are sharing describe still an expected behaviour, free memory decreases at the beginning. but should stabilise after some time. The reason is allocation of socket fields in TCP-IP stack (those fields are never freed, so when max nr of sockets is reached these values are reused).

I have tested this scenario of periodic reconnection triggered every minute over night and have seen no leak (numbers like 162724, 162720, 162728, 162728 in the log). Note: Reduced nr. of sockets in menuconfig to get around issue described above, reconnection triggered with `esp_eth_init` (tested on ethernet, but wifi behaves the same). Tested current master branch v4.0-dev

Alvin1Zhang, 2019-07-19:
@lettierimarco Thanks for your feedback, would you please help share if any updates for your issue? Thanks.

Alvin1Zhang, 2019-10-11:
@Feng573795502 Thanks for reporting, and will close the ticket for now due to short of feedback. Feel free to reopen if you still have the issue, thanks.

closed 2019-10-11 by Alvin1Zhang