espressif/esp-idf #3498: Link to documentation of esp-idf 3.3 beta 3 is not working.

Zeni241, 2019-05-19:
Below link to documentation of esp-idf 3.3 beta 3  is not working.

 https://docs.espressif.com/projects/esp-idf/en/v3.3-beta3/

projectgus, 2019-05-20:
Hi @Zeni241,

Thanks for the heads-up. ReadTheDocs (who we use) updated their version of breathe and it broke our builds. We'll have this fixed before the v3.3 final release, but the beta3 docs will probably not build.

I'll try to provide a working link to v3.3 docs ASAP.

Angus

projectgus, 2019-05-23:
Reopening until we have 3.3 docs building again.

Zeni241, 2019-05-23:
BTW when can we expect release of stable 3.3? Just curious :) .

projectgus, 2019-08-01:
Hi @Zeni241 ,

Sorry for the slow reply. Final release was blocked on one high severity Bluetooth related issue, but it should be next week at the latest.

In the meantime you can view the 3.3 prerelease docs here:
https://docs.espressif.com/projects/esp-idf/en/release-v3.3/

(Ignore "beta1" in the description, this is a bug because the b3.3-beta3 tag was not created as an annotated tag.)

projectgus, 2019-08-19:
Thanks again for your patience updating this issue.

v3.3-rc is the release candidate for v3.3 and the docs can be found here:
https://docs.espressif.com/projects/esp-idf/en/v3.3-rc/

(Note the above link will probably stop working once the v3.3 final release is made.)

closed 2019-08-19 by projectgus

Zeni241, 2019-08-19:
Thanks for update.