espressif/esp-idf #3499: Wifi WPS example not compiling as c++

exocode, 2019-05-19:
Hi there,

I am quite new on C and C++. I try to write the code for an ESP32 (Heltec) in C++.

Is this not possible? I thought I can use C++ with ESP-IDF 
_(I am reading Kolbans Book simultaneously)._

I simply changed the name of my `main.c` to `main.cpp` and prepended the `app_main()` with `extern "C"`.

## Environment

- Development Kit:      [Heltec EPS32]
- Module or chip used:  [ESP32-ESP32]
- IDF version (run ``git describe --tags`` to find it): 
    // v3.2-dev-1148-g96cd3b75c
- Build System:         [Platformio 1.8 (ESP-IDF ]
- Compiler version (run ``xtensa-esp32-elf-gcc --version`` to find it):
    // xtensa-esp32-elf-gcc (crosstool-NG crosstool-ng-1.22.0-80-g6c4433a) 5.2.0
- Operating System:     [macOS]
- Power Supply:         [USB]

## Problem Description

I try the WiFi WPS example with C++ (cpp).
FROM The **ESP-IDF Version 3.3**

### Expected Behavior

Successful build

### Actual Behavior

Won't compile

### Steps to reproduce

1. copy/paste the example code from here
    - https://github.com/espressif/esp-idf/blob/v3.3-dev/examples/wifi/wps/main/wps.c

2. Rename the `main.c` to `main.cpp`
3. Change `app_main(){}` to `extern "C" void app_main()`
4. try to compile!

// It helps if you attach a picture of your setup/wiring here.

### Code to reproduce this issue

```cpp
#include "freertos/FreeRTOS.h"
#include "freertos/event_groups.h"
#include "esp_wifi.h"
#include "esp_log.h"
#include "esp_wps.h"
#include "esp_event_loop.h"
#include "nvs_flash.h"

static const char *TAG = "example_wps";
static esp_wps_config_t wpsConfig = WPS_CONFIG_INIT_DEFAULT(WPS_TYPE_PBC);

static esp_err_t event_handler(void *ctx, system_event_t *event)
{
    switch (event->event_id)
    {
    case SYSTEM_EVENT_STA_START:
        ESP_LOGI(TAG, "SYSTEM_EVENT_STA_START");
        break;
    case SYSTEM_EVENT_STA_GOT_IP:
        ESP_LOGI(TAG, "SYSTEM_EVENT_STA_GOT_IP");
        ESP_LOGI(TAG, "GOT IP: %s\n",
                 ip4addr_ntoa(&event->event_info.got_ip.ip_info.ip));
        break;
    case SYSTEM_EVENT_STA_DISCONNECTED:
        ESP_LOGI(TAG, "SYSTEM_EVENT_STA_DISCONNECTED");
        ESP_ERROR_CHECK(esp_wifi_connect());
        break;
    case SYSTEM_EVENT_STA_WPS_ER_SUCCESS:
        /*point: the function esp_wifi_wps_start() only get ssid & password
	 * so call the function esp_wifi_connect() here
	 * */
        ESP_LOGI(TAG, "SYSTEM_EVENT_STA_WPS_ER_SUCCESS");
        ESP_ERROR_CHECK(esp_wifi_wps_disable());
        ESP_ERROR_CHECK(esp_wifi_connect());
        break;
    case SYSTEM_EVENT_STA_WPS_ER_FAILED:
        ESP_LOGI(TAG, "SYSTEM_EVENT_STA_WPS_ER_FAILED");
        ESP_ERROR_CHECK(esp_wifi_wps_disable());
        ESP_ERROR_CHECK(esp_wifi_wps_enable(&wpsConfig));
        ESP_ERROR_CHECK(esp_wifi_wps_start(0));
        break;
    case SYSTEM_EVENT_STA_WPS_ER_TIMEOUT:
        ESP_LOGI(TAG, "SYSTEM_EVENT_STA_WPS_ER_TIMEOUT");
        ESP_ERROR_CHECK(esp_wifi_wps_disable());
        ESP_ERROR_CHECK(esp_wifi_wps_enable(&wpsConfig));
        ESP_ERROR_CHECK(esp_wifi_wps_start(0));
        break;
    default:
        break;
    }
    return ESP_OK;
}

/*init wifi as sta and start wps*/
static void start_wps(void)
{
    tcpip_adapter_init();
    ESP_ERROR_CHECK(esp_event_loop_init(event_handler, NULL));

    wifi_init_config_t wifiConfig = WIFI_INIT_CONFIG_DEFAULT();

    ESP_ERROR_CHECK(esp_wifi_init(&wifiConfig));

    ESP_ERROR_CHECK(esp_wifi_set_mode(WIFI_MODE_STA));
    ESP_ERROR_CHECK(esp_wifi_start());

    ESP_LOGI(TAG, "Starting WPS...");

    ESP_ERROR_CHECK(esp_wifi_wps_enable(&wpsConfig));
    ESP_ERROR_CHECK(esp_wifi_wps_start(0));
}

// prepend `extern "C"` here
extern "C" void app_main()
{
    /* Initialize NVS — it is used to store PHY calibration data */
    esp_err_t flashInitReturn = nvs_flash_init();
    if (flashInitReturn == ESP_ERR_NVS_NO_FREE_PAGES || flashInitReturn == ESP_ERR_NVS_NEW_VERSION_FOUND)
    {
        ESP_ERROR_CHECK(nvs_flash_erase());
        flashInitReturn = nvs_flash_init();
    }
    ESP_ERROR_CHECK(flashInitReturn);

    start_wps();
}
```
// If your code is longer than 30 lines, [GIST](https://gist.github.com) is preferred.

## Debug Logs

```
> Executing task: platformio run <

Processing heltec_wifi_kit_32 (platform: espressif32; board: heltec_wifi_kit_32; framework: espidf)
--------------------------------------------------------------------------------------------------------------------------------------------------
Verbose mode can be enabled via `-v, --verbose` option
CONFIGURATION: https://docs.platformio.org/page/boards/espressif32/heltec_wifi_kit_32.html
PLATFORM: Espressif 32 > Heltec WiFi Kit 32
HARDWARE: ESP32 240MHz 320KB RAM (4MB Flash)
DEBUG: CURRENT(esp-prog) EXTERNAL(esp-prog, iot-bus-jtag, jlink, minimodule, olimex-arm-usb-ocd, olimex-arm-usb-ocd-h, olimex-arm-usb-tiny-h, olimex-jtag-tiny, tumpa)
Library Dependency Finder -> http://bit.ly/configure-pio-ldf
LDF MODES: FINDER(chain) COMPATIBILITY(soft)
Collected 0 compatible libraries
Scanning dependencies...
No dependencies
Compiling .pioenvs/heltec_wifi_kit_32/src/main.o
In file included from src/main.cpp:5:0:
/Users/janjezek/.platformio/packages/framework-espidf/components/esp32/include/esp_wps.h:86:1: error: C99 designator 'manufacturer' outside aggregate initializer
}
^
src/main.cpp:10:37: note: in expansion of macro 'WPS_CONFIG_INIT_DEFAULT'
static esp_wps_config_t wpsConfig = WPS_CONFIG_INIT_DEFAULT(WPS_TYPE_PBC);
^
/Users/janjezek/.platformio/packages/framework-espidf/components/esp32/include/esp_wps.h:86:1: error: C99 designator 'model_number' outside aggregate initializer
}
^
src/main.cpp:10:37: note: in expansion of macro 'WPS_CONFIG_INIT_DEFAULT'
static esp_wps_config_t wpsConfig = WPS_CONFIG_INIT_DEFAULT(WPS_TYPE_PBC);
^
/Users/janjezek/.platformio/packages/framework-espidf/components/esp32/include/esp_wps.h:86:1: error: C99 designator 'model_name' outside aggregate initializer
}
^
src/main.cpp:10:37: note: in expansion of macro 'WPS_CONFIG_INIT_DEFAULT'
static esp_wps_config_t wpsConfig = WPS_CONFIG_INIT_DEFAULT(WPS_TYPE_PBC);
^
/Users/janjezek/.platformio/packages/framework-espidf/components/esp32/include/esp_wps.h:86:1: error: C99 designator 'device_name' outside aggregate initializer
}
^
src/main.cpp:10:37: note: in expansion of macro 'WPS_CONFIG_INIT_DEFAULT'
static esp_wps_config_t wpsConfig = WPS_CONFIG_INIT_DEFAULT(WPS_TYPE_PBC);
^
*** [.pioenvs/heltec_wifi_kit_32/src/main.o] Error 1
=========================================================== [ERROR] Took 16.21 seconds ===========================================================
The terminal process terminated with exit code: 1

Terminal will be reused by tasks, press any key to close it.

```

## Other items if possible

- [X] platformio.ini file
    - But I am quite sure, that you will have the same compiling error.

```
[env:heltec_wifi_kit_32]
platform = espressif32
board = heltec_wifi_kit_32
framework = espidf
```

negativekelvin, 2019-05-19:
https://github.com/espressif/esp-idf/pull/2869

mahavirj, 2020-02-12:
@exocode Please help to check fix https://github.com/espressif/esp-idf/commit/50588ff3f7cab7baa99091535e3b54991d55af07#diff-64c6b86dcd8c64c62ee1ef3380d4cefc. If it works for you, please help to close the issue.

mahavirj, 2020-02-21:
Closing due to lack of response. Please feel free to re-open if you still face an issue.

closed 2020-02-21 by mahavirj