espressif/esp-idf #3500: Crash from in ppTask

hvishu, 2019-05-20:
In our board, we are seeing crash in ppTask.  Below is the partial log.  Attached is the call trace and bootup log.
After the rebooting from the assert, board doesn't recovers, assert keep's happening at the same place.  Using Reset pin or Power cycling the board recovers the problem.

Please check and provide us some pointers.

.
.
.

[Sun May 19 02:01:43.779 2019] assertion "(res == portMUX_FREE_VAL) == (mux->count == 0)" failed: file "C:/ESP_Latest/msys32/home/sjoteppagol/esp-idf-v3.2.beta1.tip/components/freertos/portmux_impl.inc.h", line 106, function: vPortCPUAcquireMutexIntsDisabledInternal

[Sun May 19 02:01:43.795 2019] abort() was called at PC 0x401090bf on core 0

[Sun May 19 02:01:43.795 2019]

[Sun May 19 02:01:43.795 2019] Backtrace: 0x40096e1c:0x3ffe53d0 0x40097049:0x3ffe53f0 0x401090bf:0x3ffe5410 0x40093bdd:0x3ffe5440 0x401758fd:0x3ffe5460 0x401759ff:0x3ffe5480 0x401c6a50:0x3ffe54a0 0x4019c616:0x3ffe5500 0x401893c1:0x3ffe5580 0x40189ccd:0x3ffe55a0 0x4018aa2d:0x3ffe56c0 0x4019437d:0x3ffe56e0 0x4008eb32:0x3ffe5730 0x40094cc5:0x3ffe5770

[Sun May 19 02:01:43.826 2019]

[Sun May 19 02:01:43.826 2019] Rebooting...

[Backtraces.txt](https://github.com/espressif/esp-idf/files/3198162/Backtraces.txt)
[bootupLog.txt](https://github.com/espressif/esp-idf/files/3198164/bootupLog.txt)

projectgus, 2019-05-21:
Hi @hvishu,

The assertion is failing because one of the "portmux" spinlocks has become invalid, in this case the `aes_spinlock` variable defined in `aes.c`. A portmux variable has two 32-bit fields - one is an "owner" field and one is a "count" variable. The assertion is failing because the "owner" seems to be set correctly, but the "count" doesn't match the expected value.

The most likely root cause is that something else has overwritten this memory address when it shouldn't be (Stray pointer, buffer overflow/underflow, etc). Unfortunately this could have happened anywhere in the app.

Can you please try looking at the output of `nm -n build/my-app.elf`, then look for the value `aes_spinlock` and look at the values before and after it in the address space? These may be useful clues as these variables have the addresses which are closest to the address of `aes_spinlock`, so a buffer overflow/underflow may lead to them overwriting the spinlock. But it's also possible for the cause to not be nearby at all.

The fact that a CPU reset doesn't solve the problem is very unusual. What peripherals are you using in your project? Do you have any interrupt service routines configured for them?

A temporary workaround for this exact failure would be to disable [AES hardware acceleration](https://docs.espressif.com/projects/esp-idf/en/latest/api-reference/kconfig.html#config-mbedtls-hardware-aes), but I think probably the root cause is not in the AES accelerator so the bug will need to be fixed in another way, disabling AES will just cause some other variable(s) to be corrupted instead.

hvishu, 2019-05-21:
Thanks @projectgus for the details.
Attached is the nm output for your reference.
From the attachment you can see that after aes_spinlock, g_wifi_event_mask is present, I see this variable is coming from 80211 library.  
3ffc0264 D rtc_spinlock
3ffc0270 d s_config
3ffc02a0 D g_wifi_osi_funcs
3ffc0424 d aes_spinlock
3ffc042c D g_wifi_event_mask
3ffc0430 D ccmp
3ffc0448 D TmpSTAAPCloseAP

Also, when the crash happens,  below log shows that it seems to be connecting to the network.  So it could be the variable "g_wifi_event_mask" which might be doing the memory corruption, though I am just speculating. 
[Sun May 05 07:31:49.460 2019] I (3629) wifi: state: init -> auth (b0)
[Sun May 05 07:31:49.491 2019] I (3649) wifi: state: auth -> assoc (0)
[Sun May 05 07:31:49.507 2019] I (3659) wifi: state: assoc -> run (10)
[Sun May 05 07:31:49.522 2019] assertion "(res == portMUX_FREE_VAL) == (mux->count == 0)" failed: file "C:/ESP_Latest/msys32/home/sjoteppagol/esp-idf-v3.2.beta1.tip/components/freertos/portmux_impl.inc.h", line 106, function: vPortCPUAcquireMutexIntsDisabledInternal
[Sun May 05 07:31:49.538 2019] abort() was called at PC 0x401090bf on core 0

We are trying to get the issue reproduced, if so, I am planning to put a print statement to make sure that this issue is indeed caused by the memory corruption aes_spinlock->count, will let you know.  Any other inputs, or suggestions please let me know.
Regards,
vishwa

[nm.txt](https://github.com/espressif/esp-idf/files/3201694/nm.txt)

projectgus, 2019-05-21:
Hi vishwa,

Thanks for the details, agree that's a very strong hint about what might be writing out of bounds.

Just to check: Do you get any warnings in the build about submodule mismatch (or anything in the "git status" output that suggests wifi library submodule is not up to date?

If you're able to dump the value of both mux->owner and mux->count when the assert fails, that could be very helpful as well.

Angus

projectgus, 2019-05-22:
I also noticed you have quite an old development version of v3.3. Please consider updating to v3.3-beta3, or the release/v3.3 branch - which should be compatible with what you have, but more stable. The v3.3 final release should be available soon and you could consider updating to this as well. See [Versions](https://docs.espressif.com/projects/esp-idf/en/latest/versions.html) for a full explanation of IDF versioning.

hvishu, 2019-05-22:
Just to check: Do you get any warnings in the build about submodule mismatch (or anything in the "git status" output that suggests wifi library submodule is not up to date?
[Vishwa] Git status doesn't give any warning like that

If you're able to dump the value of both mux->owner and mux->count when the assert fails, that could be very helpful as well.
[vishwa]  Not easy to reproduce the issue, trying to find a reproduce sequence then I will add the prints

Regarding the version, something strange,
git status shows below
On branch release/v3.2
Your branch is up-to-date with 'origin/release/v3.2'.

git log shows
root@vishwanatha-L460:~/esp/esp-idf-v3.2.beta1.tip# git log
commit 8a69ffc36f166c964d7c30188a865847bf199561
Merge: a437e4d 95c7f45
Author: Jiang Jiang Jian <jack@espressif.com>
Date:   Fri Feb 1 21:34:05 2019 +0800

Not sure why the IDF version is showing as 3.3. 

To cross check this, we used the release version 3.2 (actually plan is to migrate to 3.2, there it show the IDF version as 3.2 in bootup log)

projectgus, 2019-05-23:
Hmm, that is unusual - I see the same here with `git describe`. We tagged v3.3-dev a few commits before v3.2-beta1, due to a late-breaking bug we fast-forwarded the release/v3.2 branch instead of backporting. So there are a few very early commits on the release/v3.2 branch that `git describe` as v3.3-dev.

Please update to v3.2 release version (including `git submodule update --init --recursive`) and let us know if this crash still occurs.

projectgus, 2019-06-04:
Hi @hvishu ,

Did you get a chance to update to the v3.2 release version?

Angus

hvishu, 2019-06-16:
Still looking at the issue, mainly trying to get the reproduce the sequence with more boards.  WIll keep you posted

Alvin1Zhang, 2019-10-11:
@hvishu Thanks for reporting, would you please help share any updates for the issue? Thanks.

projectgus, 2019-12-15:
Closing due to lack of updates.

closed 2019-12-15 by projectgus