espressif/esp-idf #3503: How to reach the data read from the server using esp_http_client_perform

sysizlayan, 2019-05-20:
Hi,
I could not find a function which can be used to read the data collected during esp_http_client_perform function.

esp_http_client_read calls esp_transport_read, which requires the connection. However, I want to reach the data that is already in the client's buffer. 

Is there a function for that?

Thanks in advance.

mahavirj, 2019-05-23:
@sysizlayan Please refer to https://docs.espressif.com/projects/esp-idf/en/latest/api-reference/protocols/esp_http_client.html#application-example (especially `HTTP_EVENT_ON_DATA`)

sysizlayan, 2019-05-28:
Does it have to be in the event handler? Is there any other way of handling the data afterward? I assume the response data is buffered, right?

chegewara, 2019-05-28:
Here is some snippet from code ive been using:
```
/* ESP HTTP Client Example

   This example code is in the Public Domain (or CC0 licensed, at your option.)

   Unless required by applicable law or agreed to in writing, this
   software is distributed on an "AS IS" BASIS, WITHOUT WARRANTIES OR
   CONDITIONS OF ANY KIND, either express or implied.
*/

#include <string.h>
#include <stdlib.h>
#include "freertos/FreeRTOS.h"
#include "freertos/task.h"
#include "esp_log.h"
#include "esp_system.h"
#include "nvs_flash.h"
#include "app_wifi.h"

#include "esp_http_client.h"

#define MAX_HTTP_RECV_BUFFER 512
static const char *TAG = "HTTP_CLIENT";

extern const char server_root_cert_pem_start[] asm("_binary_server_root_cert_pem_start");
extern const char server_root_cert_pem_end[]   asm("_binary_server_root_cert_pem_end");
static uint32_t total_len = 0;
char* response;

esp_err_t _http_event_handler(esp_http_client_event_t *evt)
{
	switch (evt->event_id) {
	case HTTP_EVENT_ERROR:
		ESP_LOGD(TAG, "HTTP_EVENT_ERROR");
		break;
	case HTTP_EVENT_ON_CONNECTED:
		ESP_LOGD(TAG, "HTTP_EVENT_ON_CONNECTED");
		break;
	case HTTP_EVENT_HEADER_SENT:
		ESP_LOGD(TAG, "HTTP_EVENT_HEADER_SENT");
		break;
	case HTTP_EVENT_ON_HEADER:
		ESP_LOGD(TAG, "HTTP_EVENT_ON_HEADER, key=%s, value=%s", evt->header_key, evt->header_value);
		break;
	case HTTP_EVENT_ON_DATA:
		ESP_LOGD(TAG, "HTTP_EVENT_ON_DATA, len=%d", evt->data_len);
		
		if (!esp_http_client_is_chunked_response(evt->client)) {
			// Write out data
			memcpy(response + total_len, (char*)evt->data, evt->data_len);
			total_len += evt->data_len;
		}

		break;
	case HTTP_EVENT_ON_FINISH:
		ESP_LOGD(TAG, "HTTP_EVENT_ON_FINISH");
		break;
	case HTTP_EVENT_DISCONNECTED:
		ESP_LOGD(TAG, "HTTP_EVENT_DISCONNECTED");
		break;
	}
	return ESP_OK;
}

// #include "mbedtls/base64.h"
// #include "http_server.h"

void https(void* data) 
{
	total_len = 0;
	response = heap_caps_calloc(1024*100, 1, MALLOC_CAP_SPIRAM);
	char request[120]; 
...

	esp_http_client_config_t config = {
		.url = request,
		.event_handler = _http_event_handler,
		.cert_pem = server_root_cert_pem_start,
	//	.timeout_ms = 20000,
	};

	esp_http_client_handle_t client = esp_http_client_init(&config);
	esp_http_client_set_method(client, HTTP_METHOD_POST);
	esp_http_client_set_header(client, "Content-Type", "image/jpeg");
	// ESP_LOGW(TAG, "post data size: %d", buff_size);
	esp_http_client_set_post_field(client, (const char*)fb_data->base64_data, fb_data->base64_size);
	esp_err_t err = esp_http_client_perform(client);
	if (err == ESP_OK) {
		fb_data->resp_len = esp_http_client_get_content_length(client);
		ESP_LOGI(TAG, "HTTP POST Status = %d, content_length = %d",
			esp_http_client_get_status_code(client),
			fb_data->resp_len);
	}
	else {
		ESP_LOGE(TAG, "HTTP POST request failed: %s", esp_err_to_name(err));
	}
	esp_http_client_cleanup(client);
	// use variable response here
}
```

mahavirj, 2019-06-21:
@sysizlayan You may also refer to https://github.com/espressif/esp-idf/blob/master/examples/protocols/esp_http_client/main/esp_http_client_example.c#L420

mahavirj, 2019-07-02:
@sysizlayan Presumably this was a question, so I will go ahead and close this. Please feel free to update if you still notice any issue here.

closed 2019-07-02 by mahavirj