espressif/esp-idf #3505: SmartConfig example does not save it's credentials

exocode, 2019-05-20:
## Environment

- Development Kit:      [Heltec EPS32]
- Module or chip used:  [ESP32-ESP32]
- IDF version (run ``git describe --tags`` to find it): 
    // **v3.2-dev-1148-g96cd3b75c**
- Build System:         [Platformio 1.8 (ESP-IDF ]
- Compiler version (run ``xtensa-esp32-elf-gcc --version`` to find it):
    // xtensa-esp32-elf-gcc (crosstool-NG crosstool-ng-1.22.0-80-g6c4433a) 5.2.0
- Operating System:     [macOS]
- Power Supply:         [USB]

## Problem Description

When use the SmartConfig example, assign a network and reboot the ESP32, the credentials are not saved. 

### Expected Behavior

When I reboot ESP32 the saved credentials should be recalled
and ESP32 should connect automatically to latest working Wifi

### Actual Behavior

When I reboot a before successfully connected ESP32, it keeps hanging in SC_STATUS_FIND_CHANNEL (waiting for new credentials)

### Steps to reproduce

1. copy paste the smartConfig example: https://github.com/espressif/esp-idf/tree/master/examples/wifi/smart_config
2. build and run the example
3. Download and install one of Espressif's Mobile phone apps and connect to a Wifi Network
4. Connect the ESP32 to the Wifi Network via the mobile app
5. Wait till it's connected
6. press the reboot button on your board
7. see that SC_STATUS_FIND_CHANNEL is active (which shouldn't)

### Code to reproduce this issue

Use your example code
https://github.com/espressif/esp-idf/blob/master/examples/wifi/smart_config/main/smartconfig_main.c

## Debug Logs

#### First boot

```
[0;32mI (300) phy: phy_version: 4008, c9ae59f, Jan 25 2019, 16:54:06, 1, 0␛[0m
I (300) wifi: mode : sta (cc:50:e3:81:be:78)
I (350) smartconfig: SC version: V2.8.0
I (5170) wifi: ic_enable_sniffer
SC_STATUS_FIND_CHANNEL
I (23080) smartconfig: TYPE: ESPTOUCH
I (23080) smartconfig: T|PHONE MAC:e0:5f:45:4d:b5:72
I (23080) smartconfig: T|AP MAC:90:5c:44:1d:56:be
SC_STATUS_GETTING_SSID_PSWD
SC_TYPE:SC_TYPE_ESPTOUCH
I (26400) smartconfig: T|pswd: MYSSID_NAME
I (26400) smartconfig: T|ssid: MYSSID_PASSWORD
I (26400) smartconfig: T|bssid: 90:5c:44:1d:56:be
I (26400) wifi: ic_disable_sniffer
SC_STATUS_LINK
I (26590) wifi: n:8 0, o:1 0, ap:255 255, sta:8 0, prof:13
I (27420) wifi: state: init -> auth (b0)
I (27430) wifi: state: auth -> assoc (0)
I (27440) wifi: state: assoc -> run (10)
I (28240) wifi: connected with save_the_planet, channel 8
I (28250) wifi: pm start, type: 1

␛[0;32mI (35660) event: sta ip: 192.168.0.185, mask: 255.255.255.0, gw: 192.168.0.1␛[0m
␛[0;32mI (35660) SMARTCONFIG: WiFi Connected to ap␛[0m
SC_STATUS_LINK_OVER
Phone   ip:     192.168.0. 150
```

#### After reboot
```
ets Jun  8 2016 00:22:57

rst:0x1 (POWERON_RESET),boot:0x17 (SPI_FAST_FLASH_BOOT)
configsip: 0, SPIWP:0xee
clk_drv:0x00,q_drv:0x00,d_drv:0x00,cs0_drv:0x00,hd_drv:0x00,wp_drv:0x00
mode:DIO, clock div:2
load:0x3fff0018,len:4
load:0x3fff001c,len:6128
ho 0 tail 12 room 4
load:0x40078000,len:8816
load:0x40080400,len:6392
entry 0x40080744
␛[0;32mI (31) boot: ESP-IDF 3.30200.190418 2nd stage bootloader␛[0m
␛[0;32mI (31) boot: compile time 17:29:50␛[0m
␛[0;32mI (40) boot: Enabling RNG early entropy source...␛[0m
␛[0;32mI (40) boot: SPI Speed      : 40MHz␛[0m
␛[0;32mI (40) boot: SPI Mode       : DIO␛[0m
␛[0;32mI (44) boot: SPI Flash Size : 4MB␛[0m
␛[0;32mI (48) boot: Partition Table:␛[0m
␛[0;32mI (51) boot: ## Label            Usage          Type ST Offset   Length␛[0m
␛[0;32mI (58) boot:  0 nvs              WiFi data        01 02 00009000 00006000␛[0m
␛[0;32mI (66) boot:  1 phy_init         RF data          01 01 0000f000 00001000␛[0m
␛[0;32mI (73) boot:  2 factory          factory app      00 00 00010000 00100000␛[0m
␛[0;32mI (81) boot: End of partition table␛[0m
␛[0;32mI (85) esp_image: segment 0: paddr=0x00010020 vaddr=0x3f400020 size=0x13eb0 ( 81584) map␛[0m
␛[0;32mI (122) esp_image: segment 1: paddr=0x00023ed8 vaddr=0x3ffbdb60 size=0x02e74 ( 11892) load␛[0m
␛[0;32mI (127) esp_image: segment 2: paddr=0x00026d54 vaddr=0x40080000 size=0x00400 (  1024) load␛[0m
␛[0;32mI (129) esp_image: segment 3: paddr=0x0002715c vaddr=0x40080400 size=0x08eb4 ( 36532) load␛[0m
␛[0;32mI (152) esp_image: segment 4: paddr=0x00030018 vaddr=0x400d0018 size=0x651a0 (414112) map␛[0m
␛[0;32mI (296) esp_image: segment 5: paddr=0x000951c0 vaddr=0x400892b4 size=0x073f4 ( 29684) load␛[0m
␛[0;32mI (318) boot: Loaded app from partition at offset 0x10000␛[0m
␛[0;32mI (318) boot: Disabling RNG early entropy source...␛[0m
␛[0;32mI (318) cpu_start: Pro cpu up.␛[0m
␛[0;32mI (322) cpu_start: Starting app cpu, entry point is 0x40081fa0␛[0m
␛[0;32mI (0) cpu_start: App cpu up.␛[0m
␛[0;32mI (332) heap_init: Initializing. RAM available for dynamic allocation:␛[0m
␛[0;32mI (339) heap_init: At 3FFAE6E0 len 0000F480 (61 KiB): DRAM␛[0m
␛[0;32mI (345) heap_init: At 3FFC6AC8 len 00019538 (101 KiB): DRAM␛[0m
␛[0;32mI (351) heap_init: At 3FFE0440 len 00003AE0 (14 KiB): D/IRAM␛[0m
␛[0;32mI (358) heap_init: At 3FFE4350 len 0001BCB0 (111 KiB): D/IRAM␛[0m
␛[0;32mI (364) heap_init: At 400906A8 len 0000F958 (62 KiB): IRAM␛[0m
␛[0;32mI (370) cpu_start: Pro cpu start user code␛[0m
␛[0;32mI (53) cpu_start: Starting scheduler on PRO CPU.␛[0m
␛[0;32mI (0) cpu_start: Starting scheduler on APP CPU.␛[0m
I (140) wifi: wifi driver task: 3ffb7c48, prio:23, stack:3584, core=0
I (140) wifi: wifi firmware version: 9415913
I (140) wifi: config NVS flash: enabled
I (140) wifi: config nano formating: disabled
␛[0;32mI (150) system_api: Base MAC address is not set, read default base MAC address from BLK0 of EFUSE␛[0m
␛[0;32mI (160) system_api: Base MAC address is not set, read default base MAC address from BLK0 of EFUSE␛[0m
I (200) wifi: Init dynamic tx buffer num: 32
I (200) wifi: Init data frame dynamic rx buffer num: 32
I (200) wifi: Init management frame dynamic rx buffer num: 32
I (200) wifi: Init static rx buffer size: 1600
I (210) wifi: Init static rx buffer num: 10
I (210) wifi: Init dynamic rx buffer num: 32
␛[0;32mI (300) phy: phy_version: 4008, c9ae59f, Jan 25 2019, 16:54:06, 1, 0␛[0m
I (300) wifi: mode : sta (cc:50:e3:81:be:78)
I (350) smartconfig: SC version: V2.8.0
I (5170) wifi: ic_enable_sniffer
SC_STATUS_FIND_CHANNEL

```

## Other items if possible

- [X] using PlatformIO

espxiehang, 2019-05-27:
hi, @exocode  The example will start smartconfig after reboot.
log is : 
I (300) wifi: mode : sta (cc:50:e3:81:be:78)
I (350) smartconfig: SC version: V2.8.0
I (5170) wifi: ic_enable_sniffer
SC_STATUS_FIND_CHANNEL

In this state, it will not connect to the wireless network connection, but wait for the ESP_touch distribution network.

If you need it to save it's credentials,ou can modify the code to start smatrconfig with a command.

exocode, 2019-05-27:
Ok I understand, thank you.
I suggest for consistency extending the example with the permanent store of the credentials as it is in the WPS config example. Most use cases are using permanent storage.

espxiehang, 2019-11-06:
Thank you for your suggestion.

Alvin1Zhang, 2019-11-06:
@exocode Thanks for reporting. Feel free to reopen if you still have the issue. Thanks.

closed 2019-11-06 by Alvin1Zhang

tavdog, 2023-07-14:
Was this implemented or not ? The example still does not save the wifi creds.

isfandyarIOTMAV, 2024-02-21:
> Was this implemented or not ? The example still does not save the wifi creds.

still does not save credentials